sonyxperiadev/kernel #187: amami display

CoolDevelopment, 2015-02-22:
The display does not work with the 3.10 kernel and amami. The kernel logo and everything else does not show up. 
I have tried the kernel with freexperias combined-rootfs and inside recovery(twrp) the display works, but the recovery ui is below the kernel logo, which shows up after boot.
I have rebuilt android with 3.10 headers, but the display is still not working.

kholk, 2015-02-22:
At the moment, it is expected for Aries to not show the logo.
About Android, sure it's mandatory to build it with 3.10 headers (also you can find 'em in master branch of device-sony-shinano, device-sony-rhine, device-sony-yukon), otherwise multiple things, like display, won't work.

By the way, can you provide logcat and dmesg?

galaxyfreak, 2015-02-22:
Good to know I'm not alone with that logo in recovery. filled a separate issue on that.

CoolDevelopment, 2015-02-22:
I am using amami(z1c) not aries, i have built android with 3.10 headers
I will post a logcat and dmesg later

kholk, 2015-02-22:
Oh!!! Sorry, I misread!

About amami, there's actually a trick to make it to work. This actually is a known bug: to circumvent this we're letting it to boot into Android.
Then, you should "power off" the display (you know, like it was on, sitting at lock screen), then power it on... and it magically works.

Obviously, Android has to use the right headers and HALs for display (3.10, msm8974 hal), otherwise this trick won't work.

CoolDevelopment, 2015-02-22:
Okay thanks I will try that. Can I also add "somc,dsi-restart-hack;" to the dt of my display?

galaxyfreak, 2015-02-22:
ok, kholk, you on hangouts or irc? have one more question

kholk, 2015-02-22:
hangouts.... :)

jerpelea, 2015-03-18:
bug is fixed now !
please retest for your usecase

CoolDevelopment, 2015-03-18:
I am sorry but I have still got the issue with the newest changes of aosp/AU_LINUX_ANDROID_LA.BF.2.1_RB1.05.00.00.173.012 branch.

CoolDevelopment, 2015-03-20:
With the new 3.10 branch the issue is gone

closed 2015-03-20 by CoolDevelopment

galaxyfreak, 2015-03-20:
How about logo staying in recovery?