sonyxperiadev/kernel #275: [BF64] Kernel Build Error

necioerrante, 2015-04-12:
```
arch/arm/mach-msm/board-sony_shinano-wifi.c:25:42: fatal error: ../drivers/mmc/host/msm_sdcc.h: No such file or directory
 #include <../drivers/mmc/host/msm_sdcc.h>
                                          ^
compilation terminated.
scripts/Makefile.build:308: recipe for target 'arch/arm/mach-msm/board-sony_shinano-wifi.o' failed
make[1]: *** [arch/arm/mach-msm/board-sony_shinano-wifi.o] Error 1
Makefile:811: recipe for target 'arch/arm/mach-msm' failed
make: *** [arch/arm/mach-msm] Error 2
```

Verbose, (V=1): [ http://pastebin.com/1hsiRTPV ]

necioerrante, 2015-04-14:
With recent changes the error is now:
```
kernel/sony/msm/Android.mk:137: recipe for target 'TARGET_KERNEL_BINARIES' failed
make: [TARGET_KERNEL_BINARIES] Error 2 (ignored)
mdpath=`find out/target/product/aries/system/lib/modules -type f -name modules.order`; if [ "$mdpath" != "" ];then mpath=`dirname $mdpath`; ko=`find $mpath/kernel -type f -name *.ko`; for i in $ko; do arm-eabi-strip --strip-unneeded $i; mv $i out/target/product/aries/system/lib/modules/; done; fi
mdpath=`find out/target/product/aries/system/lib/modules -type f -name modules.order`; if [ "$mdpath" != "" ];then mpath=`dirname $mdpath`; rm -rf $mpath; fi
mdpath=`find out/target/product/aries/system/lib/modules -type f -name modules.order`; if [ "$mdpath" != "" ];then mpath=`dirname $mdpath`; ko=`find $mpath/kernel -type f -name *.ko`; for i in $ko; do arm-eabi-strip --strip-unneeded $i; mv $i out/target/product/aries/system/lib/modules/; done; fi
mdpath=`find out/target/product/aries/system/lib/modules -type f -name modules.order`; if [ "$mdpath" != "" ];then mpath=`dirname $mdpath`; rm -rf $mpath; fi
make: *** No rule to make target 'out/target/product/aries/dtbs/msm8974pro-ac-shinano_aries.dtb', needed by 'out/target/product/aries/kernel'.  Stop.

#### make failed to build some targets (23:15 (mm:ss)) ####
```
[ http://pastebin.com/98Z1gfCF ]

galaxyfreak, 2015-04-15:
1) PREFERED - make wifi build normally not as a module.
2) re-add modules support
CONFIG_MODULES=y
CONFIG_MODULE_UNLOAD=y
CONFIG_MODULE_FORCE_UNLOAD=y

jerpelea, 2015-04-15:
wifi is built in kernel and works normaly 
enabling modules is just a way of disabling a warning make: [TARGET_KERNEL_BINARIES] Error 2 (ignored)

closed 2015-04-15 by jerpelea

necioerrante, 2015-04-15:
The first error was fixed with latest commits.
But, the second error
```
make: *** No rule to make target 'out/target/product/aries/dtbs/msm8974pro-ac-shinano_aries.dtb', needed by 'out/target/product/aries/kernel'.  Stop.
```
Now occurs when trying to build.