sonyxperiadev/kernel #347: Phone do not want boot.

pawcioo132, 2015-06-28:
Hi, i try to compile aosp 5.1.1 for eagle like in sony's guide but after success compile and flash to phone, my xperia d2303 don't boot it's only one short vibrate and thats all.

galaxyfreak, 2015-06-28:
Strange, please recheck that kernel has been bulit with DTs successfully packed and included.

jerpelea, 2015-06-29:
You have to flash latest sw release first and after you have to flash the open devices bootloader (with recovery)

closed 2015-06-29 by jerpelea

pawcioo132, 2015-06-29:
i aready try that, i think problem is with DTs beacause on the end compiling kernel i have messenge "nothink to do with DTs ..." 
and next line is: "/android/kernel/sony/msm/arch/arm/boot/dts/makefile:228: warning: overriding for target 'dtbs' "
"/android/kernel/sony/msm/arch/arm/boot/dts/makefile:228: warning: ignoring old commands target 'dtbs' " 
also i add its build from fresh source, and first build.

galaxyfreak, 2015-06-29:
Do FXP builds boot for you?

kholk, 2015-06-29:
I agree with that question, knowing if FXP builds are booting is really important.

If you've got the latest bootloader and FXP builds are booting, then the problem happens on your local build, perhaps your build system may have something wrong.

In case those won't boot, then there's some unknown kernel problem. Btw, "one short vibrate and that's all" sounds like really early fail. About your DTBs messages, those should be nothing fatal.

pawcioo132, 2015-06-29:
Yes FXP build booted success for me, I use his build right now.

pawcioo132, 2015-06-29:
It's look like after vibrate phone will power off. because when I hold power button longer this short vibrations are repeat after 2 - 3 second. Maybe this information will be little helpful for you.

pawcioo132, 2015-07-02:
Okey sorry for problem but that issue was beacause i don't have some package in my build system and kernel build was not fully completed successfully.

nos1609, 2016-04-06:
I'm facing the same issue. @pawcioo132 What was the missing component?