sonyxperiadev/kernel #353: [EAGLE] GPIO Event

Noobzik, 2015-07-16:
That's affect also the D2303 model, and it's been here for a long time
http://forum.xda-developers.com/xperia-m2/general/aqua-submodels-d2403-t3111184

Linking my logcat https://gist.github.com/Noobzik/ff95417a17d4f844bbd3
the problem is the sensor HAL as pointed out by Galaxyfreak

erikcas, 2015-07-17:
@jerpelea I spend some time on this testing with an affected user some weeks ago. Tested:
* disablement of all magnetometer sensorsdrivers in kernel --> bootloop
* disablement of 1 or 2 of the magnetometer sensordrivers in all possible combinations --> bootloop
* applied all possible hacks I could trace back in stock kernel --> bootloop

* remove sensorhal on above cases: device boots w/o problems
* replace eagle blob with seagull blob with all magnetometer sensordriver enabled: device boots w/o problems, light and proximity work. magnetometer sensors donot.

for reference (on older kernel) 2 dmesgs of an M2 with this problem
* working: 
http://dl-1.va.us.xda-developers.com/3/2/7/0/7/5/9/zipkernel.txt?key=XPEJHcZE-jnBbUzOiC_WTw&ts=1437144302

* not working (bootloop):
http://dl-1.va.us.xda-developers.com/3/2/7/0/7/6/0/extrakernel.txt?key=mS1qbHJ3magnetometerdlk7yV5prbA&ts=1437144384

This is same device, the working is on stock based kernel.
Second, not working thus bootlooping, is on same rom, kernel this time is 3.5.1 unified branch.
As far as I can see, I might be wrong, magnetometer sensors are initialized correctly.

I am completely out of options concerning kernel (besides the issue I am not able to reproduce myself)

jerpelea, 2015-07-17:
Hi
Thanks for the hint
I will compare the 2 sensor sources

erikcas, 2015-07-17:
updated comment.
I see a problem on yas532 probing in later dmesgs. So there still are kernel options..

Theirs bootlooping: http://pastebin.com/HrvXvsw1 line1534 - 1538

Mine: booting as expected: https://gist.github.com/erikcas/af392942901a05d8c5bd line 742 and further

erikcas, 2015-07-18:
Update: finally received a dmesg from stock rom from this device variant. It turns out to use mmc3416x driver. (see http://pastebin.com/XhVePTxK big thanks to xda member Th3 Ant also for countless testing)

Enabled that driver in kernel but it still suffers bootloop.

updated driver in kernel: https://github.com/erikcas/kernel/commits/sony/mmc3416x

Dmesg on current change: http://pastebin.com/aMR6q1WD

Repeating sensors.msm8226.so error that crashes systemserver: http://pastebin.com/qdbZ2EVU
@jerpelea: please see line 29 in that, that is why I suspect the hal is the problem, seems to me a mismatch on definitions.

erikcas, 2015-07-19:
another small update: enabled kionix accel driver (following stock) in https://github.com/erikcas/kernel/commits/sony/kionix

still bootloop but a small change in dmesg http://pastebin.com/5wgM8CZU around line 2909.
It starts complaining about selinux for sensors.msm8226.so

This user is currently on omnirom. For full complete comparison it should be tested on aosp

galaxyfreak, 2015-07-19:
AFAIK, permissive is to print denied messages, but not actually break stuff like enforcing, so it should be okay...

erikcas, 2015-07-24:
current status:

driver is initing. Still a bootloop. Considerations:

https://github.com/erikcas/kernel/commits/sony/kionix

* added a product-id check, based on stock sources. It is done with an extern uint8_t. I have tried to replicate the product_id into both drivers but without luck, even after reading i2c documentation over and over. Possibly I am thinking off-road and solution is bloody simple. So need a solution on this

* Is mecs needed for the driver or not? As far as I can find, I understand mecs helps with translating communications from hardware to the kernel. But commit a2ce589ea470b79c2f4ee78fe405d7182d89e386 tells us that board specific configs are made to the driver to make it compatible to current kernel and hal. The former implementatations to earlier branches of this kernel are based on status on prior commit.

* if memsic still is needed, then https://github.com/erikcas/kernel/commits/sony/mecs is needed and probably shift to the "elder" mmc3416x.c driver. Which is, considering previous bullet, not compatible with current kernel. I have tested it, it makes the memsicd service function fully (i.e. no warnings or errors in strace), in commit https://github.com/erikcas/kernel 0feeee81dd4a3c18a22b89bd5c5d08bbcdd2d2fc

For now I am stuck here. Besides changing the extern uint8_t replicate to a nice clean check, I have no idea if kernelsource will work on a sensor hal. I donot know if current hal is based on situation after a2ce589ea470b79c2f4ee78fe405d7182d89e386.

@jerpelea, let me know if you want a pull request from my forks' sony/kionix branch and/or sony/mecs branch

erikcas, 2015-07-27:
https://github.com/sonyxperiadev/kernel/commit/74a9ac1e720c5aa81b95f0525f40beba4bdaf820#diff-8595c3af646bd1fa38b0820b61bf40d7R237 (line 237) gives me some hope on mmc3416x :)

Noobzik, 2015-07-27:
Anything about the two differents version of sensorhal sources @jerpelea ?

jerpelea, 2015-07-27:
unf there is no update for now

Noobzik, 2015-08-07:
What if we compile with mixed sources (sensor from stock and the rest from kernel / eagle repo)?
I doubt that it will cause more issue than now...

erikcas, 2015-08-07:
Hi,

Already did that... No success unfortunately.

jerpelea, 2015-08-07:
for now still no update on this matter
i am trying to find a device (HW) with this problem so that i can replicate locally

erikcas, 2015-08-31:
small update. Force pushed to my sony/kionix branch, see https://github.com/erikcas/kernel/commits/sony/kionix

Tiny change in DT for mmc3416x.
Follow documentary and rename mmc3416x@30 to memsic@30

Direct consequence is something showing up in dmesg I have not seen before, a selinux denial complaint of sensors.msm8226.so wants to open, read, get_attr to mmcblk0p26

See http://pastebin.com/C0BdCUT8 line 2749 (in dmesg) and line 737 (in logcat)

So progress or....??

Noobzik, 2015-10-04:
Lolipop is out, maybe there's something we can do...

erikcas, 2015-10-05:
@jerpelea I have reports of the same occurence on latest 3.4 stock sources on fresh LP blobs on the sensor issues. The rom is installed after new TA. Might this be a bootloader issue for devices with mmc3416x / kionix_accel combination?

To rule things out I will compile an "old style" combined boot/recovery on 3.4 and let that be tested on default bootloader. If the issue does not occur there, for me it is a bootloader issue and this (kernel)issue can be closed

erikcas, 2015-10-05:
on oldstyle boot.img same problem so bootloader is excluded.

aleksjp, 2015-10-07:
Can confirm for the same symptoms noticed in this [comment]( https://github.com/sonyxperiadev/kernel/issues/353#issuecomment-122602322). 
Without _sensors.msm8226.so_ library have no problem here. IMHO, that's no bootloader issue.

erikcas, 2015-11-24:
@jerpelea, this one is "solved" as far as it concerns sensorsbootloop. I.e. we are getting past the bootloop now which is NOT a kernelissue.

Solution? Might sound ridiculous, but on @galaxyfreak cm-12.1 I asked on xda to push stock system/lib/libsensorservice.so 

Question yo me is what sony changed to this. It sure calls for an extra hal present on stock eagle 18.6.A.0.175 build, i.e. system/lib/hw/sensorsecond.msm8226.so

Sensors (memsic and kionix) still need the earlier mentioned kerneldriver to work, as well as the mentioned sensorsecond.msm8226.so.

for more info see http://forum.xda-developers.com/showpost.php?p=63982527&postcount=1454 and onwards.
Only take this as a reference, the mentioned rom is running on a stockbased kernel!
It is meant to show, that the libsensorservice idea exposed the real issue

closed 2016-11-18 by jerpelea