sonyxperiadev/kernel #423: Microphone does not always work during calls

rancidfrog, 2015-09-30:
Around 2 weeks ago, I believe, there has been a re-emergence in audio routing issue regarding Mic.
Most of the time now Mic does not work, person on the other end does not hear anything even though I hear them.
Sorry I did not track the commit or exact date where the problem occurs.

http://hastebin.com/ufesexekim.xml
http://hastebin.com/iyehifugic.coffee
[ https://talk.sonymobile.com/t5/Kernel/Bugs/m-p/953195#U953195 ]

konstantinosj77, 2015-10-01:
If the device is connected to charger the problem is gone! Really strange...

humberos, 2015-11-25:
This issue was solved by this change:
https://github.com/sonyxperiadev/kernel/commit/3958115016248183a32e00e5139ca023e397b6ff

konstantinosj77, 2015-12-03:
@humberos actually not... on D2403 using latest FXP build (AOSP_L_MR1_151125_eagle.zip)

If you LOCK the phone while in call mic stops working...
If i connect my phone to my pc or to the charger and lock it everything works normally...

humberos, 2015-12-03:
@konstantinosj77 
Weird. Well, may FXP team did not get that commit ( 3958115016248183a32e00e5139ca023e397b6ff ) on it kernel tree ?

@erikcas : could you help us with it ?

erikcas, 2015-12-03:
@humberos no clue for now. The revert you refer to, does not apply to .061 (1.1) branch, we use for builds. The commit it reverts is never done on that branch.

I have experienced this on amami too some time ago, mainly in inbound calls, but cannot remember it occurred last week.

Maybe I can make a fresh build for @konstantinosj77 to test, and will also test myself (hmm.. shift from Z5 to M2 for sim :P)

konstantinosj77, 2015-12-03:
@erikcas I'm gonna build tonight using your "aosp-6.0" manifest and i'll apply that commit for testing

erikcas, 2015-12-03:
please donot, that manifest is hopeless out of date ;-)

konstantinosj77, 2015-12-03:
I've checked and it has latest kernel (LA.BF64.1.2.2_rb4.7) and *almost* up-to-date device trees
Anyway...I'm gonna start the build,if you send me anything i'll test!

erikcas, 2015-12-03:
please be advised this is a different kernelsource then that is used on L
In Dutch we say: please donot compare apples to pears...

kholk, 2015-12-04:
1.2.2_rb4.7 should need the commit referenced by @humberos.
Please note that the 1.2.2 tree is intended for Android M and, while it has been tested on L, it was just because this is what I had on my devices at the time of testing it.
The tests weren't extensive though: just a boot OK/KO test and basic functionality (modem, audio, TS, WiFi).

It is recommended to continue using the 1.1 branch on Android L because of full functionality set.
The 1.2.2 branch, for example, will support camera on Android M ONLY.

alviteri, 2015-12-04:
https://github.com/sonyxperiadev/kernel/pull/484 but @jerpelea have it alot for 1.2.2

rancidfrog, 2016-08-09:
Does not apply to new branch or blobs

closed 2016-08-09 by rancidfrog