sonyxperiadev/kernel #440: Modem missing/failed

bobobo1618, 2015-10-16:
Running on Aries (Z3 Compact) with the latest sources, binaries and instructions for Marshmallow, gives the following errors:

```
[    0.000000] cma: Found modem_region@0x8000000, memory base 0x08000000, size 82 MiB, limit 0xffffffff
[    0.000000] cma: CMA: reserved 82 MiB at 0x08000000 for modem_mem
[    0.435094] pil-q6v5-mss fc880000.qcom,mss: [modem]: Error getting proxy unvoting gpio
[    4.862099] usbserial: USB Serial support registered for Qualcomm USB modem
[    9.654970] pil-q6v5-mss fc880000.qcom,mss: modem: Failed to locate modem.mdt
[    9.654992] IPC_RTR: pil_vote_load_worker: Failed to load modem
[    9.656767] pil-q6v5-mss fc880000.qcom,mss: modem: Failed to locate modem.mdt
[    9.656787] smd_pkt_open failed on smd_pkt_dev id:0 - subsystem_get failed for modem
```

following basic problem solving steps and inserting a modem extracted from a stock Z3 Compact gives the following errors:

```
[    0.000000] cma: Found modem_region@0x8000000, memory base 0x08000000, size 82 MiB, limit 0xffffffff
[    0.000000] cma: CMA: reserved 82 MiB at 0x08000000 for modem_mem
[    0.433826] pil-q6v5-mss fc880000.qcom,mss: [modem]: Error getting proxy unvoting gpio
[    4.860271] usbserial: USB Serial support registered for Qualcomm USB modem
[   10.734125] pil-q6v5-mss fc880000.qcom,mss: modem: loading from 0x08000000 to 0x0cf00000
[   11.838341] pil-q6v5-mss fc880000.qcom,mss: modem: Invalid firmware metadata
[   11.848497] IPC_RTR: pil_vote_load_worker: Failed to load modem
[   11.851839] pil-q6v5-mss fc880000.qcom,mss: modem: loading from 0x08000000 to 0x0cf00000
[   12.951282] pil-q6v5-mss fc880000.qcom,mss: modem: Invalid firmware metadata
[   12.961448] smd_pkt_open failed on smd_pkt_dev id:0 - subsystem_get failed for modem
```

It's notable that the Sony provided binary blobs do not appear to contain a `modem.mdt`.

alviteri, 2015-10-16:
Sony Can't provide modem blobs .... you need pick them from your own device

bobobo1618, 2015-10-16:
Picking them from my own device resulted in the invalid data error I gave
in my second block of logs.

alviteri, 2015-11-09:
which device?

mikey0000, 2015-11-09:
E3 D2203

alviteri, 2015-11-09:
Did you pick modem files from stock ? Sony won't provide them on zip due to they can't be provided

mikey0000, 2015-11-09:
No I haven’t yet, I am trying to track down a stock image to grab them from.

mikey0000, 2015-11-09:
@alviteri  I've put the modem files(all of them) in and this is the last line before it reboots
```
RadioService: couldn't load radio module radio.primary (No such file or directory)
```

closed 2016-08-15 by jerpelea

mathew-dennis, 2021-06-22:
I am having the same issue on Kenzo ..... please tell me how this was fixed..