sonyxperiadev/kernel #535: Inline kernel fails to build, due to missing dtbs in output folder

rancidfrog, 2015-12-27:
Kernel fails to build inline because the dtbs folder is not copied to output directory
```
mkqcdtbootimg: failed to open '/out/target/product/aries/dtbs': No such file or directory
build/core/Makefile:566: recipe for target '/out/target/product/aries/boot.img' failed
make: *** [/out/target/product/aries/boot.img] Error 1
```
Android.mk I believe is the culprit:
https://github.com/sonyxperiadev/kernel/blob/misc-dev/Android.mk#L61
and,
https://github.com/sonyxperiadev/kernel/blob/misc-dev/Android.mk#L135

rancidfrog, 2015-12-28:
When **BUILD_KERNEL := true**
```BOARD_MKBOOTIMG_ARGS += --dt_dir $(OUT)/dtbs```
Should be
```BOARD_MKBOOTIMG_ARGS += --dt_dir $(OUT)/obj/KERNEL_OBJ/arch/arm/boot/dts```
https://github.com/sonyxperiadev/device-sony-shinano/blob/master/BoardConfig.mk#L117
https://github.com/sonyxperiadev/device-sony-shinano/blob/master/BoardConfig.mk#L39
Maybe default should be:
```BOARD_MKBOOTIMG_ARGS += --dt_dir $(OUT)/obj/KERNEL_OBJ/arch/arm/boot/dts```
And, KernelConfig.mk, https://github.com/sonyxperiadev/vendor-sony-kernel/blob/master/KernelConfig.mk , is changing to **BUILD_KERNEL := false** it should change output to
```BOARD_MKBOOTIMG_ARGS += --dt_dir $(OUT)/dtbs```

mengxiaoxi, 2015-12-28:
hi

1 check if you have downloaded and extracted prebuilt kernel in
vendor/sony/kernel folder

2 it should be boot.img  insteadof bootimg...

rancidfrog, 2016-01-06:
That is translation problem, you using translator google?
Vendor/sony/kernel is for aosp only.
And, guideline should be followed: 
```
BUILD_KERNEL := true
-include vendor/sony/kernel/KernelConfig.mk
```
Kernelconfig.mk Should change output folder to: BOARD_MKBOOTIMG_ARGS += --dt_dir $(OUT)/dtbs
The same way it changes build_kernel to false.
Because, if kernel project not included, build kernel is true, which will need output folder to be: $(OUT)/obj/KERNEL_OBJ/arch/arm/boot/dts

rancidfrog, 2016-03-30:
With new appended dtb not needed anymore

closed 2016-03-30 by rancidfrog

Mikdadali, 2020-04-02:
pls help
[image]

[image]

[image]