sonyxperiadev/kernel #658: flash the controller touchscreen Cypress TrueTouch TMA4XX

Kota-Kun, 2016-03-24:
That's my problem. To flash your device and how it decided to flash the firmware from another phone (similar in iron) and of accident either ask whether zater touchscreen controller firmware.

touch himself firm Cypress TrueTouch TMA4XX
It turned out that the touch turned to the left

I enclose a video how the phone.
http://www.youtube.com/watch?v=ZoW8iOvAXRU

**Completed actions:**

Compiling from source sonyxperiadev kernel kernel with options enabled (no they did not stitch the firmware is available):

CONFIG_INPUT_TOUCHSCREEN = y
CONFIG_TOUCHSCREEN_CYPRESS_CYTTSP4 = y
CONFIG_TOUCHSCREEN_CYPRESS_CYTTSP4_DEVICETREE_SUPPORT = y
CONFIG_TOUCHSCREEN_CYPRESS_CYTTSP4_I2C = y
CONFIG_TOUCHSCREEN_CYPRESS_CYTTSP4_MT_B = y
CONFIG_TOUCHSCREEN_CYPRESS_CYTTSP4_BINARY_FW_UPGRADE = y
CONFIG_TOUCHSCREEN_CYPRESS_CYTTSP4_PLATFORM_YUKON = y

And of course compilation for the insertion according to the instructions from Sony.

As a result, a working firmware, but the method (forced_upgrade) to stitch a forced firmware, namely forced_upgrade interface is not available. Available only manual_upgrade but how to use it do not know.

**As of adb as root**

> root@seagull:/sys/bus/i2c/devices/5-0024/main_ttsp_core.cyttsp4_i2c_adapter # 
cd cyttsp4_device_access.main_ttsp_core
root@seagull:/sys/bus/i2c/devices/5-0024/main_ttsp_core.cyttsp4_i2c_adapter/cyttsp4_device_access.main_ttsp_core # ls
driver
get_panel_data
ic_grpdata
ic_grpnum
ic_grpoffset
modalias
power
subsystem
uevent
root@seagull:/sys/bus/i2c/devices/5-0024/main_ttsp_core.cyttsp4_i2c_adapter/cyttsp4_device_access.main_ttsp_core # cd ../
root@seagull:/sys/bus/i2c/devices/5-0024/main_ttsp_core.cyttsp4_i2c_adapter # 
cd cyttsp4_loader.main_ttsp_core
root@seagull:/sys/bus/i2c/devices/5-0024/main_ttsp_core.cyttsp4_i2c_adapter/cyttsp4_loader.main_ttsp_core # ls
driver
manual_upgrade
modalias
power
subsystem
uevent
root@seagull:/sys/bus/i2c/devices/5-0024/main_ttsp_core.cyttsp4_i2c_adapter/cyttsp4_loader.main_ttsp_core # cd ../
root@seagull:/sys/bus/i2c/devices/5-0024/main_ttsp_core.cyttsp4_i2c_adapter # 
cd cyttsp4_mt.main_ttsp_core
root@seagull:/sys/bus/i2c/devices/5-0024/main_ttsp_core.cyttsp4_i2c_adapter/cyttsp4_mt.main_ttsp_core # ls
driver
input
modalias
power
subsystem
uevent
root@seagull:/sys/bus/i2c/devices/5-0024/main_ttsp_core.cyttsp4_i2c_adapter/cyttsp4_mt.main_ttsp_core # 

Tell me how can I be?

kholk, 2016-03-24:
First of all, use the latest 3.10 kernel: the PLATFORM_YUKON config option was available only in 3.4: that driver was buggy as hell, as well as the whole kernel.

First of all, a good disclaimer:
******* IF YOU DECIDE TO DO THIS, YOU ARE DOING THIS AT YOUR OWN RISK. NEITHER ME, SONY OR ANYONE WILL BE RESPONSIBLE FOR YOUR ACTIONS. THIS IS NOT A RECOMMENDED PROCEDURE AND HAS BEEN GUESSED (AND TESTED IN THE PAST) BY ME AND ONLY ME. **********

Then, we may proceed.

Enable the configuration option CONFIG_TOUCHSCREEN_CYPRESS_CYTTSP4_BINARY_FW_UPGRADE in the kernel, recompile and flash.

- ALL THE OPERATIONS FROM NOW ON HAS TO BE DONE FROM A PHONE ROOT SHELL -

Put the firmware for CYTTSP4 (I took the one from Tianchi to recover my Seagull) in /system/etc/firmware. The firmware file has to be named "cyttsp4_loader.main_ttsp_core".
Write "1" to "manual_upgrade", found in cyttsp4 sysfs path.
Write "1" to "upgrade_now", found in the same path.
Check the upgrade by reading "upgrade_step", found in the same path.

If your touchscreen still doesn't work, then it's not a firmware problem: it's either a HW problem or you screwed up your CYTTSP4 TTCONFIG. To upgrade the configuration, you will need to first get a known good conf FOR YOUR SPECIFIC DEVICE, then you will need to enable the CONFIG_TOUCHSCREEN_CYPRESS_CYTTSP4_MANUAL_TTCONFIG_UPGRADE and you will need to:
- Feed the new conf to the sysfs "config_data" (dd if=seagull_config of=/sys/xxxxxxxxx/config_data)
- Start the configuration upgrade by writing "1" to "config_loading" (same sysfs path).

By the way, messing with the touchscreen firmware is stupid!

Good luck!
-- Angelo G. Del Regno

Kota-Kun, 2016-03-25:
Excuse me, but could you explain more how to perform these commands?
> Write "1" to "manual_upgrade", found in cyttsp4 sysfs path.
Write "1" to "upgrade_now", found in the same path.

and

> Feed the new conf to the sysfs "config_data" (dd if=seagull_config of=/sys/xxxxxxxxx/config_data)
Start the configuration upgrade by writing "1" to "config_loading" (same sysfs path).

Kota-Kun, 2016-03-25:
I do not specifically, the firmware was not the one that was necessary(
> By the way, messing with the touchscreen firmware is stupid!

Or could you help me with the firmware assembly?
The server can provide (3 x 3 GHz 4 Gb RAM 200 Gb Disk)

Kota-Kun, 2016-03-25:
Just do not understand this point
> Put the firmware for CYTTSP4 (I took the one from Tianchi to recover my Seagull) in /system/etc/firmware.

Kota-Kun, 2016-03-25:
Just do not understand this point. Where can I get the firmware for CYTTSP4 ??

> Put the firmware for CYTTSP4 (I took the one from Tianchi to recover my Seagull) in /system/etc/firmware.

Or maybe you have a ready-made firmware for the phone itself? Or at least for most touchscreen

Kota-Kun, 2016-03-25:
The firmware file (cyttsp4_fw.bin) for the touchscreen, I found, and took him to the Tianchi.
Recompile the kernel as described here: [How to build and flash a Linux kernel for AOSP supported devices](http://developer.sonymobile.com/knowledge-base/open-source/open-devices/how-to-build-and-flash-a-linux-kernel/how-to-build-and-flash-a-linux-kernel-for-aosp-supported-devices/)

then operate these instructions: [How to build AOSP KitKat for unlocked Xperia devices](http://developer.sonymobile.com/knowledge-base/open-source/open-devices/aosp-build-instructions/how-to-build-aosp-kitkat-for-unlocked-xperia-devices/)

After compiling ask to come to the phone in the terminal as root, and copy cyttsp4_fw.bin in / system / etc / firmware.

Then run:

> Write "1" to "manual_upgrade", found in cyttsp4 sysfs path.
Write "1" to "upgrade_now", found in the same path.
Check the upgrade by reading "upgrade_step", found in the same path.

if it does not work:

> Feed the new conf to the sysfs "config_data" (dd if=seagull_config of=/sys/xxxxxxxxx/config_data)
Start the configuration upgrade by writing "1" to "config_loading" (same sysfs path).

I understood everything correctly? Or there will be some nuances about the version of the kernel?

Kota-Kun, 2016-05-09:
What I did wrong?

Kota-Kun, 2016-05-13:
kernel 3.10 and android 4.4 bootloop((

jerpelea, 2016-08-15:
please flash the latest Sony official software instead of cross flashing the TS firmware

closed 2016-08-15 by jerpelea

Kota-Kun, 2016-08-15:
> please flash the latest Sony official software
latest Sony official software it's android 4.4.4
To flash many times, it did not help