sonyxperiadev/kernel #681: Slightly faster memutils

threader, 2016-04-14:
Implemented this in memutils - https://github.com/rachanta/kernel_n5_racer/commit/f9ee304323c497dbffdced7948cea669b26682ab.patch

It is slightly faster then memutils in LA.BF64.1.2.2_rb4.7 according to benchmarks, but needs some professional input. 

I messed up my initial commit with some rej. orig files and the entire original memutils but tried stitching one together in the zip
https://github.com/threader/kernel-copyleft-14.6.A.1.xxx-backports/commit/d90b9cf611add9544aecc5e39598b36f2c772ded

[memutils-cortex-a15-mike.patch.zip](https://github.com/sonyxperiadev/kernel/files/219713/memutils-cortex-a15-mike.patch.zip)

threader, 2016-04-14:
https://github.com/threader/kernel-copyleft-14.6.A.1.xxx-backports/commit/3cfce01e4d03e1e8a0f38a46112beb854159eb2f &&  +#define HWCAP_VFPD32    (1 << 19)       /* set if VFP has 32 regs (not 16) */ , really speed things up, the above crazyness even more. Tho very slightly. .

threader, 2016-04-15:
[image]
Went over the assembly again today, apparently i've done the right thing first time around and what seemed like logic ended up not working, though i spotted one mistake.

kholk, 2016-04-16:
Interesting find, but usually nVidia makes things for _their_ SoCs.... in any case, an Android benchmark is not enough to say that this has better performance: those will not use ALL of the libraries, but just memcpy and memmove, basically.

You need to test it on a terminal (make small programs to test, it's very easy, like 20 lines per program)!

As a reference, look at the commitlog in the memutils commit, which contains good tests:
https://github.com/sonyxperiadev/kernel/commit/0f492d7aa419daedc09247d31b7b28ea7f4c5c87

threader, 2016-04-16:
The RAM test of antutu is the only one that changed, and it did not change much until that VFP flag was changed. The results doubled. I'll find or write some test case. Would be nice to have exactly the case referenced in the memutils commit.

kholk, 2016-04-16:
Well, just re-read it by yourself: #if __LINUX_ARM_ARCH__ <= 6

MSM8974 is ARMv7, hence __LINUX_ARM_ARCH__ is 7.

threader, 2016-04-16:
Ah, spotted, could it be the changes in vfpmodule then? 
https://github.com/threader/kernel-copyleft-14.6.A.1.xxx-backports/commit/4626c69f918490a4d9c7f27e0d94231a3e4090c2 ? 

 			if (((fmrx(MVFR0) & MVFR0_A_SIMD_MASK)) == 1)
-				elf_hwcap |= HWCAP_VFPv3D16;
+				elf_hwcap |= HWCAP_VFPv3D16; /* also v4-D16 */
+			else
+				elf_hwcap |= HWCAP_VFPD32;

kholk, 2016-04-16:
No, MSM8974 has VFPv3-D16 support, hence that will be selected. No change with that else.

threader, 2016-04-26:
So, is this tricking the benchmark? I would have expected to see an even across the board increase. Instead i see a doubling of mark which bump the total mark...

kholk, 2016-04-26:
Nothing tricks the benchmark. It's just a casualty.
The right way of benchmarking a device is to use a clean system (no apps, preferrably no GAPPS as well) and without any internet connection to make sure that no background process will interfere with the process.

threader, 2016-04-26:
Noted, i have some time this week to figure it out. Cheers

closed 2016-05-15 by jerpelea