sonyxperiadev/kernel #751: Issue with rqbalance

RyTekk, 2016-07-10:
After many times to change between rqbalance and runnable, rqbalance seems not to handle little cores, it leads to reduce point in benchmark, specially in Geekbench.

Device: Xperia Z5 Dual
OS + Kernel: 6.0.1 Stock Sony with my own kernel, CPUQuiet framework ported from LA.BR.1.3.3_rb2.14 branch.

kholk, 2016-07-10:
This can be because of incorrect porting by you on your own kernel, that may have its own problems.
Please post issues only if they are referring to Open Devices (meaning AOSP + associated kernel as per Sony's build guide).

RyTekk, 2016-07-10:
I just post to you and others check it, and if there's the problem, you will fix it, I simply think so. Btw, did you port CPUQuiet from Tegra X1 source? I think it will work better on big.LITTLE devices, such as my Z5.

kholk, 2016-07-10:
It was ported from the latest version found in the Chromium kernel at that time. Probably X1 support was there.

RyTekk, 2016-07-10:
https://github.com/RyTekk/sony_kernel_kitakami_8994/commit/4d7064054e954f35e87b5a9927174fd5606cd903

Could you take a look at my commit? I think I ported successful CPUQuiet.

Is this the latest version from Chromium that you mentioned? https://chromium.googlesource.com/chromiumos/third_party/kernel/+log/chromeos-3.10/drivers/cpuquiet

Latest source from NVIDIA is on Mar 2016. (https://developer.nvidia.com/embedded/linux-tegra-r232)

closed 2016-08-15 by jerpelea