sonyxperiadev/kernel #796: Volume in calls on stable branch is constant

rancidfrog, 2016-08-09:
Changing volume during calls, whether using speaker or in-ear speaker, does not change anything.

However, if we use WhatsApp for calls changing volume affects both speaker and in-ear speaker as it should.

rancidfrog, 2016-08-09:
Not kernel related: https://github.com/sonyxperiadev/kernel/issues/797#issuecomment-238561607

kholk, 2016-08-09:
Userspace configuration issues

rancidfrog, 2017-01-12:
@kholk 
Apparently it is a kernel issue:
https://gerrit.omnirom.org/#/q/status:merged+project:android_kernel_sony_msm+branch:android-6.0+topic:acdb
And: https://forum.xda-developers.com/showpost.php?p=70510280&postcount=74

mcgi5sr2, 2017-01-15:
requires blobs that are not distributed by sony, but instead some unknown benevolent people....

rancidfrog, 2017-01-15:
which means blobs that are provided by sony open dev are not complete and meant for caf?
I believe hammerhead also has msm8974 so there should be blobs available based on aosp code?

mcgi5sr2, 2017-01-17:
hammerhead does have acdb blobs, although i can't find the same kernel commits to use them in hammerhead though most likely because it is using on 3.4. The above mentioned kernel commits are in from the huawei angler kernel which is also 3.10 (The commits can also be found in the latest copy-left kernels which I believe are stock.). Although if you check the vendor blobs for angler there are no acdb libs suggesting another way of audio calibration/routing. My guess would be that is why we have audio, just no volume control. In fact audio quality is better without the acdb blobs.

AFAIK the missing blobs are the only audio and the only ones that are missing for shinano board devices.

rancidfrog, 2017-02-27:
7.1.1 Aries n-mr1
Not much difference with kernel patches and qcom extra/stock acdb files
Volume hardly changes if at all.
There might be some more missing caf commits which acdb blobs rely on!
Any ideas on how to get proper aosp based solution?
@kholk 
Or can we just set up mixer-paths to use same route as VoIP?
As WhatsApp calls have better volume for headset and speaker
Is there a way to force voice calls to use different routing?
(Since no resolution seems likely and for some strange reason aodp acdb blobs do not work properly and seem to rely on some missing caf commits )

mcgi5sr2, 2017-02-28:
reports are that hangouts routing still has in call echo.

rancidfrog, 2017-02-28:
Even after enabling fluence?
https://github.com/sonyxperiadev/device-sony-common/commit/d5123e26f7d85bf6a49fe4bf7fdb9bd50c22e004

So what does that mean, then?
Blobs issue?
Or configurations?
https://github.com/sonyxperiadev/device-sony-common/blob/master/voice_processing/voice_processing_descriptors.c#L33

closed 2017-09-02 by rancidfrog