sonyxperiadev/kernel #797: Phone reboots when using certain apps

rancidfrog, 2016-08-09:
Instagram app,  Maps.me **beta** (https://github.com/mapsme/omim), and sometimes vlc app, cause phone to reboot.

Z3-compact, "stable" 1.2.2 branch

kholk, 2016-08-09:
This should happen because of a vidc/iommu bug reproducible by opening and closing multiple vidc hardware enc/decoder instances while c2d composition (hardware assisted rendering) is enabled. 

After small tests, I couldn't reproduce the issue on the new 1.3.3 branch. 
The new 1.3.3 branch will be considered as main/stable very soon! Please, test the still bleeding edge 1.3.3 branch to confirm that the bug is gone. 

Thanks!

rancidfrog, 2016-08-09:
Okay will test it, 
I reverted back to 1.2.2 because camera blobs were missing on exp blobs.
And was waiting for new blob release as it was mentioned they would be added soon.
I guess now it is very soon.
Will also try to reproduce in call volume not changing.

kholk, 2016-08-09:
The in call volume bug is not kernel related!

rancidfrog, 2016-08-09:
okay

rancidfrog, 2016-08-16:
I am unable to test "bleeding edge 1.3.3 branch" because settings/security FC and I am unable to enable unkown sources to install apps.
```
01-02 15:43:58.611  9558  9558 D SubSettings: Launching fragment com.android.settings.SecuritySettings
01-02 15:43:58.639  9558  9558 V SecuritySettings: No fingerprint hardware detected!!
01-02 15:43:58.649  9558  9558 D AndroidRuntime: Shutting down VM
01-02 15:43:58.653  9558  9558 E AndroidRuntime: FATAL EXCEPTION: main
01-02 15:43:58.653  9558  9558 E AndroidRuntime: Process: com.android.settings, PID: 9558
01-02 15:43:58.653  9558  9558 E AndroidRuntime: Theme: themes:{}
01-02 15:43:58.653  9558  9558 E AndroidRuntime: java.lang.RuntimeException: Unable to resume activity {com.android.settings/com.android.settings.SubSettings}: java.lang.NullPointerException: Attempt to invoke interface method 'int android.security.IKeystoreService.is_hardware_backed(java.lang.String)' on a null object reference
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3131)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3162)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2509)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.app.ActivityThread.-wrap11(ActivityThread.java)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1347)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:102)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:148)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:5458)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to invoke interface method 'int android.security.IKeystoreService.is_hardware_backed(java.lang.String)' on a null object reference
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.security.KeyStore.isHardwareBacked(KeyStore.java:381)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.security.KeyStore.isHardwareBacked(KeyStore.java:376)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at com.android.settings.SecuritySettings.createPreferenceHierarchy(SecuritySettings.java:307)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at com.android.settings.SecuritySettings.onResume(SecuritySettings.java:630)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.app.Fragment.performResume(Fragment.java:2263)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1008)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1148)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1130)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.app.FragmentManagerImpl.dispatchResume(FragmentManager.java:1963)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.app.FragmentController.dispatchResume(FragmentController.java:174)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.app.Activity.performResume(Activity.java:6348)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3120)
01-02 15:43:58.653  9558  9558 E AndroidRuntime: 	... 10 more
01-02 15:43:58.654  1389  3990 W ActivityManager:   Force finishing activity com.android.settings/.SubSettings
01-02 15:43:58.656  1389  3990 W ActivityManager:   Force finishing activity com.android.settings/.Settings
```
Could it be related to `QCOMKeyMaster: Loading keymaster app failed`:
```
01-02 15:44:01.183  9594  9594 I keystore: Found keymaster0 module Keymaster QCOM HAL, version 3
01-02 15:44:01.184  9594  9594 I SoftKeymaster: system/keymaster/soft_keymaster_device.cpp, Line 122: Creating device
01-02 15:44:01.184  9594  9594 D SoftKeymaster: system/keymaster/soft_keymaster_device.cpp, Line 123: Device address: 0xb6b24000
01-02 15:44:01.184  9594  9594 D QSEECOMAPI: : QSEECom_get_handle sb_length = 0x2000
01-02 15:44:01.189  9594  9594 E QSEECOMAPI: : Error::ioctl call query if app is loaded failed with ret = -1, errno = 22
01-02 15:44:01.189  9594  9594 D QSEECOMAPI: : QSEECom_get_handle sb_length = 0x2000
01-02 15:44:01.195  9594  9594 E QSEECOMAPI: : Error::ioctl call query if app is loaded failed with ret = -1, errno = 22
01-02 15:44:01.195  9594  9594 E QCOMKeyMaster: Loading keymaster app failed
01-02 15:44:01.195  9594  9594 E keystore: Error opening keystore keymaster0 device.
01-02 15:44:01.195  9594  9594 E keystore: keystore keymaster could not be initialized; exiting
01-02 15:44:04.025   339  2156 D audio_hw_primary: disable_audio_route: reset and update mixer path: low-latency-playback speaker
[... cut ...]
01-02 15:44:11.308  9603  9603 E QSEECOMAPI: : Error::ioctl call query if app is loaded failed with ret = -1, errno = 22
01-02 15:44:11.308  9603  9603 D QSEECOMAPI: : QSEECom_get_handle sb_length = 0x2000
01-02 15:44:11.313  9603  9603 E QSEECOMAPI: : Error::ioctl call query if app is loaded failed with ret = -1, errno = 22
01-02 15:44:11.313  9603  9603 E QCOMKeyMaster: Loading keymaster app failed
01-02 15:44:11.313  9603  9603 E keystore: Error opening keystore keymaster0 device.
01-02 15:44:11.313  9603  9603 E keystore: keystore keymaster could not be initialized; exiting
```

mcgi5sr2, 2016-09-14:
@kholk could you share some info on what is causing the in call volume issue?

kholk, 2016-09-14:
It looks like android tries to manage the wrong mixer for the in call volume function. That's all I know and it's only my supposition.

rancidfrog, 2016-09-21:
The thing is for calling through whatsapp there is no such volume problem!
Confusing

jerpelea, 2016-09-21:
voip apps use different audio paths

rancidfrog, 2016-10-25:
On 1.3.3 branch video playback is much better,
But got a reboot when using VLC
```
[ 6919.530260] IRQ42 no longer affine to CPU2
[ 6928.014781] msm_iommu_v1: Timed out waiting for TLB SYNC to complete for venus_iommu (client: venus_ns)
[ 6928.023270] ------------[ cut here ]------------
[ 6928.027768] kernel BUG at ../../../../../../kernel/sony/msm/drivers/iommu/msm_iommu-v1.c:312!
[ 6928.036276] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP ARM
[ 6928.042097] CPU: 0 PID: 153 Comm: kworker/u8:3 Not tainted 3.10.84-g929d052 #1
[ 6928.049317] Workqueue: msm_vidc_workerq_venus venus_hfi_core_work_handler
[ 6928.056067] task: f37fd140 ti: c545a000 task.ti: c545a000
[ 6928.061455] PC is at __sync_tlb+0x78/0x90
[ 6928.065450] LR is at __atomic_notifier_call_chain+0x3c/0x48
[ 6928.071002] pc : [<c0708de4>]    lr : [<c0051190>]    psr: 60000093
[ 6928.071002] sp : c545bc30  ip : 00000000  fp : 00000002
[ 6928.082451] r10: d3836dc0  r9 : 60000013  r8 : c0e54568
[ 6928.087666] r7 : f3464810  r6 : c57e9b80  r5 : c5d487f4  r4 : 00000000
[ 6928.094174] r3 : 00000000  r2 : c545bbf8  r1 : c0c07d28  r0 : 00000000
[ 6928.100688] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[ 6928.108068] Control: 10c5787d  Table: 03f0006a  DAC: 00000015
[ 6928.113789] 
[ 6928.113789] PC: 0xc0708d64:
[ 6928.118037] 8d64  c0d05819 c0d05bd6 e92d41f0 e1a01601 e590300c e1a06002 e2812e7f e1a07000
[ 6928.126199] 8d84  e3a00000 e0832002 e5820000 e59f4050 e2815e7f e59f804c e2855004 e0835005
[ 6928.134357] 8da4  e5953000 e3130001 08bd81f0 e59f0038 e5983004 e12fff33 e2544001 1afffff7
[ 6928.142517] 8dc4  e5971030 e5962014 e59f0020 eb0a2873 e3a01001 e5962014 e59f0014 ebe520ed
[ 6928.150674] 8de4  e7f001f2 0007a120 c0e54568 0001a36e c0d0592a c118ae04 e92d40f8 e1a05000
[ 6928.158837] 8e04  e5b5400c e1a07000 e2444008 e2843008 e1530005 0a00001c e5943004 e3530000
[ 6928.166998] 8e24  0a000002 e5930010 e3500000 1a000000 e7f001f2 ebf20d87 e2506000 1a000000
[ 6928.175156] 8e44  e7f001f2 ebfffece e3500000 18bd80f8 e5943000 e596200c e1a03603 e2833e61
[ 6928.183317] 
[ 6928.183317] LR: 0xc0051110:
[ 6928.187568] 1110  e1a00005 e8bd8070 e92d4070 e1a04000 e1a05001 eb2534c3 e1a06000 e1a01005
[ 6928.195726] 1130  e2840004 ebffffc1 e1a05000 e1a01006 e1a00004 eb25350a eb00bcec e1a00005
[ 6928.203890] 1150  e8bd8070 e92d40f7 e1a04003 e1a07000 e1a06001 e1a05002 eb00a357 e59d3020
[ 6928.212047] 1170  e1a01006 e1a02005 e2870004 e58d3000 e1a03004 ebffffbb e1a04000 eb00a356
[ 6928.220207] 1190  e1a00004 e28dd00c e8bd80f0 e3a03000 e92d4007 e58d3000 e3e03000 ebffffe8
[ 6928.228365] 11b0  e28dd00c e49df004 e59f3048 e92d4070 e1a04000 e1a06001 e2805010 e5933000
[ 6928.236528] 11d0  e3530000 1a000002 e1a00005 e8bd4070 eaffff89 eb2531a4 e1a01006 e1a00005
[ 6928.244685] 11f0  ebffff85 e1a05000 e1a00004 eb0071f9 e1a00005 e8bd8070 c0e1a4c4 e92d4038
[ 6928.252850] 
[ 6928.252850] SP: 0xc545bbb0:
[ 6928.257101] bbb0  00000000 00000000 00000000 00000000 c0f6de86 0000005b c099c2a0 c0e10780
[ 6928.265257] bbd0  00000000 c0708de8 00000000 c000e258 00000000 c000dfac 00000000 c0c07d28
[ 6928.273417] bbf0  c545bbf8 00000000 00000000 c5d487f4 c57e9b80 f3464810 c0e54568 60000013
[ 6928.281578] bc10  d3836dc0 00000002 00000000 c545bc30 c0051190 c0708de4 60000093 ffffffff
[ 6928.289737] bc30  f3f04810 c57e9b8c f3464810 c57e9b80 c118ae00 c0708e80 2cc4d05a 00259000
[ 6928.297899] bc50  dcc00000 c57e9b80 c57e9b40 c0708ee4 c0708e98 d383659c 00000002 d3836580
[ 6928.306058] bc70  00000001 d3836c9c d3836c80 c0706394 00000000 c07079f8 d3836584 00000001
[ 6928.314218] bc90  00000002 00000001 00000001 c070817c d3836dc0 00000002 00000001 00000001
[ 6928.322381] 
[ 6928.322381] R1: 0xc0c07ca8:
[ 6928.326634] 7ca8  61662063 000a6c69 6d6f6371 73646d2c 69702d73 632d6570 6f737275 69782d72
[ 6928.334789] 7cc8  64692d6e 6f637100 646d2c6d 702d7373 2d657069 73727563 632d726f 632d6b6c
[ 6928.342950] 7ce8  2d6c7274 7366666f 00737465 73253601 6564203a 61636964 20646574 63207076
[ 6928.351109] 7d08  6f737275 64207372 63657465 2c646574 6d756e20 0a64253d 70646d00 00736e5f
[ 6928.359267] 7d28  5f70646d 75636573 01006572 3a732533 20626320 20726f66 20424c54 434e5953
[ 6928.367428] 7d48  6d697420 74756f65 7544202e 6e69706d 4c582067 7327474f 646d000a 62645f70
[ 6928.375590] 7d68  75625f67 73250073 6572203a 726f7473 20676e69 7373646d 63657320 20657275
[ 6928.383748] 7d88  666e6f63 000a6769 73253401 6373203a 65725f6d 726f7473 65735f65 66635f63
[ 6928.391910] 
[ 6928.391910] R2: 0xc545bb78:
[ 6928.396162] bb78  00003886 00000000 c31e14b4 00000001 00000006 c00bccd8 c0e1b338 c0e0a4b0
[ 6928.404320] bb98  c0e0a4b0 00000001 00000000 00000001 00000006 c0030158 00000000 00000000
[ 6928.412482] bbb8  00000000 00000000 c0f6de86 0000005b c099c2a0 c0e10780 00000000 c0708de8
[ 6928.420642] bbd8  00000000 c000e258 00000000 c000dfac 00000000 c0c07d28 c545bbf8 00000000
[ 6928.428800] bbf8  00000000 c5d487f4 c57e9b80 f3464810 c0e54568 60000013 d3836dc0 00000002
[ 6928.436962] bc18  00000000 c545bc30 c0051190 c0708de4 60000093 ffffffff f3f04810 c57e9b8c
[ 6928.445118] bc38  f3464810 c57e9b80 c118ae00 c0708e80 2cc4d05a 00259000 dcc00000 c57e9b80
[ 6928.453281] bc58  c57e9b40 c0708ee4 c0708e98 d383659c 00000002 d3836580 00000001 d3836c9c
[... 121 lines cut ...]
```
:03.260   459  7361 D VLC     : [8a7947d8/1cc1] core input: Buffering 98%
10-25 07:13:03.261   459  7361 D VLC     : [8a7947d8/1cc1] core input: Stream buffering done (1503 ms in 47 ms)
[... cut ...]
10-25 07:13:06.810   459  7361 E NdkMediaCodec: sf error code: -2147479543
10-25 07:13:06.811   459  7361 E NdkMediaCodec: sf error code: -38
10-25 07:13:06.811   459  7364 W AHierarchicalStateMachine: Warning message AMessage(what = 'shut', target = 10) = {
10-25 07:13:06.811   459  7364 W AHierarchicalStateMachine:   int32_t keepComponentAllocated = 0
10-25 07:13:06.811   459  7364 W AHierarchicalStateMachine: } unhandled in root state.
10-25 07:13:09.812   459  7364 I ACodec  : [OMX.qcom.video.decoder.avc] forcing the release of codec
```

kholk, 2016-10-25:
This looks really like a buffer size mismatch for the encoded chunk of data sent to the Venus block. 
Reiterating in more understandable words, it looks like the system sends too much data to Venus HW, more than it can handle (or improperly aligned).

rancidfrog, 2016-10-25:
[media_profiles.xml](https://github.com/sonyxperiadev/device-sony-shinano/blob/master/rootdir/system/etc/media_profiles.xml#L131) still commented:
```
        <!--
        <EncoderProfile quality="1080p" fileFormat="mp4" duration="30">
            <Video codec="h264"
                   bitRate="17000000"
                   width="1920"
                   height="1080"
                   frameRate="30" />
            <Audio codec="aac"
                   bitRate="96000"
                   sampleRate="48000"
                   channels="1" />
        </EncoderProfile>
        -->
```

rancidfrog, 2016-10-28:
Aries, 1.3.3 branch, phone still reboots during calls,
I testing with [qcom-extras](https://github.com/SuperBenevolent/aosp-vendor-qcom-extras/tree/m-mr1), but audio still not working, speaker issues and volume stuck:
```
[10688.078954] afe_get_cal_topology_id: [AFE_TOPOLOGY_CAL] not initialized for this port 16385
[10688.086268] send_afe_cal_type cal_block not found!!
[10688.137967] afe_get_cal_topology_id: [AFE_TOPOLOGY_CAL] not initialized for this port 4102
[10688.145455] send_afe_cal_type cal_block not found!!
[10688.162298] voice_get_cal: No cal data for cal 7!
[10688.165994] voice_send_cvs_register_cal_cmd: Voice_get_cal failed for cal 7!
[10688.173295] voice_get_cal: No cal data for cal 4!
[10688.177833] voice_send_cvp_register_dev_cfg_cmd: Voice_get_cal failed for cal 4!
[10688.185178] voice_get_cal: No cal data for cal 2!
[10688.189962] voice_send_cvp_register_cal_cmd: Voice_get_cal failed for cal 2!
[10688.196808] voice_get_cal: No cal data for cal 3!
[10688.201613] voice_send_cvp_register_vol_cal_cmd: Voice_get_cal failed for cal 3!
[10688.224291] qdsp_cvp_callback: cmd = 0x112c2 returned error = 0x1
[10688.377566] setEQ: failed 0:2
[10688.379511] setEQ: failed 0:2
[10688.382482] setEQ: failed 0:2
[10688.385428] setEQ: failed 0:2
[10688.388537] setEQ: failed 0:2
[10688.391361] setEQ: failed 0:2
[10688.394313] setEQ: failed 0:2
[10688.397302] setEQ: failed 0:2
[10688.400208] setEQ: failed 0:2
[10688.403214] setEQ: failed 0:2
[10688.443984] qdsp_cvp_callback: cmd = 0x112c2 returned error = 0x1
[10691.704755] qdsp_cvp_callback: cmd = 0x11375 returned error = 0x9
[10691.710160] qdsp_cvp_callback: cmd = 0x11276 returned error = 0x9
[10691.807214] msm-stub-codec qcom,msm-stub-codec.63: set format for codec dai failed
[10691.813946] afe_get_cal_topology_id: [AFE_TOPOLOGY_CAL] not initialized for this port 4103
[10691.822166] send_afe_cal_type cal_block not found!!
[10691.832844] afe_get_cal_topology_id: [AFE_TOPOLOGY_CAL] not initialized for this port 4102
[10691.840193] send_afe_cal_type cal_block not found!!
[10691.938508] setEQ: failed 0:2
[10691.940595] setEQ: failed 0:2
[10691.943420] setEQ: failed 0:2
[10691.946416] setEQ: failed 0:2
[10691.949461] setEQ: failed 0:2
[10691.952321] setEQ: failed 0:2
[10691.955224] setEQ: failed 0:2
[10691.958378] setEQ: failed 0:2
[10691.961140] setEQ: failed 0:2
[10691.964122] setEQ: failed 0:2
[10691.972342] voice_get_cal: No cal data for cal 4!
[10691.976128] voice_send_cvp_register_dev_cfg_cmd: Voice_get_cal failed for cal 4!
[10691.983740] voice_get_cal: No cal data for cal 2!
[10691.988155] voice_send_cvp_register_cal_cmd: Voice_get_cal failed for cal 2!
[10691.995157] voice_get_cal: No cal data for cal 3!
[10691.999947] voice_send_cvp_register_vol_cal_cmd: Voice_get_cal failed for cal 3!
[10692.013115] afe_get_cal_topology_id: [AFE_TOPOLOGY_CAL] not initialized for this port 16385
[10692.020506] send_afe_cal_type cal_block not found!!
[10692.699812] migrate_irqs: 979 callbacks suppressed
[10760.092154] IRQ42 no longer affine to CPU1
[10762.021025] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[10762.027062] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
[10763.313006] Freezing of tasks aborted after 0.001 seconds[10763.313938] Error: returning -512 value

[10763.437927] Error: returning -512 value
[10763.452033] Unable to handle kernel NULL pointer dereference at virtual address 0000001c
[10763.452046] pgd = c4a2c000
[10763.452065] [0000001c] *pgd=00000000
[10763.452087] Internal error: Oops: 5 [#1] PREEMPT SMP ARM
[10763.452116] CPU: 1 PID: 1259 Comm: system_server Tainted: G        W    3.10.84-gd77ff96 #1
[10763.452133] task: c42cddc0 ti: ea9ba000 task.ti: ea9ba000
[10763.452164] PC is at snd_pcm_do_suspend+0x44/0x54
[10763.452181] LR is at snd_pcm_action_single+0x38/0x64
[10763.452202] pc : [<c0730dac>]    lr : [<c0730a0c>]    psr: 60000093
[10763.452202] sp : ea9bbcd0  ip : 000ff749  fp : c0e43f6c
[10763.452215] r10: 00000002  r9 : c0c4a388  r8 : c038db04
[... 47 lines cut ...]
[10763.453637] [<c01031e4>] (SyS_write+0x38/0x68) from [<c000e340>] (ret_fast_syscall+0x0/0x38)
[10763.453666] Code: 1a000004 e59320e0 e1a00003 e3a01005 (e592201c) 
[10763.453686] ---[ end trace e5cc1fca497956d7 ]---
[10763.536236] Kernel panic - not syncing: Fatal exception
[10763.536432] CPU0: stopping
[10763.536459] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G      D W    3.10.84-gd77ff96 #1
[10763.536514] [<c00139c0>] (unwind_backtrace+0x0/0x120) from [<c0011c58>] (show_stack+0x10/0x14)
[10763.536550] [<c0011c58>] (show_stack+0x10/0x14) from [<c0012cec>] (handle_IPI+0xe0/0x1b8)
[10763.536585] [<c0012cec>] (handle_IPI+0xe0/0x1b8) from [<c0008628>] (gic_handle_irq+0x58/0x60)
[10763.536623] [<c0008628>] (gic_handle_irq+0x58/0x60) from [<c000df00>] (__irq_svc+0x40/0x70)
[10763.536638] Exception stack(0xc0e13eb8 to 0xc0e13f00)
[10763.536654] 3ea0:                                                       c0e13f28 00000000
[10763.536679] 3ec0: 8d4fdf3b 00004b1e 00963ca0 00000000 01256e68 00000000 00000003 c0e12000
[10763.536702] 3ee0: 00000001 c31e6f88 00000008 c0e13f00 c00824bc c064705c 00000013 ffffffff
[10763.536754] [<c000df00>] (__irq_svc+0x40/0x70) from [<c064705c>] (lpm_cpuidle_enter+0x678/0x744)
[10763.536796] [<c064705c>] (lpm_cpuidle_enter+0x678/0x744) from [<c06438ac>] (cpuidle_enter_state+0x18/0x60)
[10763.536832] [<c06438ac>] (cpuidle_enter_state+0x18/0x60) from [<c06439d4>] (cpuidle_idle_call+0xe0/0x144)
[... cut ...] _call+0xe0/0x144) from [<c000ef94>] (arch_cpu_idle+0x8/0x38)
[10763.536915] [<c000ef94>] (arch_cpu_idle+0x8/0x38) from [<c007212c>] (cpu_startup_entry+0x1d4/0x268)
[10763.536963] [<c007212c>] (cpu_startup_entry+0x1d4/0x268) from [<c098bdbc>] (rest_init+0x8c/0xa8)
[10763.537011] [<c098bdbc>] (rest_init+0x8c/0xa8) from [<c0db0a14>] (start_kernel+0x310/0x36c)
[10765.423962] Rebooting in 5 seconds..
[10770.429561] Going down for restart now
[10770.433400] Calling SCM to disable SPMI PMIC arbiter
```

rancidfrog, 2016-11-07:
@jerpelea @kholk 
Is the incall reboots and volume stuck missing blobs, i.e. [ qcom-extras](https://github.com/SuperBenevolent/aosp-vendor-qcom-extras/tree/m-mr1)
Or something else; i.e. missing[ mixer_path or audio_config](https://source.codeaurora.org/quic/la/platform/vendor/qcom/copper/tree/mixer_paths_auxpcm.xml?h=LA.BF.1.1)
Any advice on debugging issue?
 
I tried adding qcom-extra, but volume still was stuck

kholk, 2016-11-08:
It looks like your device started the suspend sequence while you were in call. 
This should not happen.... While in call your device should stay awake, so something is not setting the wakelock.

mcgi5sr2, 2016-11-08:
@rancidfrog frog have you tried using GApps for calling, this would tell you if it was an apk issue...

rancidfrog, 2016-11-08:
Do not use gapps, unfortunately.
But, issue with phone rebooting during calls has been present for a while now: https://talk.sonymobile.com/t5/Kernel/Phone-reboots-sometimes-during-calls-Aries-z3-compact/td-p/1107771
Also, iommu still has bugs every now and then, although much better than 1.2.2 branch
latest reboot issue:
```
[43528.353729] msm_iommu_v1: Timed out waiting for TLB SYNC to complete for mdp_iommu (client: mdp_ns) [43528.361738] mdss_iommu_tlb_timeout_notify: cb for TLB SYNC timeout. Dumping XLOG's [43528.369363] ------------[ cut here ]------------ [43528.373901] WARNING: at ../../../../../../kernel/sony/msm/kernel/smp.c:436 smp_call_function_many+0x90/0x298() [43528.384420] ---[ end trace 83c94a5a35ebdd61 ]--- [43528.388802] BUG: scheduling while atomic: mdss_fb0/28483/0x00000002 [43528.395259]   [43528.446615] BUG: scheduling while atomic: mdss_fb0/28483/0x00000002
```

rancidfrog, 2016-12-16:
Aries 7.0 built 15 dec,
After taking picture with camera app, ( no preview, but tested with freedcam and it had preview - so problem with app?).
Going to gallery app and clicking photo taken causes phone to reboot:
```
[  175.647592] ------------[ cut here ]------------
[  175.651179] kernel BUG at ../../../../../../kernel/sony/msm/fs/ext4/inode.c:2347!
[  175.658641] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP ARM
[  175.664460] CPU: 0 PID: 276 Comm: kworker/u8:4 Not tainted 3.10.84-g39eb498 #1
[  175.671670] Workqueue: writeback bdi_writeback_workfn (flush-179:0)
[  175.677910] task: c53076c0 ti: c52f6000 task.ti: c52f6000
[  175.683297] PC is at mpage_prepare_extent_to_map+0x1b0/0x230
[  175.688937] LR is at mpage_prepare_extent_to_map+0x108/0x230
[  175.694577] pc : [<c0165db8>]    lr : [<c0165d10>]    psr: 40000013
[  175.694577] sp : c52f7c50  ip : 00000000  fp : 00000000
[  175.706037] r10: cdf8e43c  r9 : c52f7d20  r8 : da51c164
[  175.711244] r7 : 00000000  r6 : 00000bff  r5 : ffffffff  r4 : 00000000
[  175.717753] r3 : 00000003  r2 : 40400039  r1 : 40400039  r0 : 00000000
[  175.724266] Flags: nZcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[  175.731556] Control: 10c5787d  Table: 1edc006a  DAC: 00000015
[  175.737284] 
[  175.737284] PC: 0xc0165d38:
[  175.741537] 5d38  e5993004 e5933018 e3530000 0a000002 e59a3004 e1530008 0a000002 e1a0000a
[  175.749697] 5d58  ebfd8908 ea00001d e59a3000 e3130a02 0a000002 e1a0000a e3a0100d[  175.756672] mdss_mdp_video_wait4comp: vsync wait timeout 0, fallback to poll mode
 ebfd880d
[  175.765320] 5d78  e59a3000 e3130a02 0a000000 e7f001f2 e5993024 e3530000 059a3008 05893008
[  175.773483] 5d98  e59a3008 e2833001 e589300c e59a1000 e59a3008 e3110b02 e1a03b13 1a000000
[  175.781640] 5db8  e7f001f2 e59a101c e1a00009 e1a02001 ebffff32 e2503000 da00000e e2466001
[  175.789799] 5dd8  e59d3014 e2844001 e1540003 1affffb4 e59d3020 e3530000 0a000001 e28d0020
[  175.797959] 5df8  ebfdb901 eb210b73 eaffff9a e3a00000 ea000008 e3a03000 e59d2020 e3520000
[  175.806120] 5e18  0a000003 e28d0020 e58d300c ebfdb8f6 e59d300c e1a00003 e28dd064 e8bd8ff0
[  175.814280] 
[  175.814280] LR: 0xc0165c90:
[  175.818534] 5c90  e2833001 e28d201c e58d3000 e1a03007 ebfdb969 e3500000 e58d0014 0a000056
[  175.826693] 5cb0  e59d2010 e3a04000 e262b00c e28d2020 e0823104 e593a008 e59a3008 e1530005
[  175.834853] 5cd0  8a00004d e5991004 e5911018 e3510000 1a000001 e3560000 da000047 e5991024
[  175.843012] 5cf0  e3510000 0a000002 e599100c e1510003 1a000041 e3a00000 e1a0100a eb02f626
[  175.851173] 5d10  e3500000 0a000001 e1a0000a ebfd8939 e59a3000 e3130010 0a000009 e59a3000
[  175.859331] 5d30  e3130a02 0a000003 e5993004 e5933018 e3530000 0a000002 e59a3004 e1530008
[  175.867490] 5d50  0a000002 e1a0000a ebfd8908 ea00001d e59a3000 e3130a02 0a000002 e1a0000a
[  175.875651] 5d70  e3a0100d ebfd880d e59a3000 e3130a02 0a000000 e7f001f2 e5993024 e3530000
[  175.883812] 
[  175.883812] SP: 0xc52f7bd0:
[  175.888064] 7bd0  00000000 00000000 00000100 00000008 da542504 c52f7c6c 0000000e da51c164
[  175.896223] 7bf0  00000000 c0165dbc 00000000 c000e258 00000000 c000dfac 00000000 40400039
[  175.904384] 7c10  40400039 00000003 00000000 ffffffff 00000bff 00000000 da51c164 c52f7d20
[  175.912544] 7c30  cdf8e43c 00000000 00000000 c52f7c50 c0165d10 c0165db8 40000013 ffffffff
[  175.920704] 7c50  0000000e c10cefc8 00000008 f2cef6e0 0000000c 00000008 00000050 0000000e
[  175.928864] 7c70  00000008 00000000 cdf8e43c cdf8e418 cdacb38c cdacb368 cdf95984 cdacb2fc
[  175.937024] 7c90  cdacb2d8 cdf9611c 00000002 c016a8c4 00000000 c0191754 00000002 000009ba
[  175.945181] 7cb0  00000002 00000000 f2cef6e0 00000000 00000000 da51c088 c52f7dc8 da51c164
[  175.953344] 
[  175.953344] R8: 0xda51c0e4:
[  175.957595] c0e4  00000000 0000000c 00000000 00000008 00000000 00000006 00000087 00000001
[  175.965755] c104  00000000 da51c108 da51c108 00000000 00000000 ffffcd75 00000000 ce545104
[  175.973914] c124  c5048228 da51f1d4 da51c12c da51c12c da546cc4 da51f1e4 dbbaa190 c017e410
[  175.982074] c144  00000000 00000001 00000000 00000002 00000000 00000001 c09c5880 00000000
[  175.990234] c164  da51c088 00000001 0c000020 da5424c1 00360036 00000000 00000000 da51c180
[  175.998394] c184  da51c180 00000001 00000000 da51c190 da51c190 00000000 00000000 0000001e
[  176.006554] c1a4  00000000 c09c5a18 000200da c50480d0 000f000f da51c1b8 da51c1b8 00000000
[  176.014714] c1c4  00000000 00000000 da51c1cc da51c1cc 00000000 33e97879 00000000 00000000
[  176.022873] 
[  176.022873] R9: 0xc52f7ca0:
[  176.027129] 7ca0  00000000 c0191754 00000002 000009ba 00000002 00000000 f2cef6e0 00000000
[  176.035287] 7cc0  00000000 da51c088 c52f7dc8 da51c164 00000000 c016a788 00000000 c0e33338
[  176.043445] 7ce0  c52f7d84 c09a87d4 000000c8 00000001 00000001 c4e81400 00000c00 00000001
[  176.051606] 7d00  c52f7dc8 c52f7d50 009a820c 91827364 dcf72af8 dcf72af8 c52f7d18 c52f7d18
[  176.059764] 7d20  da51c088 c52f7dc8 00000003 00000004 ffffffff 60000013 00000d60 00000000
[  176.067927] 7d40  00000000 00000000 00000000 00000000 00000001 00000000 c4502a28 f2e12f84
[  176.076086] 7d60  00000d61 00000000 c52f6000 da51c088 c52f7dc8 c52f7e8c da51c164 00000c00
[  176.084244] 7d80  da51c0e0 c5048238 da51c124 c00d24bc 00000087 c012488c c50481c8 da51c088
[... 84 lines cut ...]
[  176.758321] [<c000df00>] (__irq_svc+0x40/0x70) from [<c0727b90>] (binder_ioctl+0x4d0/0x910)
[... cut ...]
[  176.887448] [<c064fe20>] (cpuidle_enter_state+0x18/0x60) from [<c064ff48>] (cpuidle_idle_call+0xe0/0x144)
[  176.896997] [<c064ff48>] (cpuidle_idle_call+0xe0/0x144) from [<c000ef94>] (arch_cpu_idle+0x8/0x38)
[  176.905939] [<c000ef94>] (arch_cpu_idle+0x8/0x38) from [<c007212c>] (cpu_startup_entry+0x1d4/0x268)
[  176.914964] [<c007212c>] (cpu_startup_entry+0x1d4/0x268) from [<0099b264>] (0x99b264)
[  178.127501] Rebooting in 5 seconds..
[  183.131450] Going down for restart now
[  183.134967] Calling SCM to disable SPMI PMIC arbiter
```

rancidfrog, 2016-12-26:
```
[249982.315968] mdss_mdp_pipe_fetch_halt: mdss_mdp_overlay_kickoff+0xec8/0x130c: pipe6 is not idle. xin_id=2
[249982.324744] mdss_mdp_pipe_fetch_halt: mdss_mdp_overlay_kickoff+0xec8/0x130c: pipe7 is not idle. xin_id=10
[249982.351440] mdss_mdp_wait_for_xin_halt: VBIF client 10 not halting. TIMEDOUT.
[249982.357692] mdss_mdp_overlay_cleanup: performing recovery sequence for fb0
[249983.793328] msm_iommu_v1: Timed out waiting for TLB SYNC to complete for mdp_iommu (client: mdp_ns)
[249983.801426] mdss_iommu_tlb_timeout_notify: cb for TLB SYNC timeout. Dumping XLOG's
[249983.809143] ------------[ cut here ]------------
[249983.813761] WARNING: at .../kernel/sony/msm/kernel/smp.c:436 smp_call_function_many+0x90/0x298()
[249983.824387] ---[ end trace 16fb0de21ffa22dd ]---
```

rancidfrog, 2017-02-01:
@kholk I believe the following error happens on first boot after flashing:
```
[ 1079.972266] migrate_irqs: 1526 callbacks suppressed
[ 1079.972273] IRQ32 no longer affine to CPU1
[ 1079.972279] IRQ33 no longer affine to CPU1
[ 1079.972285] IRQ34 no longer affine to CPU1
[ 1079.972290] IRQ35 no longer affine to CPU1
[ 1079.972296] IRQ36 no longer affine to CPU1
[ 1079.972302] IRQ37 no longer affine to CPU1
[ 1079.972307] IRQ38 no longer affine to CPU1
[ 1079.972312] IRQ39 no longer affine to CPU1
[ 1079.972318] IRQ40 no longer affine to CPU1
[ 1079.972323] IRQ41 no longer affine to CPU1
[ 1080.065809] pil-q6v5-mss fc880000.qcom,mss: MBA boot done
[ 1080.071318] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[ 1080.078569] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
[ 1080.087995] ------------[ cut here ]------------
[ 1080.091857] WARNING: at ../../../../../../kernel/sony/msm/drivers/base/firmware_class.c:1246 _request_firmware+0x26c/0x9e4()
[ 1080.105055] mdss_fb_release_all: try to close unopened fb 2! from pid:286 name:healthd
[ 1080.113104] mdss_fb_release_all: try to close unopened fb 1! from pid:286 name:healthd
[ 1080.120195] CPU: 0 PID: 267 Comm: kworker/0:3 Tainted: G        W    3.10.84-g7fce554dd62 #1
[ 1080.130415] Workqueue: ssr_wq subsystem_restart_wq_func
[ 1080.134669] [<c00139c0>] (unwind_backtrace+0x0/0x120) from [<c0011c58>] (show_stack+0x10/0x14)
[ 1080.143979] [<c0011c58>] (show_stack+0x10/0x14) from [<c002e204>] (warn_slowpath_common+0x48/0x68)
[ 1080.153079] [<c002e204>] (warn_slowpath_common+0x48/0x68) from [<c002e41c>] (warn_slowpath_null+0x18/0x20)
[ 1080.162298] mdss_dsi_panel_off: ctrl=f3f46010 ndx=0
[ 1080.166814] [<c002e41c>] (warn_slowpath_null+0x18/0x20) from [<c039db00>] (_request_firmware+0x26c/0x9e4)
[ 1080.177201] [<c039db00>] (_request_firmware+0x26c/0x9e4) from [<c039e370>] (request_firmware_direct+0x7c/0xa4)
[ 1080.187725] [<c039e370>] (request_firmware_direct+0x7c/0xa4) from [<c02ffbdc>] (pil_boot+0x6b4/0x9cc)
[ 1080.196463] [<c02ffbdc>] (pil_boot+0x6b4/0x9cc) from [<c030b648>] (subsystem_powerup+0x5c/0xe8)
[ 1080.205309] [<c030b648>] (subsystem_powerup+0x5c/0xe8) from [<c09a4958>] (for_each_subsys_device.constprop.14+0x2c/0x38)
[ 1080.216568] [<c09a4958>] (for_each_subsys_device.constprop.14+0x2c/0x38) from [<c030b9e4>] (subsystem_restart_wq_func+0x310/0x3cc)
[ 1080.227557] [<c030b9e4>] (subsystem_restart_wq_func+0x310/0x3cc) from [<c00472ac>] (process_one_work+0x290/0x420)
[ 1080.239064] [<c00472ac>] (process_one_work+0x290/0x420) from [<c0047690>] (worker_thread+0x228/0x35c)
[ 1080.247880] [<c0047690>] (worker_thread+0x228/0x35c) from [<c004cb60>] (kthread+0xdc/0xf0)
[ 1080.255785] [<c004cb60>] (kthread+0xdc/0xf0) from [<c000e3e0>] (ret_from_fork+0x14/0x34)
[ 1080.263860] ---[ end trace 5b7f5d1b7a27072e ]---
[ 1080.268209] pil-q6v5-mss fc880000.qcom,mss: firmware: modem.b02 will not be loaded
[ 1080.275915] pil-q6v5-mss fc880000.qcom,mss: modem: Failed to locate blob modem.b02 or blob is too big.
[ 1080.286569] pil-q6v5-mss fc880000.qcom,mss: MBA unlock returned err status: -2
[ 1080.377971] reset disable: pinctrl not enabled
[ 1080.386971] ------------[ cut here ]------------
[ 1080.390594] kernel BUG at ../../../../../../kernel/sony/msm/drivers/video/msm/mdss/mdss_mdp_overlay.c:1254!
[ 1080.400308] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP ARM
[ 1080.406139] CPU: 0 PID: 286 Comm: healthd Tainted: G        W    3.10.84-g7fce554dd62 #1
[ 1080.414204] task: c57ef6c0 ti: c4da6000 task.ti: c4da6000
[ 1080.419610] PC is at mdss_mdp_overlay_off+0x2e8/0x494
[ 1080.424632] LR is at sched_clock_32+0x84/0xa0
[ 1080.428963] pc : [<c028ab94>]    lr : [<c0089adc>]    psr: 800b0013
[ 1080.428963] sp : c4da7d58  ip : 0000000b  fp : f3eb2e28
[ 1080.440414] r10: c57ef6c0  r9 : f3f3545c  r8 : f3f3545c
[ 1080.445626] r7 : f3f35424  r6 : f3f35400  r5 : f3f35400  r4 : f3eb2270
[ 1080.452136] r3 : f3f354a8  r2 : c56a16d0  r1 : f3f35428  r0 : c0f43910
[ 1080.458649] Flags: Nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 1080.465765] Control: 10c5787d  Table: 04dac06a  DAC: 00000015
[ 1080.471490] Process healthd (pid: 286, stack limit = 0xc4da6238)
[ 1080.477480] Stack: (0xc4da7d58 to 0xc4da8000)
[ 1080.481825] 7d40:                                                       c028a8ac f3eb2270
[ 1080.489991] 7d60: 00000000 f3f46038 00000001 f3eb2890 00000000 c57ef6c0 f3eb2e28 c02d1080
[ 1080.498151] 7d80: f3eb2270 00000000 c4da7da8 f3eb2270 c0f49b20 00000004 c4da6028 00000001
[ 1080.506311] 7da0: 0000011e c02d1444 00000004 00000004 20306266 6e616c62 0034206b c4da7db8
[ 1080.514471] 7dc0: c0f49b20 f3f21400 00000001 0000011e ffffff94 f3eb2270 c4d37bc0 00000001
[ 1080.522631] 7de0: 0000011e ffffff94 f3eb2000 c02d39b0 c57efab0 c1156d00 f36a5e34 c000ebf4
[ 1080.530790] 7e00: c4d37bc0 00000000 c4da6030 00000000 00000003 00000000 c1156d00 f35d5844
[ 1080.538950] 7e20: c4da7e4c c005944c 00000000 f3eb2270 00000001 f3eb2a80 c1156d00 c0f84c9c
[ 1080.547111] 7e40: c1156d00 f35d5844 bea8aa60 c02d3b44 f35d581c f35d5810 f3eca410 c03920a4
[ 1080.555267] 7e60: c0392090 c038f000 00000000 c0e5a5e4 01234567 00000000 fee1dead c4da6000
[ 1080.563427] 7e80: 00000000 c0041e6c 01234567 c00420f0 0000002c c011f4f4 c4d43980 c4da7ea8
[ 1080.571586] 7ea0: c017f7d4 c0c00290 0000002c 00000800 00000000 00000009 c0bfc0af 00000000
[ 1080.579746] 7ec0: b1837400 c487632f c4d43980 00000000 00000000 00000000 c4d43980 00020000
[... cut ...]
[ 1080.730111] [<c0041e6c>] (kernel_restart+0xc/0x50) from [<c00420f0>] (SyS_reboot+0xd8/0x1b0)
[ 1080.738534] [<c00420f0>] (SyS_reboot+0xd8/0x1b0) from [<c000e340>] (ret_fast_syscall+0x0/0x38)
[ 1080.747113] Code: e59620a8 e28630a8 e1520003 0a000000 (e7f001f2) 
[ 1080.753183] ---[ end trace 5b7f5d1b7a27072f ]---
[ 1080.769417] Kernel panic - not syncing: Fatal exception
[ 1080.985085] Rebooting in 5 seconds..
[ 1085.990298] Going down for restart now
[ 1085.994546] Calling SCM to disable SPMI PMIC arbiter
```

rancidfrog, 2017-02-28:
Random reboot while browsing web on data:
```
[   11.811141] mdss_dsi_panel_off: ctrl=f35cb010 ndx=0
[   12.003573] reset disable: pinctrl not enabled
[   12.210730] reset enable: pinctrl not enabled
[   12.399658] mdss_dsi_panel_disp_on: ctrl=f35cb010 ndx=0
[   12.410039] mdss_dsi_panel_off: ctrl=f35cb010 ndx=0
[   12.613513] reset disable: pinctrl not enabled
[   12.616785] clearpad clearpad: clearpad_set_power: state=running
[   12.636993] clearpad clearpad: clearpad_vreg_suspend: vdd: set mode (LPM) ok, new mode=4
[   12.636995] clearpad clearpad: suspend mode OK
[   12.817412] reset enable: pinctrl not enabled
[   13.003488] mdss_dsi_panel_disp_on: ctrl=f35cb010 ndx=0
[   13.003529] clearpad clearpad: clearpad_set_power: state=running
[   13.010669] clearpad clearpad: clearpad_vreg_suspend: vdd: set mode (HPM) ok, new mode=2
[   13.040311] mdss_mdp_pipe_destroy: unable to free pipe 3 while still in use
[   13.040318] ------------[ cut here ]------------
[   13.040328] WARNING: at ../../../../../../kernel/sony/msm/drivers/video/msm/mdss/mdss_mdp_pipe.c:745 mdss_mdp_pipe_unmap+0x98/0xb0()
[   13.040329] Unexpected free pipe during unmap
[   13.040333] CPU: 1 PID: 294 Comm: mdss_fb0 Not tainted 3.10.84-g05d7347e42e #1
[   13.040344] [<c00139c0>] (unwind_backtrace+0x0/0x120) from [<c0011c58>] (show_stack+0x10/0x14)
[   13.040352] [<c0011c58>] (show_stack+0x10/0x14) from [<c002e204>] (warn_slowpath_common+0x48/0x68)
[   13.040358] [<c002e204>] (warn_slowpath_common+0x48/0x68) from [<c002e27c>] (warn_slowpath_fmt+0x2c/0x3c)
[   13.040365] [<c002e27c>] (warn_slowpath_fmt+0x2c/0x3c) from [<c026b090>] (mdss_mdp_pipe_unmap+0x98/0xb0)
[   13.040376] [<c026b090>] (mdss_mdp_pipe_unmap+0x98/0xb0) from [<c028a494>] (mdss_mdp_overlay_pan_display+0x1d4/0x2fc)
[   13.040385] [<c028a494>] (mdss_mdp_overlay_pan_display+0x1d4/0x2fc) from [<c02d4500>] (__mdss_fb_display_thread+0x2c0/0x430)
[   13.040393] [<c02d4500>] (__mdss_fb_display_thread+0x2c0/0x430) from [<c004cb60>] (kthread+0xdc/0xf0)
[   13.040401] [<c004cb60>] (kthread+0xdc/0xf0) from [<c000e3e0>] (ret_from_fork+0x14/0x34)
[   13.040404] ---[ end trace 1facb8f83d258d69 ]---
[   13.050079] clearpad clearpad: normal mode OK

[   13.347796] EXT4-fs (mmcblk0p24): recovery complete
[   13.352478] EXT4-fs (mmcblk0p24): mounted filesystem with ordered data mode. Opts: 
[   13.372316] EXT4-fs (mmcblk0p25): recovery complete
[   13.377020] EXT4-fs (mmcblk0p25): mounted filesystem with ordered data mode. Opts: 
[   13.393261] mdss_fb_do_ioctl: unsupported ioctl (5401)
[   13.418066] EXT4-fs (mmcblk0p25): mounted filesystem with ordered data mode. Opts: 
[   13.435605] mdss_fb_do_ioctl: unsupported ioctl (5401)
[   13.459198] EXT4-fs (mmcblk0p25): mounted filesystem with ordered data mode. Opts: 
[   13.475452] mdss_fb_do_ioctl: unsupported ioctl (5401)
[   13.498760] EXT4-fs (mmcblk0p25): warning: maximal mount count reached, running e2fsck is recommended
[   13.507388] EXT4-fs (mmcblk0p25): mounted filesystem with ordered data mode. Opts: 
[   13.524840] mdss_fb_do_ioctl: unsupported ioctl (5401)
[   14.077295] EXT4-fs (mmcblk0p23): mounted filesystem with ordered data mode. Opts: 
[   14.897362] FAT-fs (mmcblk1p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   15.823938] FAT-fs (mmcblk1p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   16.152419] EXT4-fs (mmcblk0p23): mounted filesystem with ordered data mode. Opts: 
[   16.943451] mtp_open
[   24.030099] EXT4-fs (mmcblk0p23): warning: maximal mount count reached, running e2fsck is recommended
[   24.039340] EXT4-fs (mmcblk0p23): mounted filesystem with ordered data mode. Opts: 
[   24.056359] EXT4-fs (mmcblk0p23): warning: maximal mount count reached, running e2fsck is recommended
[   24.065192] EXT4-fs (mmcblk0p23): mounted filesystem with ordered data mode. Opts: 

[   24.224360] SysRq : Emergency Remount R/O
[   24.231504] EXT4-fs (mmcblk0p24): re-mounted. Opts: (null)
[   24.276255] EXT4-fs (mmcblk0p25): re-mounted. Opts: (null)
[   24.302808] Emergency Remount complete
[   24.328272] Disabling non-boot CPUs ...
[   24.331424] IRQ32 no longer affine to CPU1
[   24.331431] IRQ33 no longer affine to CPU1
[   24.331437] IRQ34 no longer affine to CPU1
[   24.331443] IRQ35 no longer affine to CPU1
[   24.331448] IRQ36 no longer affine to CPU1
[   24.331454] IRQ37 no longer affine to CPU1
[   24.342811] mdss_fb_release_all: try to close unopened fb 2! from pid:1 name:init
[   24.349907] mdss_fb_release_all: try to close unopened fb 1! from pid:1 name:init
[   24.439131] mdss_dsi_panel_off: ctrl=f35cb010 ndx=0
[   24.634021] reset disable: pinctrl not enabled
[   24.648095] ------------[ cut here ]------------
[   24.651721] kernel BUG at ../../../../../../kernel/sony/msm/drivers/video/msm/mdss/mdss_mdp_overlay.c:1254!
[... 27 lines cut ...]
[   24.848168] 1ee0: c4cbf548 f33cb8a0 f3da0018 f3da1f80 00000400 c4cbf540 00000400 c01da978
[   24.856325] 1f00: 00000000 00000000 ce4733d0 c0e5ca00 ce4733d0 c0e5ca00 c4cbf540 c007e51c
[   24.864487] 1f20: ce476c60 c00fd860 00000000 c4cbf540 00000000 c09ab9ec c365c6e8 c0114c14
[   24.872647] 1f40: 00002710 c4cbf540 00000000 c09abb44 00002710 c0104330 00000020 00000000
[   24.880804] 1f60: c0e26b60 00000000 00000000 c006da6c c0e26b60 f3d984e4 beb68b58 c006daec
[   24.888965] 1f80: beb68b58 c004a854 00000000 00070307 b6e95304 dead0003 00000000 00000058
[... cut ...]
[   24.990374] [<c0041e6c>] (kernel_restart+0xc/0x50) from [<c0042168>] (SyS_reboot+0x150/0x1b0)
[   24.998889] [<c0042168>] (SyS_reboot+0x150/0x1b0) from [<c000e340>] (ret_fast_syscall+0x0/0x38)
[   25.007546] Code: e59620a8 e28630a8 e1520003 0a000000 (e7f001f2) 
[   25.013614] ---[ end trace 1facb8f83d258d6a ]---
[   25.028471] Kernel panic - not syncing: Fatal exception
[   26.243467] Rebooting in 5 seconds..
[   31.249119] Going down for restart now
[   31.253057] Calling SCM to disable SPMI PMIC arbiter
```

rancidfrog, 2017-04-12:
@kholk @jerpelea 
There is still iommu bug in 1.3.3 branch,
When using wifi phone reboots, happened multiple times with same error:
```
 808.525595] IRQ41 no longer affine to CPU1
[  808.530350] __qpnpint_handle_irq: 292 triggered [0x0, 0x08,0x0] qpnp_kpdpwr_status
[  808.613793] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[  808.613798] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
[  808.743093] NOHZ: local_softirq_pending 202
[  811.697241] afe_get_cal_topology_id: [AFE_TOPOLOGY_CAL] not initialized for this port 4102
[  811.705151] send_afe_cal_type cal_block not found!!
[  814.300351] migrate_irqs: 996 callbacks suppressed
[ 1731.335367] IRQ40 no longer affine to CPU2
[ 1731.335395] IRQ41 no longer affine to CPU2
[ 1731.335424] IRQ42 no longer affine to CPU2
```
**[ 1732.959893] msm_iommu_v1: Timed out waiting for TLB SYNC to complete for mdp_iommu (client: mdp_ns)
[ 1732.967903] mdss_iommu_tlb_timeout_notify: cb for TLB SYNC timeout. Dumping XLOG's
[ 1732.975514] ------------[ cut here ]------------
[ 1732.980053] WARNING: at ../../../../../../kernel/sony/msm/kernel/smp.c:436 0xc008efbc()
[ 1732.988209] ---[ end trace c8435ec68c00ccee ]---**

closed 2017-09-02 by rancidfrog