sonyxperiadev/kernel #809: "bleeding edge 1.3.3 branch" errors

rancidfrog, 2016-08-16:
Aries, build date Tue Aug 16 00:32:58
I have the following errors, 
```
01-02 15:43:47.798  1389  2194 I Simple PowerHAL: Device is awake.
01-02 15:43:47.798  1389  2194 I Simple PowerHAL: Setting normal power mode
01-02 15:43:47.798  1389  2194 E Simple PowerHAL: Error opening /sys/devices/system/cpu/cpuquiet/nr_power_max_cpus: Permission denied
01-02 15:43:47.798  1389  2194 E Simple PowerHAL: Error opening /sys/devices/system/cpu/cpuquiet/nr_min_cpus: Permission denied
```
Flood of following error in log,
```
01-02 15:44:16.369  9607  9607 I keystore: Found keymaster0 module Keymaster QCOM HAL, version 3
01-02 15:44:16.371  9607  9607 I SoftKeymaster: system/keymaster/soft_keymaster_device.cpp, Line 122: Creating device
01-02 15:44:16.372  9607  9607 D SoftKeymaster: system/keymaster/soft_keymaster_device.cpp, Line 123: Device address: 0xb6b24000
01-02 15:44:16.372  9607  9607 D QSEECOMAPI: : QSEECom_get_handle sb_length = 0x2000
01-02 15:44:16.377  9607  9607 E QSEECOMAPI: : Error::ioctl call query if app is loaded failed with ret = -1, errno = 22
01-02 15:44:16.377  9607  9607 D QSEECOMAPI: : QSEECom_get_handle sb_length = 0x2000
01-02 15:44:16.383  9607  9607 E QSEECOMAPI: : Error::ioctl call query if app is loaded failed with ret = -1, errno = 22
01-02 15:44:16.383  9607  9607 E QCOMKeyMaster: Loading keymaster app failed
01-02 15:44:16.383  9607  9607 E keystore: Error opening keystore keymaster0 device.
01-02 15:44:16.383  9607  9607 E keystore: keystore keymaster could not be initialized; exiting
01-02 15:44:21.454  9610  9610 I keystore: Found keymaster0 module Keymaster QCOM HAL, version 3
01-02 15:44:21.454  9610  9610 I SoftKeymaster: system/keymaster/soft_keymaster_device.cpp, Line 122: Creating device
01-02 15:44:21.455  9610  9610 D SoftKeymaster: system/keymaster/soft_keymaster_device.cpp, Line 123: Device address: 0xb6ba4000
01-02 15:44:21.456  9610  9610 D QSEECOMAPI: : QSEECom_get_handle sb_length = 0x2000
01-02 15:44:21.463  9610  9610 E QSEECOMAPI: : Error::ioctl call query if app is loaded failed with ret = -1, errno = 22
01-02 15:44:21.463  9610  9610 D QSEECOMAPI: : QSEECom_get_handle sb_length = 0x2000
01-02 15:44:21.469  9610  9610 E QSEECOMAPI: : Error::ioctl call query if app is loaded failed with ret = -1, errno = 22
01-02 15:44:21.469  9610  9610 E QCOMKeyMaster: Loading keymaster app failed
01-02 15:44:21.469  9610  9610 E keystore: Error opening keystore keymaster0 device.
01-02 15:44:21.470  9610  9610 E keystore: keystore keymaster could not be initialized; exiting
```
```
[16393.535279] init: Starting service 'keystore'...
[16393.595935] QSEECOM: qseecom_ioctl: Invalid IOCTL: 0xc024970f
[16393.600844] QSEECOM: qseecom_ioctl: Invalid IOCTL: 0xc024970f
[16393.607739] init: Service 'keystore' (pid 9594) exited with status 1
[16393.613282] init: Service 'keystore' (pid 9594) killing any children in process group
```
```
[16384.419460] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[16384.426068] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
[16384.520883] android_usb gadget: high-speed config #1: fe8050c8.android_usb
[16384.521012] dwc3 f9200000.dwc3: Avail curr from USB = 500
[16384.521500] android_work: android_work: sent uevent USB_STATE=CONNECTED
```
Also, the following errors still occurs in new branch
```
[16394.160516] migrate_irqs: 983 callbacks suppressed
[16394.160523] IRQ32 no longer affine to CPU1
[16394.160530] IRQ33 no longer affine to CPU1
[16394.160536] IRQ34 no longer affine to CPU1
[16394.160541] IRQ35 no longer affine to CPU1
[16394.160547] IRQ36 no longer affine to CPU1
[16394.160553] IRQ37 no longer affine to CPU1
[16394.160559] IRQ38 no longer affine to CPU1
[16394.160565] IRQ39 no longer affine to CPU1
[16394.160571] IRQ40 no longer affine to CPU1
[16394.160576] IRQ41 no longer affine to CPU1
```
```
[16390.983727] afe_get_cal_topology_id: [AFE_TOPOLOGY_CAL] not initialized for this port 4102
[16390.991059] send_afe_cal_type cal_block not found!!
```
 Also, camera connects but is half-green after latest commits by kholk and alviteri

SuperLamic, 2016-08-16:
Qseecom seems to be WIP by alvitery - https://github.com/SonyAosp/kernel/commits/n-crypto

rancidfrog, 2016-08-31:
Qseecom fixed with latest blobs and kernel commits

rancidfrog, 2016-09-05:
```
[   22.432530] msm_qti_pp_get_rms_value_control, back not active to query rms be_idx:3
[   22.440747] core_get_license_status: cmdrsp_license_result.result = 0x15 for module 0x1
[   22.440840] msm_dolby_dap_param_to_get_control_get, port_id not set, do not query ADM
[   22.447862] APR: Wrong parameters
[   22.450927] adm_get_params_v2: Failed to Get Params on port_id 0x0 -22
[   22.457438] msm_dolby_dap_param_visualizer_control_get: get parameters failed
[   22.465690] msm_pcm_volume_ctl_get substream runtime not found
[   22.470745] msm_pcm_volume_ctl_get substream runtime not found
[   22.476446] msm_pcm_volume_ctl_get substream runtime not found
[   22.482399] msm_compr_audio_effects_config_get: stream or effects inactive
[   22.489572] msm_pcm_volume_ctl_get substream runtime not found
[   22.494914] msm_compr_audio_effects_config_get: stream or effects inactive
[   22.502060] msm_compr_audio_effects_config_get: stream or effects inactive
[   22.509032] msm_pcm_volume_ctl_get substream runtime not found
[   12.314731] taiko-slim taiko-slim-pgd: Platform data from device tree
[   12.320327] taiko-slim taiko-slim-pgd: cdc-vdd-buck: vol=[2150000 2150000]uV, curr=[650000]uA, ond 0
[   12.329328] taiko-slim taiko-slim-pgd: cdc-vdd-tx-h: vol=[1800000 1800000]uV, curr=[25000]uA, ond 0
[   12.338368] taiko-slim taiko-slim-pgd: cdc-vdd-rx-h: vol=[1800000 1800000]uV, curr=[25000]uA, ond 0
[   12.347354] taiko-slim taiko-slim-pgd: cdc-vddpx-1: vol=[1800000 1800000]uV, curr=[10000]uA, ond 0
[   12.356332] taiko-slim taiko-slim-pgd: cdc-vdd-a-1p2v: vol=[1225000 1225000]uV, curr=[10000]uA, ond 0
[   12.365543] taiko-slim taiko-slim-pgd: cdc-vddcx-1: vol=[1225000 1225000]uV, curr=[10000]uA, ond 0
[   12.374463] taiko-slim taiko-slim-pgd: cdc-vddcx-2: vol=[1225000 1225000]uV, curr=[10000]uA, ond 0
[   12.383370] taiko-slim taiko-slim-pgd: Looking up qcom,cdc-micbias4-mv property in node /soc/slim@fe12f000/taiko_codec failed
[   12.394705] taiko-slim taiko-slim-pgd: Looking up qcom,cdc-mad-dmic-rate property in node /soc/slim@fe12f000/taiko_codec failed, err = -22
[   12.407061] taiko-slim taiko-slim-pgd: wcd9xxx_validate_dmic_sample_rate: Invalid mad_dmic_rate = 0, for mclk 9600000
[   12.417744] extcodec_get_pinctrl: Unable to get pinctrl handle
[   12.488033] slimbus:1 laddr:0xcb, EAPC:0x1:0xa0<6>[   12.491760] taiko-slim taiko-slim-pgd: wcd9xxx_slim_device_up: slim device up
[   12.498653] slimbus:1 laddr:0xca, EAPC:0x0:0xa0
[   12.498654] taiko-slim taiko-slim-pgd: wcd9xxx_check_codec_type: detected taiko_codec, major 0x102, minor 0x1, ver 0x2
[   12.531411] msm8974-asoc-taiko fe02c000.sound: msm8974_asoc_machine_probe(): hdmi audio support present
[   12.534914] taiko_codec taiko_codec: taiko_codec_probe()
[   12.575182] msm-pcm-routing qcom,msm-pcm-routing.57: ASoC: no dapm match for VOICEMMODE1_DL --> VoiceMMode1 --> SEC_MI2S_RX_Voice Mixer
[   12.586870] msm-pcm-routing qcom,msm-pcm-routing.57: ASoC: Failed to add route VOICEMMODE1_DL -> VoiceMMode1 -> SEC_MI2S_RX_Voice Mixer
[   12.598670] msm-pcm-routing qcom,msm-pcm-routing.57: ASoC: no dapm match for VOICEMMODE2_DL --> VoiceMMode2 --> SEC_MI2S_RX_Voice Mixer
[   12.610834] msm-pcm-routing qcom,msm-pcm-routing.57: ASoC: Failed to add route VOICEMMODE2_DL -> VoiceMMode2 -> SEC_MI2S_RX_Voice Mixer
[   12.627043] msm-pcm-routing qcom,msm-pcm-routing.57: ASoC: no dapm match for SLIMBUS_0_TX --> SLIM_0_TX --> QUAT_MI2S_RX Port Mixer
[   12.637982] msm-pcm-routing qcom,msm-pcm-routing.57: ASoC: Failed to add route SLIMBUS_0_TX -> SLIM_0_TX -> QUAT_MI2S_RX Port Mixer
[   12.652254] mmc2: Starting deferred resume
[   12.652842] msm-pcm-routing qcom,msm-pcm-routing.57: ASoC: no dapm match for VOICE2_STUB_DL --> Voice2 Stub --> INTERNAL_BT_SCO_RX_Voice Mixer
[   12.652846] msm-pcm-routing qcom,msm-pcm-routing.57: ASoC: Failed to add route VOICE2_STUB_DL -> Voice2 Stub -> INTERNAL_BT_SCO_RX_Voice Mixer
[   12.653625] msm-pcm-routing qcom,msm-pcm-routing.57: ASoC: no sink widget found for SENARY_MI2S_TX
[   12.653629] msm-pcm-routing qcom,msm-pcm-routing.57: ASoC: Failed to add route BE_IN -> direct -> SENARY_MI2S_TX
[   12.654743] msm-pcm-routing qcom,msm-pcm-routing.57: ASoC: failed to add widget PRI_MI2S_RX Port Mixer dapm kcontrol PRI_MI2S_RX Port Mixer QUAT_MI2S_TX: -16
[   12.655711] msm-pcm-routing qcom,msm-pcm-routing.57: ASoC: failed to add widget SLIMBUS_0_RX Port Mixer dapm kcontrol SLIMBUS_0_RX Port Mixer SLIM_0_TX: -16
[   12.683361] msm-pcm-routing qcom,msm-pcm-routing.57: ASoC: mux SLIM_0_RX AANC MUX has no paths
[   12.699913] tfa98xx_codec_probe
[   12.700361] TOP channel open 0
[   12.700673] BOTTOM channel open 1
[   12.702111] msm-dai-q6-dev qcom,msm-dai-q6-sb-2-rx.194: ASoC: no source widget found for SLIMBUS_2_RX
[   12.702115] msm-dai-q6-dev qcom,msm-dai-q6-sb-2-rx.194: ASoC: Failed to add route SLIMBUS_2_RX -> direct -> Slimbus2 Playback
[   12.702139] msm-dai-q6-dev qcom,msm-dai-q6-sb-2-tx.195: ASoC: no sink widget found for SLIMBUS_2_TX
[   12.702142] msm-dai-q6-dev qcom,msm-dai-q6-sb-2-tx.195: ASoC: Failed to add route Slimbus2 Capture -> direct -> SLIMBUS_2_TX
[   12.703671] msm8974-asoc-taiko fe02c000.sound:  snd-soc-dummy-dai <-> MultiMedia1 mapping ok
[   12.704042] msm8974-asoc-taiko fe02c000.sound:  snd-soc-dummy-dai <-> MultiMedia2 mapping ok
[   12.704269] msm8974-asoc-taiko fe02c000.sound:  snd-soc-dummy-dai <-> CS-VOICE mapping ok
[   12.704492] msm8974-asoc-taiko fe02c000.sound:  snd-soc-dummy-dai <-> VoIP mapping ok
[... cut ...]
[   12.983566] input: msm8974-taiko-mtp-snd-card Button Jack as /devices/fe02c000.sound/sound/card0/input6
[   12.983684] input: msm8974-taiko-mtp-snd-card Headset Jack as /devices/fe02c000.sound/sound/card0/input7
[   12.997782] msm8974-asoc-taiko fe02c000.sound: property qcom,us-euro-gpios not detected in node /soc/sound
[   12.998269] platform qcom,smp2pgpio-ssr-smp2p-4-in.28: Driver smp2pgpio requests probe deferral
[   12.998613] platform qcom,smp2pgpio-ssr-smp2p-4-out.29: Driver smp2pgpio requests probe deferral
```

kholk, 2016-09-05:
IRQ not affine is not a fatal message. This is not prejudicial for kernel stability, don't worry. 
About the audio errors, most of those are well known and normal: our kernel covers a lot of devices and different platforms using different SoC and of course different audio mappings. To not create a mess in kernel we decided to not ifdef all the code, also because it's not even degrading performance.
About other audio errors, specifically the license ones, those are also normal as some SoC versions and some platforms haven't got the licenses to run the Dolby SW or other proprietary 3rd party extension.

rancidfrog, 2016-09-06:
Okay.
In aries I still have audio issue where volume does not change for normal calls: https://github.com/sonyxperiadev/device-sony-aries/issues/49
Also, I believe that it is somewhat random, sometimes when you answer call it routes to speaker sometimes to earpiece. In 1.2.2 it also causes frequent reboots during calls, but have not tested long with 1.3.3 yet to verify

mcgi5sr2, 2016-09-14:
Aries 1.3.3 Nougat I get consistent routing to earpiece or speaker. The in call volume is still fixed, and doesn't change.

rancidfrog, 2016-10-25:
On 1.3.3
```
[ 6085.791670] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[ 6085.797713] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
```
It appears >20-40 times

mcgi5sr2, 2016-11-22:
we don't have the widevine library file being referred to in the second set of erros. I would imagine that is what is causing the omx errors

rancidfrog, 2016-12-16:
```
[   16.537701] msm_qti_pp_get_rms_value_control, back not active to query rms be_idx:3
[   16.553899] msm_dolby_dap_param_to_get_control_get, port_id not set, do not query ADM
[   16.560906] APR: Wrong paramfound
[   16.564099] adm_get_params_v2: Failed to Get Params on port_id 0x0 -22
[   16.570549] msm_dolby_dap_param_visualizer_control_get: get parameters failed
[   16.580429] msm_voice_sound_focus_get: Error getting Sound Focus Params, err=-22
[   16.589924] msm_voice_source_tracking_get: Error getting Source Tracking Params, err=-22
[   16.600442] msm_audio_get_copp_idx_from_port_id: Invalid FE, exiting
[   16.607079] msm_audio_sound_focus_get: Could not get copp idx for port_id=16385
[   16.615068] msm_audio_get_copp_idx_from_port_id: Invalid FE, exiting
[   16.622156] msm_audio_source_tracking_get: Could not get copp idx for port_id=16385
[   16.630468] msm_voice_sound_focus_get: Error getting Sound Focus Params, err=-22
[   16.638675] msm_voice_source_tracking_get: Error getting Source Tracking Params, err=-22
[   16.645853] msm_audio_get_copp_idx_from_port_id: Invalid FE, exiting
[   16.653436] msm_audio_sound_focus_get: Could not get copp idx for port_id=4101
[   16.661766] msm_audio_get_copp_idx_from_port_id: Invalid FE, exiting
[   16.668550] msm_audio_source_tracking_get: Could not get copp idx for port_id=4101
[   16.675676] msm_pcm_volume_ctl_get substream runtime not found
[   16.683511] msm_pcm_volume_ctl_get substream runtime not found
[   16.691275] msm_pcm_volume_ctl_get substream runtime not found
[   16.697614] msm_compr_audio_effects_config_get: stream or effects inactive
[   16.704798] msm_pcm_volume_ctl_get substream runtime not found
[   16.710306] msm_compr_audio_effects_config_get: stream or effects inactive
[   16.716897] msm_compr_audio_effects_config_get: stream or effects inactive
[   16.723675] msm_pcm_volume_ctl_get substream runtime not found
```

```
[  154.177125] msm_sensor_config:1446 sensor state 0
[  154.242279] msm_camera_pinctrl_init:1260 Getting pinctrl handle failed
[  154.248662]  msm_csid_init: Failed in getting TOP gdscr regulator handle[  154.255084] msm_camera_power_up:1301 Initialization of pinctrl failed
[  154.272385] MSM-CPP cpp_init_hardware:1066 stream_cnt:0
[  154.293446] msm_cci_pinctrl_init:1111 devm_pinctrl_get cci_pinctrl failed
[  154.299284] msm_cci_init:1258 Initialization of pinctrl failed
[  154.305173]  msm_cci_init: Failed in getting TOP gdscr regulator handle[  154.312210] msm_sensor_config:1416 sensor state 1
[  154.470916] msm_isp_cfg_set_stats_ab: STATS AB value is 0/-1, set to MSM_ISP_MIN_AB
[  154.477903] msm_isp_cfg_set_stats_ab: STATS AB value is 0/-1, set to MSM_ISP_MIN_AB
[  154.549665] msm_isp_notify: PIX0 frame id: 1
[  154.552951] msm_isp_notify: PIX0 frame id: 1
[  154.613662] msm_isp_notify: PIX0 frame id: 2
[  154.617060] msm_isp_notify: PIX0 frame id: 2
[  154.677393] msm_isp_notify: PIX0 frame id: 3
[  154.680701] msm_isp_notify: PIX0 frame id: 3
[  154.741319] msm_isp_notify: PIX0 frame id: 4
[  154.744570] msm_isp_notify: PIX0 frame id: 4
[  154.805279] msm_isp_notify: PIX0 frame id: 5
[  154.808556] msm_isp_notify: PIX0 frame id: 5
[  159.644696] afe_get_cal_topology_id: [AFE_TOPOLOGY_CAL] not initialized for this port 4102
[  159.651961] send_afe_cal_type cal_block not found!!
[  171.126799] msm_sensor_config:1446 sensor state 0
```

rancidfrog, 2016-12-26:
```
<6>[   12.784584] reset enable: pinctrl not enabled
<6>[   12.970819] mdss_dsi_panel_disp_on: ctrl=f36e1010 ndx=0
<6>[   12.970859] clearpad clearpad: clearpad_set_power: state=running
<6>[   12.978042] clearpad clearpad: clearpad_vreg_suspend: vdd: set mode (HPM) ok, new mode=2
<3>[   13.007648] mdss_mdp_pipe_destroy: unable to free pipe 3 while still in use
<4>[   13.007654] ------------[ cut here ]------------
<4>[   13.007664] WARNING: at /sony/msm/drivers/video/msm/mdss/mdss_mdp_pipe.c:745 mdss_mdp_pipe_unmap+0x98/0xb0()
<4>[   13.007665] Unexpected free pipe during unmap
<4>[   13.007669] CPU: 1 PID: 292 Comm: mdss_fb0 Not tainted 3.10.84-g0437da4cbe2 #1
<4>[   13.007681] [<c00139c0>] (unwind_backtrace+0x0/0x120) from [<c0011c58>] (show_stack+0x10/0x14)
<4>[   13.007688] [<c0011c58>] (show_stack+0x10/0x14) from [<c002e204>] (warn_slowpath_common+0x48/0x68)
<4>[   13.007694] [<c002e204>] (warn_slowpath_common+0x48/0x68) from [<c002e27c>] (warn_slowpath_fmt+0x2c/0x3c)
<4>[   13.007700] [<c002e27c>] (warn_slowpath_fmt+0x2c/0x3c) from [<c026a970>] (mdss_mdp_pipe_unmap+0x98/0xb0)
<4>[   13.007711] [<c026a970>] (mdss_mdp_pipe_unmap+0x98/0xb0) from [<c0289d5c>] (mdss_mdp_overlay_pan_display+0x1d4/0x2fc)
<4>[   13.007720] [<c0289d5c>] (mdss_mdp_overlay_pan_display+0x1d4/0x2fc) from [<c02d3dc8>] (__mdss_fb_display_thread+0x2c0/0x430)
<4>[   13.007729] [<c02d3dc8>] (__mdss_fb_display_thread+0x2c0/0x430) from [<c004cb60>] (kthread+0xdc/0xf0)
<4>[   13.007737] [<c004cb60>] (kthread+0xdc/0xf0) from [<c000e3e0>] (ret_from_fork+0x14/0x34)
<4>[   13.007740] ---[ end trace f7feeb62599a087d ]---
```

rancidfrog, 2017-01-16:
```
[ 8106.076343] ------------[ cut here ]------------
[ 8106.076356] WARNING: at ../../../../../../kernel/sony/msm/kernel/irq/manage.c:457 resume_irqs+0x6c/0x84()
[ 8106.076358] Unbalanced enable for IRQ 61
[ 8106.076364] CPU: 0 PID: 1304 Comm: system_server Tainted: G        W    3.10.84-g7fce554dd62 #1
[ 8106.076379] [<c00139c0>] (unwind_backtrace+0x0/0x120) from [<c0011c58>] (show_stack+0x10/0x14)
[ 8106.076389] [<c0011c58>] (show_stack+0x10/0x14) from [<c002e204>] (warn_slowpath_common+0x48/0x68)
[ 8106.076398] [<c002e204>] (warn_slowpath_common+0x48/0x68) from [<c002e27c>] (warn_slowpath_fmt+0x2c/0x3c)
[ 8106.076407] [<c002e27c>] (warn_slowpath_fmt+0x2c/0x3c) from [<c0077da8>] (resume_irqs+0x6c/0x84)
[ 8106.076417] [<c0077da8>] (resume_irqs+0x6c/0x84) from [<c0397300>] (dpm_resume_noirq+0x228/0x258)
[ 8106.076425] [<c0397300>] (dpm_resume_noirq+0x228/0x258) from [<c0397590>] (dpm_resume_start+0xc/0x18)
[ 8106.076433] [<c0397590>] (dpm_resume_start+0xc/0x18) from [<c0070a88>] (suspend_enter+0x24c/0x298)
[ 8106.076441] [<c0070a88>] (suspend_enter+0x24c/0x298) from [<c0070bf8>] (suspend_devices_and_enter+0x124/0x258)
[ 8106.076449] [<c0070bf8>] (suspend_devices_and_enter+0x124/0x258) from [<c0070e9c>] (pm_suspend+0x170/0x2a0)
[ 8106.076456] [<c0070e9c>] (pm_suspend+0x170/0x2a0) from [<c006fc84>] (state_store+0x40/0x68)
[ 8106.076465] [<c006fc84>] (state_store+0x40/0x68) from [<c0225c60>] (kobj_attr_store+0x14/0x20)
[ 8106.076475] [<c0225c60>] (kobj_attr_store+0x14/0x20) from [<c0157f90>] (sysfs_write_file+0x104/0x148)
[ 8106.076487] [<c0157f90>] (sysfs_write_file+0x104/0x148) from [<c0103568>] (vfs_write+0xcc/0x1b4)
[ 8106.076496] [<c0103568>] (vfs_write+0xcc/0x1b4) from [<c01036f0>] (SyS_write+0x38/0x68)
[ 8106.076507] [<c01036f0>] (SyS_write+0x38/0x68) from [<c000e340>] (ret_fast_syscall+0x0/0x38)
[ 8106.076512] ---[ end trace 70699a5f73a8fab7 ]---
[ 8106.077076] PM: early resume of devices complete after 0.527 msecs
[ 8106.151386] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[ 8106.151389] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
[ 8106.151459] PM: resume of devices complete after 74.378 msecs
[ 8106.187335] PM: Finishing wakeup.
[ 8106.187339] Restarting tasks ... done.
[ 8106.193403] Resume caused by IRQ 200, qcom,smd-rpm
[ 8106.193409] Resume caused by IRQ 58, qcom,smsm-modem
[ 8106.193468] PM: suspend exit 2017-01-16 15:56:17.468949106 UTC
[ 8110.072314] PM: suspend entry 2017-01-16 15:56:21.347792698 UTC
[ 8110.072321] PM: Syncing filesystems ... done.
[ 8110.111151] PM: Preparing system for mem sleep
[ 8110.112062] Error: returning -512 value
[ 8110.111612] Freezing user space processes ... (elapsed 0.004 seconds) done.
[ 8110.116335] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[ 8110.118257] PM: Entering mem sleep
[ 8110.118266] Suspending console(s) (use no_console_suspend to debug)
[ 8110.132610] PM: suspend of devices complete after 13.744 msecs
[ 8110.133325] PM: late suspend of devices complete after 0.708 msecs
[ 8110.134525] PM: noirq suspend of devices complete after 1.192 msecs
[ 8110.134533] Disabling non-boot CPUs ...
[ 8110.139579] CPU0:msm_cpu_pm_enter_sleep mode:3 during suspend
[ 8110.139579] Suspended for 0.000 seconds
[ 8110.140347] Enabling non-boot CPUs ...
[ 8110.143829] CPU1 is up
[ 8110.144448] PM: noirq resume of devices complete after 0.466 msecs
[ 8110.145143] ------------[ cut here ]------------
[ 8110.145157] WARNING: at ../../../../../../kernel/sony/msm/kernel/irq/manage.c:457 resume_irqs+0x6c/0x84()
[ 8110.145159] Unbalanced enable for IRQ 61
[ 8110.145166] CPU: 0 PID: 1304 Comm: system_server Tainted: G        W    3.10.84-g7fce554dd62 #1
[ 8110.145181] [<c00139c0>] (unwind_backtrace+0x0/0x120) from [<c0011c58>] (show_stack+0x10/0x14)
[ 8110.145194] [<c0011c58>] (show_stack+0x10/0x14) from [<c002e204>] (warn_slowpath_common+0x48/0x68)
[ 8110.145205] [<c002e204>] (warn_slowpath_common+0x48/0x68) from [<c002e27c>] (warn_slowpath_fmt+0x2c/0x3c)
[ 8110.145217] [<c002e27c>] (warn_slowpath_fmt+0x2c/0x3c) from [<c0077da8>] (resume_irqs+0x6c/0x84)
[ 8110.145230] [<c0077da8>] (resume_irqs+0x6c/0x84) from [<c0397300>] (dpm_resume_noirq+0x228/0x258)
[ 8110.145240] [<c0397300>] (dpm_resume_noirq+0x228/0x258) from [<c0397590>] (dpm_resume_start+0xc/0x18)
[ 8110.145249] [<c0397590>] (dpm_resume_start+0xc/0x18) from [<c0070a88>] (suspend_enter+0x24c/0x298)
[ 8110.145259] [<c0070a88>] (suspend_enter+0x24c/0x298) from [<c0070bf8>] (suspend_devices_and_enter+0x124/0x258)
[ 8110.145268] [<c0070bf8>] (suspend_devices_and_enter+0x124/0x258) from [<c0070e9c>] (pm_suspend+0x170/0x2a0)
[ 8110.145276] [<c0070e9c>] (pm_suspend+0x170/0x2a0) from [<c006fc84>] (state_store+0x40/0x68)
[ 8110.145286] [<c006fc84>] (state_store+0x40/0x68) from [<c0225c60>] (kobj_attr_store+0x14/0x20)
[ 8110.145298] [<c0225c60>] (kobj_attr_store+0x14/0x20) from [<c0157f90>] (sysfs_write_file+0x104/0x148)
[ 8110.145311] [<c0157f90>] (sysfs_write_file+0x104/0x148) from [<c0103568>] (vfs_write+0xcc/0x1b4)
[ 8110.145322] [<c0103568>] (vfs_write+0xcc/0x1b4) from [<c01036f0>] (SyS_write+0x38/0x68)
[ 8110.145334] [<c01036f0>] (SyS_write+0x38/0x68) from [<c000e340>] (ret_fast_syscall+0x0/0x38)
[ 8110.145339] ---[ end trace 70699a5f73a8fab8 ]---
[ 8110.145936] PM: early resume of devices complete after 0.560 msecs
[ 8110.220941] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[... cut ...]
[ 7012.577834] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
[ 7061.083653] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[ 7061.089794] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
[ 7200.251674] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[ 7200.258187] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
[ 7244.231295] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[ 7244.237436] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
```

rancidfrog, 2017-01-19:
When using recovery:
```
<6>[    5.825080] sdhci-pltfm: SDHCI platform and OF driver helper
<4>[    5.831093] sdhci_msm msm_sdcc.1: sdhci_msm_probe: ICE device is not enabled
<3>[    5.837449] ## mmc1: gpio_is_valid(pdata->uim2_gpio)=-2: failure
<6>[    5.846821] sdhci_msm msm_sdcc.1: host_use_default_qos = 1
<6>[    5.851294] sdhci_msm msm_sdcc.1: For qos_policy(default qos) = 0, tbl_sz = 1, qos type = 2
<6>[    5.859598] sdhci_msm msm_sdcc.1: 	dma_latency = 200
<6>[    5.864570] sdhci_msm msm_sdcc.1: For qos_policy(modified dynamic qos) = 1, tbl_sz = 1, qos type = 2
<6>[    5.873684] sdhci_msm msm_sdcc.1: 	dma_latency = 200
<6>[    5.878610] sdhci_msm msm_sdcc.1: For qos_policy(modified dynamic qos) = 2, tbl_sz = 1, qos type = 2
<6>[    5.887746] sdhci_msm msm_sdcc.1: 	dma_latency = 200
<6>[    5.892693] sdhci_msm msm_sdcc.1: Host using default pm_qos
<6>[    5.898831] mmc1: no vqmmc regulator found
<6>[    5.902345] mmc1: no vmmc regulator found
<6>[    5.906454] mmc1: SDHCI controller on msm_sdcc.1 [msm_sdcc.1] using 32-bit ADMA in legacy mode
<4>[    5.943577] sdhci_msm msm_sdcc.2: sdhci_msm_probe: ICE device is not enabled
<3>[    5.949663] ## mmc2: gpio_is_valid(pdata->uim2_gpio)=-2: failure
<6>[    5.961200] sdhci_msm msm_sdcc.2: host_use_default_qos = 1
<6>[    5.965687] sdhci_msm msm_sdcc.2: For qos_policy(default qos) = 0, tbl_sz = 1, qos type = 2
<6>[    5.974002] sdhci_msm msm_sdcc.2: 	dma_latency = 200
<6>[    5.978925] sdhci_msm msm_sdcc.2: For qos_policy(modified dynamic qos) = 1, tbl_sz = 1, qos type = 2
<6>[    5.988061] sdhci_msm msm_sdcc.2: 	dma_latency = 200
<6>[    5.993008] sdhci_msm msm_sdcc.2: For qos_policy(modified dynamic qos) = 2, tbl_sz = 1, qos type = 2
<6>[    6.002130] sdhci_msm msm_sdcc.2: 	dma_latency = 200
<6>[    6.007184] sdhci_msm msm_sdcc.2: Host using default pm_qos
<6>[    6.023918] mmc2: no vqmmc regulator found
<6>[    6.027007] mmc2: no vmmc regulator found
<6>[    6.031200] mmc2: SDHCI controller on msm_sdcc.2 [msm_sdcc.2] using 32-bit ADMA in legacy mode
<6>[    6.052404] mmc1: BKOPS_EN bit = 1
<6>[    6.054762] mmc1: eMMC FW version: 0x0b
<3>[    6.069634] qcom,leds-qpnp 01-qcom,leds-d300: Flash wa regulator get failed(-517)
<3>[    6.076147] qcom,leds-qpnp 01-qcom,leds-d300: Regulator get failed(-517)
<3>[    6.082780] qcom,leds-qpnp 01-qcom,leds-d300: Unable to read flash config data
<3>[    6.083796] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
<3>[    6.083799] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
<3>[   11.531872] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
<3>[   11.537874] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
```
I still seem to get a lot of `Failed to enable auto-bkops: err: -74` in logs.
And,
```
<4>[   12.912667] ------------[ cut here ]------------
<4>[   12.912677] WARNING: at ../../../../../../kernel/sony/msm/drivers/video/msm/mdss/mdss_mdp_pipe.c:745 mdss_mdp_pipe_unmap+0x98/0xb0()
<4>[   12.912679] Unexpected free pipe during unmap
<4>[   12.912682] CPU: 3 PID: 296 Comm: mdss_fb0 Not tainted 3.10.84-gb31bea27bfb #1
<4>[   12.912693] [<c00139c0>] (unwind_backtrace+0x0/0x120) from [<c0011c58>] (show_stack+0x10/0x14)
<4>[   12.912702] [<c0011c58>] (show_stack+0x10/0x14) from [<c002e204>] (warn_slowpath_common+0x48/0x68)
<4>[   12.912708] [<c002e204>] (warn_slowpath_common+0x48/0x68) from [<c002e27c>] (warn_slowpath_fmt+0x2c/0x3c)
<4>[   12.912715] [<c002e27c>] (warn_slowpath_fmt+0x2c/0x3c) from [<c026b090>] (mdss_mdp_pipe_unmap+0x98/0xb0)
<4>[   12.912726] [<c026b090>] (mdss_mdp_pipe_unmap+0x98/0xb0) from [<c028a494>] (mdss_mdp_overlay_pan_display+0x1d4/0x2fc)
<4>[   12.912736] [<c028a494>] (mdss_mdp_overlay_pan_display+0x1d4/0x2fc) from [<c02d4500>] (__mdss_fb_display_thread+0x2c0/0x430)
<4>[   12.912745] [<c02d4500>] (__mdss_fb_display_thread+0x2c0/0x430) from [<c004cb60>] (kthread+0xdc/0xf0)
<4>[   12.912753] [<c004cb60>] (kthread+0xdc/0xf0) from [<c000e3e0>] (ret_from_fork+0x14/0x34)
<4>[   12.912756] ---[ end trace 57799c43e8326730 ]---
```

rancidfrog, 2017-02-27:
Andoird 7.1.1 compiled n-mr1 branch
In Recovery:
```
[   10.503512] ------------[ cut here ]------------
[   10.503522] WARNING: at ../../../../../../kernel/sony/msm/drivers/video/msm/mdss/mdss_mdp_pipe.c:745 mdss_mdp_pipe_unmap+0x98/0xb0()
[   10.503523] Unexpected free pipe during unmap
[   10.503527] CPU: 2 PID: 294 Comm: mdss_fb0 Not tainted 3.10.84-g1f01bc82cd1 #1
[   10.503538] [<c00139c0>] (unwind_backtrace+0x0/0x120) from [<c0011c58>] (show_stack+0x10/0x14)
[   10.503546] [<c0011c58>] (show_stack+0x10/0x14) from [<c002e204>] (warn_slowpath_common+0x48/0x68)
[   10.503552] [<c002e204>] (warn_slowpath_common+0x48/0x68) from [<c002e27c>] (warn_slowpath_fmt+0x2c/0x3c)
[   10.503559] [<c002e27c>] (warn_slowpath_fmt+0x2c/0x3c) from [<c026b090>] (mdss_mdp_pipe_unmap+0x98/0xb0)
[   10.503569] [<c026b090>] (mdss_mdp_pipe_unmap+0x98/0xb0) from [<c028a494>] (mdss_mdp_overlay_pan_display+0x1d4/0x2fc)
[   10.503579] [<c028a494>] (mdss_mdp_overlay_pan_display+0x1d4/0x2fc) from [<c02d4500>] (__mdss_fb_display_thread+0x2c0/0x430)
[   10.503587] [<c02d4500>] (__mdss_fb_display_thread+0x2c0/0x430) from [<c004cb60>] (kthread+0xdc/0xf0)
[   10.503595] [<c004cb60>] (kthread+0xdc/0xf0) from [<c000e3e0>] (ret_from_fork+0x14/0x34)
[   10.503598] ---[ end trace 2f5743a9c86f1c89 ]---
[   25.131273] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[   25.137355] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
[  199.501392] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[  199.507474] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
[  203.102642] EXT4-fs (mmcblk0p23): warning: maximal mount count reached, running e2fsck is recommended
[  203.111709] EXT4-fs (mmcblk0p23): mounted filesystem with ordered data mode. Opts: 
[  203.118539] SELinux: initialized (dev mmcblk0p23, type ext4), uses xattr
[  203.129851] mdss_fb_do_ioctl: unsupported ioctl (5401)
[  203.269547] mdss_fb_do_ioctl: unsupported ioctl (5401)
[  203.302676] mdss_fb_do_ioctl: unsupported ioctl (5401)
[  203.338787] mdss_fb_do_ioctl: unsupported ioctl (5401)
[  234.847380] EXT4-fs (mmcblk0p23): mounted filesystem with ordered data mode. Opts: 
[  234.854015] SELinux: initialized (dev mmcblk0p23, type ext4), uses xattr
[  234.865354] mdss_fb_do_ioctl: unsupported ioctl (5401)
[  234.972426] mdss_fb_do_ioctl: unsupported ioctl (5401)
[  235.009886] mdss_fb_do_ioctl: unsupported ioctl (5401)
[  235.044221] mdss_fb_do_ioctl: unsupported ioctl (5401)
[  237.022659] mdss_fb_do_ioctl: unsupported ioctl (5401)
[  237.594559] EXT4-fs (mmcblk0p23): mounted filesystem with ordered data mode. Opts: 
[  237.601397] SELinux: initialized (dev mmcblk0p23, type ext4), uses xattr
[  335.702069] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[  335.708105] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
[  360.090024] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[  360.096052] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
[  360.228073] mdss_fb_do_ioctl: unsupported ioctl (5401)
```

rancidfrog, 2017-03-12:
```
[103297.729698] IRQ40 no longer affine to CPU3
[103297.729704] IRQ41 no longer affine to CPU3
[103297.729709] IRQ42 no longer affine to CPU3
[103299.826836] SysRq : Emergency Remount R/O
[103299.869198] loop: Write error at byte offset 1093632, length 4096.
[103299.874511] JBD2: Error -5 detected when updating journal superblock for loop0-8.
[103299.883474] loop: Write error at byte offset 1093632, length 4096.
[103299.888830] JBD2: Error -5 detected when updating journal superblock for loop0-8.
[103299.896327] loop: Write error at byte offset 0, length 4096.
[103300.820875] msm-dwc3 msm_dwc3: failed to get vbus_rec
[103301.031206] Restarting system with command ''.
[103301.034746] Going down for restart now
[103301.040330] Calling SCM to disable SPMI PMIC arbiter
```

rancidfrog, 2017-04-12:
```
790.517837] IRQ42 no longer affine to CPU1
[  790.539174] ------------[ cut here ]------------
[  790.539192] WARNING: at ../../../../../../kernel/sony/msm/kernel/irq/manage.c:457 0xc0078328()
[  790.539202] Unbalanced enable for IRQ 61
[  790.539454] ---[ end trace c8435ec68c00cce9 ]---
[  790.623668] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[  790.623682] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
[  793.573227] Error: returning -512 value
[  793.612378] ------------[ cut here ]------------
[  793.612396] WARNING: at ../../../../../../kernel/sony/msm/kernel/irq/manage.c:457 0xc0078328()
[  793.612406] Unbalanced enable for IRQ 61
[  793.612655] ---[ end trace c8435ec68c00ccea ]---
[  793.696582] mmc1: mmc_set_auto_bkops: error in setting auto bkops to 1 (-74)
[  793.696596] mmc1: mmc_init_card: Failed to enable auto-bkops: err: -74
[  795.644252] Error: returning -512 value
[  795.666473] migrate_irqs: 494 callbacks suppressed
[  795.666478] IRQ32 no longer affine to CPU1
```

closed 2017-09-02 by rancidfrog