sonyxperiadev/kernel #856: Screen flickering on Z3 D6603

Swidilator, 2016-09-06:
Every few moments, the screen will flicker/glitch while using this kernel on the Z3 D6603

kholk, 2016-09-06:
Are you using AOSP built as per Sony guides?

aclegg2011, 2016-09-06:
The issue happens on AOSP and CM roms using the 3.10 kernel. Using the 3.4 kernel on CM there's no screen flicker on the Z3 D6603.

alviteri, 2016-09-14:
@kholk can be panel bug JDI/AUO

@Swidilator please do and post here
adb reboot
adb shell dmesg | grep panel

nailyk-fr, 2016-09-14:
I'm porting cm-13.0 on a broken z3 and will receive some new panels in a few time. Could tests with those new panels be relevant? 
Also, iirc when enabling debugging in the adreno 300 driver and the HAL part, it seems the flickering happens when logcat said 'VSYNC ENABLED' (but not each time this message appears). 
Hope it helps.

diegosolor, 2016-09-14:
Same problem here on z3c (aries)

 adb shell dmesg | grep panel:

>     0.522581] mdss_dsi_config_panel:2493:dsi specific cfg not present
[    0.522871]  mdss_dsi_panel: mdss_dsi_panel_init: spec_pdata->vsn_gpio=-2 Invalid
[    0.522881]  mdss_dsi_panel: mdss_dsi_panel_init: spec_pdata->vsp_gpio=-2 Invalid
[    0.522994] panel_detect_setup: DriverIC GPIO: 1
[    0.523005] do_panel_detect: Starting ADC Panel Detection...
[    0.523014] adc_panel_detect: Found panel ADC: 376026
[    0.523061] mdss_panel_parse_dt: Panel Name = sharp novatek 720p vid
[    0.523110] mdss_dsi_panel_timing_from_dt: found new timing "somc,novatek_sharp_720p_video_panel" (f3da18f4)
[    0.523165] mdss_dsi_panel_timing_switch: ndx=0 switching to panel timing "somc,novatek_sharp_720p_video_panel"
[    0.523285] mdss_panel_parse_dt:3840, Unable to read SoMC Phy regulator 					settings
[    0.523417] mdss_dsi_parse_panel_features: ulps feature disabled
[    0.523426] mdss_dsi_parse_panel_features: ulps during suspend feature disabled
[    0.523446] mdss_dsi_parse_panel_features:2965, Disp_en gpio not specified
[    0.523494] mdss_dsi_parse_dcs_cmds: Cannot find property:  qcom,mdss-dsi-post-panel-on-command
[    0.523508] mdss_dsi_panel_init:4283 Continuous splash flag not found.
[    0.523522] mdss_panel_parse_bl_settings: WLED bkl-trigger registered!
[    0.524738] dsi_panel_device_register: Continuous splash disabled
[    0.525055] mdss_register_panel: adding framebuffer device fd922800.qcom,mdss_dsi_ctrl0
[    0.525589] mdss_dsi_config_panel:2493:dsi specific cfg not present
[    0.525596] mdss_dsi_pref_prim_panel:can't find panel phandle
[    0.525602] mdss_dsi_config_panel: can't find panel node 
[    0.525608] mdss_dsi_ctrl_probe: panel configuration failed
[    0.527017] mdss_register_panel: adding framebuffer device fd922100.qcom,hdmi_tx
[    0.529629] mdss_register_panel: adding framebuffer device qcom,mdss_wb_panel.8
[    0.541318] mdss_dsi_ctrl fd922800.qcom,mdss_dsi_ctrl0: mipi_dsi_panel_create_debugfs: create folder fd922800.qcom,mdss_dsi_ctrl0
[    0.736513] mdss_dsi_panel_pcc_setup: (1975):(ru[0], ru[1])=(0, 4), (rv[0], rv[1])=(1, 2)
[    1.203850] mdss_dsi_panel_pcc_setup: (1984):ct=0 area=112 ud=34 vd=28 r=0x00007A00 g=0x00008000 b=0x00007F00
[    1.203851] mdss_dsi_panel_disp_on: ctrl=f34db010 ndx=0
[    4.962971] clearpad clearpad: no wakeup_gesture_large_panel

Please tell me if you need anything else

I also attach the complete dmesg 
[dmesg_z3c.txt](https://github.com/sonyxperiadev/kernel/files/472085/dmesg_z3c.txt)

kholk, 2016-09-14:
Thanks for the feedback. I currently have zero problems with both Z3 and Z3 Compact, though, I have different panel(s).

I will see what can I do for you, hoping that I get some more people with the same panels.

nailyk-fr, 2016-09-14:
d6603 on cm-13.0 builds:  
kernel [3.4](https://github.com/nailyk-fr/android_kernel_sony_msm8974/tree/cm-13.0) : [dmesg | panel ](https://github.com/sonyxperiadev/kernel/files/472683/dmesg_3.4.panel.txt) / [full dmesg](https://github.com/sonyxperiadev/kernel/files/472685/dmesg_3.4.txt) (display is perfect on this kernel)
kernel [3.10](https://github.com/sonyxperiadev/kernel/tree/aosp/LA.BF64.1.2.2_rb4.7) : [dmesg | panel](https://github.com/sonyxperiadev/kernel/files/472684/dmesg_3.10.panel.txt) / [full dmesg](https://github.com/sonyxperiadev/kernel/files/472686/dmesg_3.10.txt)

Tell me if a can do something else (enable debug flags, try patches, etc...)

nailyk-fr, 2016-09-17:
Confirmed it is a panel problem: 
With this "default_panel_1"no screen flickering (same motherboard, same cm-13.0 zip flash). 

> [    0.610034] adc_panel_detect: Found panel ADC: 1826916
> [    0.610115] mdss_panel_parse_dt: Panel Name = default video 1
> [    0.610159] mdss_dsi_panel_timing_from_dt: found new timing "somc,default_panel_1" (f3d97b2c)
> [    0.610208] mdss_dsi_panel_timing_switch: ndx=0 switching to panel timing "somc,default_panel_1"

aclegg2011, 2016-09-18:
C:\Users\Aren Clegg\Desktop\Minimal ADB and Fastboot>adb shell
leo:/ $ dmesg | grep panel
[ 1131.996376] mdss_dsi_panel_on: ctrl=f2f35810 ndx=0
[ 1145.383529] mdss_dsi_panel_off: ctrl=f2f35810 ndx=0
[49072.042461] mdss_dsi_panel_on: ctrl=f2f35810 ndx=0
[49107.469884] mdss_dsi_panel_off: ctrl=f2f35810 ndx=0
[50626.769201] mdss_dsi_panel_on: ctrl=f2f35810 ndx=0
[50660.226226] mdss_dsi_panel_off: ctrl=f2f35810 ndx=0

this is the dump of Android N preview 3 (uses 3.4 kernel no screen glicthes) dmesg | grep panel

alviteri, 2016-09-18:
@aclegg2011 this don't say you panel name...please 

adb reboot
adb shell dmesg | grep panel

aclegg2011, 2016-09-18:
@alviteri panel info has been updated in previous post

alviteri, 2016-09-18:
@aclegg2011 yeah auo/sharp have those glitches .... unfortunatelly but even my leo has JDI also seems main devs have JDI too....

aclegg2011, 2016-09-18:
@alviteri so the only way to fix this is, is by using a 3.4x kernel? if I wanted to fix this myself locally, should

alviteri, 2016-09-18:
@aclegg2011 I mean you having that panel .... you can play and fix it eventually ..... becuase main devs having not the hardware to test ... is little complicated

opticron, 2016-10-31:
Oddly enough, I have a Z3 Compact and don't experience this issue. Someone suggested that I post my dmesg|grep panel here (3.10.84-g083cfc0 kernel, original stock panel):
[    0.512257] mdss_dsi_config_panel:2493:dsi specific cfg not present
[    0.512542]  mdss_dsi_panel: mdss_dsi_panel_init: spec_pdata->vsn_gpio=-2 Invalid
[    0.512551]  mdss_dsi_panel: mdss_dsi_panel_init: spec_pdata->vsp_gpio=-2 Invalid
[    0.512663] panel_detect_setup: DriverIC GPIO: 1
[    0.512674] do_panel_detect: Starting ADC Panel Detection...
[    0.512683] adc_panel_detect: Found panel ADC: 1152774
[    0.512723] mdss_panel_parse_dt: Panel Name = jdi novatek 720p vid
[    0.512769] mdss_dsi_panel_timing_from_dt: found new timing "somc,novatek_jdi_720p_video_panel" (f3da18f4)
[    0.512818] mdss_dsi_panel_timing_switch: ndx=0 switching to panel timing "somc,novatek_jdi_720p_video_panel"
[    0.512941] mdss_panel_parse_dt:3836, Unable to read SoMC Phy regulator 					settings
[    0.513071] mdss_dsi_parse_panel_features: ulps feature disabled
[    0.513079] mdss_dsi_parse_panel_features: ulps during suspend feature disabled
[    0.513100] mdss_dsi_parse_panel_features:2961, Disp_en gpio not specified
[    0.513149] mdss_dsi_parse_dcs_cmds: Cannot find property:  qcom,mdss-dsi-post-panel-on-command
[    0.513162] mdss_dsi_panel_init:4279 Continuous splash flag not found.
[    0.513176] mdss_panel_parse_bl_settings: WLED bkl-trigger registered!
[    0.514378] dsi_panel_device_register: Continuous splash disabled
[    0.514698] mdss_register_panel: adding framebuffer device fd922800.qcom,mdss_dsi_ctrl0
[    0.515241] mdss_dsi_config_panel:2493:dsi specific cfg not present
[    0.515248] mdss_dsi_pref_prim_panel:can't find panel phandle
[    0.515254] mdss_dsi_config_panel: can't find panel node 
[    0.515260] mdss_dsi_ctrl_probe: panel configuration failed
[    0.516683] mdss_register_panel: adding framebuffer device fd922100.qcom,hdmi_tx
[    0.519162] mdss_register_panel: adding framebuffer device qcom,mdss_wb_panel.8
[    0.530977] mdss_dsi_ctrl fd922800.qcom,mdss_dsi_ctrl0: mipi_dsi_panel_create_debugfs: create folder fd922800.qcom,mdss_dsi_ctrl0
[    0.725347] mdss_dsi_panel_pcc_setup: (1980):ct=255 area=0 ud=30 vd=30 r=0x00008000 g=0x00008000 b=0x00008000
[    0.725348] mdss_dsi_panel_disp_on: ctrl=f357a010 ndx=0
[    4.802808] clearpad clearpad: no wakeup_gesture_large_panel
[   32.445926] mdss_dsi_panel_off: ctrl=f357a010 ndx=0

nailyk-fr, 2016-11-28:
Should be fixed by this [commit](https://github.com/sonyxperiadev/kernel/pull/1005). 
Please report back if some screen are still flickering. 

Humberos [PR](https://github.com/sonyxperiadev/kernel/pull/992) is reported as solving black display for 2 different users with replacement panels. 

I think this issue could be closed when PR got merged. Thanks.

Swidilator, 2016-11-29:
I see this is now fixed, so no need for this issue anymore :P

closed 2016-11-29 by Swidilator