sonyxperiadev/kernel #882: KCAL support?

leeoniya, 2016-09-17:
Hey guys,

Would you consider adding support for KCAL [1]? I'm finding that the screen on my Z5C (suzuran) is way oversaturated and there are no apps to adjust vibrance/intensity/saturation, only contrast, temperature and RGB balance (which already works fine in CM13's display settings). This feature would be really great.

thanks!

[1] http://forum.xda-developers.com/android/software-hacking/dev-kcal-advanced-color-control-t3032080

scafroglia93, 2017-01-08:
Are you still interested ?

leeoniya, 2017-01-08:
The CM13/AOSP kernel had too many issues on the Z5C (camera issues, call volume/echo). I had to resort to the XPower v5.0 modified stock rom with a sunkernel [1] which seems to have the KCAL patches.

[image]

[1] https://forum.xda-developers.com/z5-compact/development/kernel-zachs-kernel-playground-z5c-wip-t3350690/page20

leeoniya, 2017-01-08:
that being said...if CM (i guess LineageOS now) or another lightweight AOSP rom presents itself without all the driver bugs, i would be interested in switching. so yes, it would be nice to have these patches especially if it's not a lot of work.

scafroglia93, 2017-01-08:
This is my personal branch based on 1.3.3 with Kcal V2 fully functional on my Z3 Aosp

https://github.com/scafroglia93/kernel/tree/kcal

leeoniya, 2017-01-08:
i'm not a android dev, so unless these patches make it into some actively maintained and flashable Z5C kernel [or downstreamed as part of a Z5C rom] they won't affect me.

i guess the first step would be to submit a PR into this repo :/

scafroglia93, 2017-01-08:
Yep but it's really hard to merge this into master repo

leeoniya, 2017-01-08:
¯\_(ツ)_/¯

bartcubbins, 2017-11-11:
unfortunately I think we will never get KCAL support on our 1.3.3 AOSP kernel, sorry.
@jerpelea can it be closed?

closed 2017-11-11 by jerpelea