sonyxperiadev/kernel #893: Kernel can't be compiled

fergy, 2016-09-25:
**LA.BR.1.3.3_rb2.14**
**aosp config for aosp_kitakami_sumire**

`drivers/built-in.o: In function `set_device_mode':
/home/fergy/Development/nougat/kernel/sony/kitakami/drivers/cpuidle/lpm-levels.c:329: undefined reference to `use_psci'
/home/fergy/Development/nougat/kernel/sony/kitakami/drivers/cpuidle/lpm-levels.c:329: undefined reference to `use_psci'
/home/fergy/Development/nougat/kernel/sony/kitakami/drivers/cpuidle/lpm-levels.c:329: undefined reference to `use_psci'
/home/fergy/Development/nougat/kernel/sony/kitakami/drivers/cpuidle/lpm-levels.c:329: undefined reference to `use_psci'
/home/fergy/Development/nougat/kernel/sony/kitakami/drivers/cpuidle/lpm-levels.c:329: undefined reference to `use_psci'
drivers/built-in.o:/home/fergy/Development/nougat/kernel/sony/kitakami/drivers/cpuidle/lpm-levels.c:329: more undefined references to `use_psci' follow
drivers/built-in.o: In function `parse_cluster_params':
/home/fergy/Development/nougat/kernel/sony/kitakami/drivers/cpuidle/lpm-levels-of.c:278: undefined reference to `set_cci_mode'
/home/fergy/Development/nougat/kernel/sony/kitakami/drivers/cpuidle/lpm-levels-of.c:278: undefined reference to `set_cci_mode'
drivers/built-in.o: In function `ipa_q6_monitor_holb_mitigation':
/home/fergy/Development/nougat/kernel/sony/kitakami/drivers/platform/msm/ipa/ipa.c:1378: undefined reference to `ipa_uc_monitor_holb'
drivers/built-in.o: In function `ipa_controller_static_bind':
/home/fergy/Development/nougat/kernel/sony/kitakami/drivers/platform/msm/ipa/ipa_utils.c:4415: undefined reference to `__ipa_commit_hdr_v2_6L'
/home/fergy/Development/nougat/kernel/sony/kitakami/drivers/platform/msm/ipa/ipa_utils.c:4415: undefined reference to `__ipa_commit_hdr_v2_6L'
/home/fergy/Development/nougat/kernel/sony/kitakami/drivers/platform/msm/ipa/ipa_utils.c:4416: undefined reference to `__ipa_generate_rt_hw_rule_v2_6L'
/home/fergy/Development/nougat/kernel/sony/kitakami/drivers/platform/msm/ipa/ipa_utils.c:4416: undefined reference to `__ipa_generate_rt_hw_rule_v2_6L'
drivers/built-in.o: In function `sps_pipe_inject_zlt':
/home/fergy/Development/nougat/kernel/sony/kitakami/drivers/platform/msm/sps/sps.c:2578: undefined reference to `sps_bam_pipe_inject_zlt'
drivers/built-in.o: In function `msm_pc_debug_counters_phys':
cpuquiet-msm.c:(.data+0x1477e8): undefined reference to `clk_ops_byte_multiparent'
cpuquiet-msm.c:(.data+0x1479d8): undefined reference to `clk_ops_byte_multiparent'
cpuquiet-msm.c:(.data+0x149950): undefined reference to `clk_ops_pixel_multiparent'
cpuquiet-msm.c:(.data+0x149b40): undefined reference to `clk_ops_pixel_multiparent'
mm/built-in.o: In function `get_pageblock_migratetype':
/home/fergy/Development/nougat/kernel/sony/kitakami/include/linux/mmzone.h:90: undefined reference to `get_pageblock_flags_group'
/home/fergy/Development/nougat/kernel/sony/kitakami/include/linux/mmzone.h:90: undefined reference to `get_pageblock_flags_group'
/home/fergy/Development/nougat/kernel/sony/kitakami/include/linux/mmzone.h:90: undefined reference to `get_pageblock_flags_group'
/home/fergy/Development/nougat/kernel/sony/kitakami/include/linux/mmzone.h:90: undefined reference to `get_pageblock_flags_group'
/home/fergy/Development/nougat/kernel/sony/kitakami/include/linux/mmzone.h:90: undefined reference to `get_pageblock_flags_group'`

fergy, 2016-09-25:
also `SYSTEM__MODULE_LOAD` is never defined from backport so please remove it as give error

erikcas, 2016-10-08:
builds fine here

zachariasmaladroit, 2016-10-08:
compile it with 4.9 toolchain,

5.3, 6.0, 7.0 toolchain might cause these kind of errors due to ABI change

erikcas, 2016-10-08:
I compile with default aosp toolchain. Compiles fine

fergy, 2016-10-09:
Problem solved. I revert to 4.9 compiler and it compile.
Closing issue.
Thank You for reply.

closed 2016-10-09 by fergy