sonyxperiadev/kernel #904: Some issues on Xperia Z2 (D6503, SonyAosp)

danielhoset27, 2016-09-30:
1) When switching off the Flashlight it makes a noise, sometimes the QS tile for Flashlight disappears.

Also, when using the "Super-Bright LED Torch" app from the Google Play Store, it says in the app that "It's detected that your hardware does not support LED Cam light. Screen brightness will be used to light up."

2) There is a 'Charging issue'... when you plug the USB cable to your phone it charges the phone but when you remove the USB cable, it will stop charging the phone but the system still thinks that the device is charging. 

^^ So, when you switch off the phone, you won't be able to turn on the phone using the power button. You've to press the "Power button + Volume Up button" to completely stop all the power from the board and then you can power on your phone by using the "Power button"

rcstar6696, 2016-09-30:
1. Is due a very unstable camera on android N 

2. Yes its the currently biggest issue i hope @kholk can help with a fix for it

scafroglia93, 2016-11-17:
Same problem on Z3

RyTekk, 2017-01-09:
"sometimes the QS tile for the Flashlight disappears"
Even this happens on latest Z5 Dual firmware 32.2.A.5.11

bartcubbins, 2017-11-11:
@danielhoset27 did it can be closed or you still have this problem?

closed 2017-11-11 by danielhoset27