sonyxperiadev/kernel #984: dma_mapping-fast breaking build.

HaoZeke, 2016-11-08:
Hey, using the aosp_kitakami_sumire_dsds_defconfig and trying from the AOSP instructions,  I'm getting 
In file included from ./arch/arm64/include/asm/dma-iommu.h:12:0,
                 from arch/arm64/mm/dma-mapping.c:33:
include/linux/dma-mapping-fast.h:46:15: warning: 'struct dma_iommu_mapping' declared inside parameter list

jerpelea, 2016-11-09:
what branch are you building?

HaoZeke, 2016-11-09:
LA.UM.5.5.r1
That's the nougat 7.1 one, though I had to use the defconfig from the default branch...

kholk, 2016-11-09:
MSM8994 devices are not supported yet on 3.18 branches.
The current WIP devices are based on MSM8996 and MSM8976/56, Tone and Loire platform. 

MSM8994 support will be evaluated and probably done after full implementation of those platforms and after MSM8939.

HaoZeke, 2016-11-09:
So, umm... is there a branch or something I should be trying to build against to see if I can help or should I just wait?

jerpelea, 2016-11-09:
follow 7.0 build guide and switch the android tag to the one you need

closed 2016-11-09 by jerpelea