sonyxperiadev/kernel #1121: MSM8994-Satsuki Camera orientation

BustyLoli-Chan, 2017-01-09:
Right now on the pre-built kernel images the front facing camera for the satsuki is rotated 90 degrees counter-clockwise.

I'm trying to edit line 
https://github.com/sonyxperiadev/kernel/blob/aosp/LA.BR.1.3.3_rb2.14/arch/arm/boot/dts/qcom/msm8994-kitakami_satsuki_camera.dtsi#L55
And change it's value to either <0> or <270> because I believe that is the value it needs to be in order to be oriented properly, but in all of my attempts to get a built version of this kernel actually running on the satsuki I've failed.

Can anyone more informed help me out? Can anyone tell me what I need to do to build a working version of this kernel for the E6853? 
Or can anyone fix the camera orientation in the prebuilt kernels since they seem to always work fine?

scafroglia93, 2017-01-09:
You can't build the kernel ?

BustyLoli-Chan, 2017-01-09:
I can "build" it, but putting it into my boot image and flashing it onto the device gives me the triple blinking red LED I've learned to associate with kernel failure. 

I'm using this build script though
`#!/bin/bash
./clean
rm -fr build_satsuki
mkdir -pv build_satsuki/out/modules

export ARCH=arm64
export PATH=/home/busty/android/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/bin/:$PATH
export CROSS_COMPILE=aarch64-linux-android-
export STRIP=aarch64-linux-android-strip

make aosp_kitakami_satsuki_defconfig
make menuconfig
make -j 4

tools/dtbTool --force-v2 -o build_satsuki/out/boot.img-dtb -s 2048 -p ./scripts/dtc/ ./arch/arm/boot/dts/qcom
cp -v arch/arm64/boot/Image build_satsuki/out/boot.img-zImage
cp -v arch/arm64/boot/Image.gz-dtb build_satsuki/out/boot.img-zImage_gz
make modules
find -type f -name *.ko -exec cp {} build_satsuki/out/modules/ \;
ls -al build_satsuki/out/modules/
cd build_satsuki/out/modules/
$STRIP --strip-unneeded *.ko
cd ../../../
ls -al build_satsuki/*`

Maybe I'm using the wrong defconfig? I've tried several toolchains including ubertc and the stock 4.9 one, I just can't get it to boot.

scafroglia93, 2017-01-09:
Wait a moment, i'm build the kernel for you

BustyLoli-Chan, 2017-01-09:
Alright, I'll stand by to put it in a boot image and flash it.

scafroglia93, 2017-01-09:
You use twrp ?

BustyLoli-Chan, 2017-01-09:
Yes!

scafroglia93, 2017-01-09:
This is good otherwise there's a little problem. I need boot.img for patch my image but if you use twrp no problem. I build with your repo

BustyLoli-Chan, 2017-01-09:
My repo isn't up to date. I can give you my current boot image if you need it.

scafroglia93, 2017-01-09:
This is build with your repo

[kernel-flasher-5.0-20170109-0247.zip](https://github.com/sonyxperiadev/kernel/files/692607/kernel-flasher-5.0-20170109-0247.zip)

BustyLoli-Chan, 2017-01-09:
I'll flash it with twrp, and it seems to be booting!
I have noticed however that adjusting the mount angle in the DTSI didn't actually change the camera orientation so my assumptions about that may have been wrong.

This does make me wonder though, why don't my builds work?

scafroglia93, 2017-01-09:
So your change is good ? have you fixed the problem related to orientation ?

BustyLoli-Chan, 2017-01-09:
I have not :(

scafroglia93, 2017-01-09:
It can happen...

scafroglia93, 2017-01-09:
You are interested ?

https://github.com/scafroglia93/kernel/commits/kcal

BustyLoli-Chan, 2017-01-09:
What is it?

Also what is your build setup/script?

BustyLoli-Chan, 2017-01-09:
Also I'm wondering if 
https://github.com/BustyLoli-Chan/kernel/blob/aosp/LA.BR.1.3.3_rb2.14/arch/arm/boot/dts/qcom/msm8994-camera-sensor-mtp.dtsi#L337

is the file that gets pulled instead...
This is why I want to get my build working.

scafroglia93, 2017-01-09:
for my kernel i use this

https://github.com/jcadduono/lazyflasher

Follow the instruction and don't forget, donate for this work is awesome

jerpelea, 2017-01-09:
hi,
i  checked reported problem and the camera is working finewith the AOSP camera.
What camera app are you using ?

BustyLoli-Chan, 2017-01-09:
I'm trying to use it with snapchat. 
The rear camera works fine, but the front camera is 90 degrees counter clockwise.

jerpelea, 2017-01-09:
i will install snapchat and test 
thanks

jerpelea, 2017-01-09:
what android version?

BustyLoli-Chan, 2017-01-09:
I guess I should also clarify this is on Android 7.1.1 but the problem also persists on 7.1

I'm currently using this xml file
`<?xml version="1.0" encoding="UTF-8"?>

<manifest>
<remote name="sony" fetch="https://github.com/sonyxperiadev/" />
<remove-project name="platform/hardware/qcom/camera" />
<project path="device/sony/sepolicy" name="device-sony-sepolicy" groups="device" remote="sony" revision="n-mr0" />
<project path="device/sony/common" name="device-sony-common" groups="device" remote="sony" revision="n-mr0" />
<project path="device/sony/common-headers" name="device-sony-common-headers" groups="device" remote="sony" revision="aosp/LA.BR.1.3.3_rb2.14" />
<project path="device/sony/common-kernel" name="vendor-sony-kernel" groups="device" remote="sony" revision="aosp/LA.BR.1.3.3_rb2.14" />
<project path="device/sony/common-init" name="device-sony-common-init" groups="device" remote="sony" revision="master" />
<project path="device/sony/kitakami" name="device-sony-kitakami" groups="device" remote="sony" revision="n-mr0" />
<project path="device/sony/satsuki" name="device-sony-satsuki" groups="device" remote="sony" revision="n-mr0" />
<project path="hardware/qcom/camera" name="camera" groups="device" remote="sony" revision="aosp/LA.BR.1.3.3_rb2.14" />
<project path="kernel/sony/msm" name="kernel" groups="device" remote="sony" revision="aosp/LA.BR.1.3.3_rb2.14" />
<project path="vendor/qcom/opensource/dataservices" name="vendor-qcom-opensource-dataservices" groups="device" remote="sony" revision="master" />
<project path="vendor/qcom/opensource/fm" name="vendor-qcom-opensource-fm" groups="device" remote="sony" revision="master" />
<project path="vendor/sony-oss/thermanager" name="thermanager" groups="device" remote="sony" revision="master" />
<project path="vendor/sony-oss/macaddrsetup" name="macaddrsetup" groups="device" remote="sony" revision="master" />
<project path="vendor/sony-oss/timekeep" name="timekeep" groups="device" remote="sony" revision="master" />
<project path="packages/apps/ExtendedSettings" name="packages_apps_ExtendedSettings" groups="device" remote="sony" revision="master" />
<project path="packages/apps/FMRadio" name="packages-apps-FMRadio" groups="device" remote="sony" revision="master" />
</manifest>`

and applying these patches
`cd external/toybox
git fetch https://android.googlesource.com/platform/external/toybox refs/changes/74/265074/1 && git cherry-pick FETCH_HEAD
git cherry-pick d3e8dd1bf56afc2277960472a46907d419e4b3da
git cherry-pick 1c028ca33dc059a9d8f18daafcd77b5950268f41
git cherry-pick cb49c305e3c78179b19d6f174ae73309544292b8
cd ../../hardware/qcom/audio
git revert --no-edit 66796eef5ebf71befa37b74f6507efae80d51ea0
git fetch https://android.googlesource.com/platform/hardware/qcom/audio refs/changes/91/294291/1 && git cherry-pick FETCH_HEAD
cd ../bt
git revert --no-edit c7dc913784965e4ce705c2045f0a8b43fcd1db1c
cd ../keymaster
git revert --no-edit 583ecf5ed2a4be0d05229b8c6726680c3836be8b
git fetch https://android.googlesource.com/platform/hardware/qcom/keymaster refs/changes/70/212570/5 && git cherry-pick FETCH_HEAD
git fetch https://android.googlesource.com/platform/hardware/qcom/keymaster refs/changes/80/212580/2 && git cherry-pick FETCH_HEAD
git fetch https://android.googlesource.com/platform/hardware/qcom/keymaster refs/changes/61/213261/1 && git cherry-pick FETCH_HEAD
cd ../../../system/core
git fetch https://android.googlesource.com/platform/system/core refs/changes/52/269652/1 && git cherry-pick FETCH_HEAD
cd ../../packages/apps/Music
git cherry-pick 6036ce6127022880a3d9c99bd15db4c968f3e6a3
cd ../../../frameworks/base/
git fetch https://github.com/AOSPA/android_frameworks_base nougat-caf
git cherry-pick d36582165d4694da101cc65755af0841d443c80e
git cherry-pick b70f5994464cf6b3b29cedcc4efdd73807a27b0f
cd ../../`

jerpelea, 2017-01-09:
thanks for the clarification

PDesire, 2017-01-11:
@BustyLoli-Chan Made once a guide do make kernel work on satsuki devices, maybe that can help you because i had same problem on E6833 

Link: https://forum.xda-developers.com/z5-premium/general/kernel-info-solving-kernel-boot-problems-t3530718/post70364270#post70364270

BustyLoli-Chan, 2017-01-11:
@PDesire 
I'll give that a try! 
Just out of curiosity, have you managed to apply those steps to get LA.UM.5.5.r1 working on the z5p at all?

scafroglia93, 2017-01-11:
Use lazyflasher

https://github.com/jcadduono/lazyflasher

seeliuh, 2017-01-18:
hello everyone!
I try to custom kernel on my device E6883, and i was failed. 
I followed the steps for this link: http://developer.sonymobile.com/open-devices/how-to-build-and-flash-a-linux-kernel/how-to-build-and-flash-a-linux-kernel-for-aosp-supported-devices/ .   All steps was OK. I got the boot.img and I flash onto my E6883, but it can’t boot, black screen.

I read and try this issues’s idea: @scafroglia93 use lazyflasher and twrp.  @PDesire and this Link https://forum.xda-developers.com/z5-premium/general/kernel-info-solving-kernel-boot-problems-t3530718/post70364270#post70364270   .  It’s not work to me.

Can anyone help me?

mcgi5sr2, 2017-01-19:
@seeliuh unlock the bootloader

PDesire, 2017-01-19:
@seeliuh I think I've found the problem. AOSP Kernels just accept one mkbootimg which is included into Compiling process of AOSP (no-prebuilt).

It's kinda different from any other mkbootimg by Google and so on so that could be the possible key to make it work

Will share soon the own built AOSP mkbootimg, in hope it fixes the problems

scafroglia93, 2017-01-19:
@seeliuh Wait a moment, i compile your kernel

jerpelea, 2017-01-19:
the mkbootimg source is in the AOSP and is automaticaly compiled and used to during the AOSP buid
please note that we use the appended dtb image

seeliuh, 2017-01-20:
@jerpelea you mean i should use the 'Image.gz-dtb'  to make the boot.img ?

seeliuh, 2017-01-20:
@scafroglia93 thx, I'm waiting

scafroglia93, 2017-01-20:
[kernel-flasher-5.1-20170120-0433.zip](https://github.com/sonyxperiadev/kernel/files/718361/kernel-flasher-5.1-20170120-0433.zip)

seeliuh, 2017-01-20:
hi @scafroglia93 , It's not working. 
I copy the zip file to sdcard, and run the twrp(version 2.8), install the zip file, reboot. 
When the power on, the sony logo will displayed about 1/2 second, and then,  black screen.

scafroglia93, 2017-01-20:
It's dualsim versione ?

seeliuh, 2017-01-20:
@scafroglia93   yes, it is. "XPERIA Z5 Premium Dual E6883"

scafroglia93, 2017-01-20:
Oppps this is for single... Excuse me

PDesire, 2017-01-20:
@seeliuh if you have that bootlogo problem, it's because it's made with a non Google Toolchain, UBERTC, Linaro and others don't work on our Z5P

scafroglia93, 2017-01-20:
@PDesire The're trying all ... You made some mistake 
[image]

scafroglia93, 2017-01-20:
Z5 Premium Dualsim is aosp_kitakami_satsuki_dsds_defconfig ?

ItzLevvie, 2017-01-20:
@scafroglia93 Yes, 'dsds' indicates Dual Sim

BustyLoli-Chan, 2017-01-20:
@seeliuh You mentioned you see the Sony logo when you boot. I've never seen the Sony logo while booting AOSP just the Android logo. 
Also try building branch LA.BR.1.3.3 I've never gotten LA.UM.5.5.r1 to build for the satsuki. 

Also @scafroglia93 I used the lazyflasher the other day. Thank you for the tip!

scafroglia93, 2017-01-20:
No I'm not at home, 30min and I release the kernel for z5 premium with lazy flasher :)

ItzLevvie, 2017-01-20:
@BustyLoli-Chan He meant the Sony splash screen logo not bootanimation from OS

BustyLoli-Chan, 2017-01-20:
@ItzLevvie after building and flashing aosp 7.1 the splash screen logo is gone. It's just a black screen and then the bootloader, at least for me.

zachariasmaladroit, 2017-01-20:
LA.UM.5.5.r1 just got Kitakami support via PR but it's still in heavy development, so 1.3.3 currently is the branch to use

scafroglia93, 2017-01-21:
DSDS Version
[kernel-flasher-5.1-20170121-1749.zip](https://github.com/sonyxperiadev/kernel/files/721317/kernel-flasher-5.1-20170121-1749.zip)

seeliuh, 2017-01-22:
hi @scafroglia93, I was downloaded kernel-flasher-5.1-20170121-1749.zip and try, it still not working.  after 1/2 seconds "sony logo", black screen

seeliuh, 2017-01-22:
@BustyLoli-Chan Only use the lazyflasher, can see the "sony logo". If I follow the link:http://developer.sonymobile.com/open-devices/how-to-build-and-flash-a-linux-kernel/how-to-build-and-flash-a-linux-kernel-for-aosp-supported-devices/ , make the boot.img myself and flash, just black screen.

ZeroX25, 2017-01-28:
In my builds of 7.0 using the updated sources I am still having issues with the camera. The UI is rotated (in any app) as well as the pictures being taken in that same orientation. This also includes disconnects and the front camera taking tv static resembling photos (the picture shows only the colors like when the TV lost the connection to whatever you were watching).

I'm also getting an unknown baseband/no imei. I have the satsuki_dsds variant.

seeliuh, 2017-02-13:
hi @scafroglia93 . my rom version is: 32.0.A.6.209 . android version is 5.1.   
is this the reason that I use your kernel boot failed?
What should i do next?

jerpelea, 2017-02-13:
please update your bootloader ( flash the latest sw) using Sony flashtool
http://developer.sonymobile.com/open-devices/flash-tool/

BustyLoli-Chan, 2017-02-14:
@jerpelea does updating the bootloader actually fix the camera orientation?

jerpelea, 2017-02-14:
i tested the v8 blobs and camera is ok

BustyLoli-Chan, 2017-02-14:
@jerpelea Forgive me for asking, but what are the `v8 blobs`?

jerpelea, 2017-02-14:
the binaries published for the OpenDevices program
http://developer.sonymobile.com/open-devices/

BustyLoli-Chan, 2017-02-14:
I understand now!

kholk, 2017-02-14:
The latest bootloader is required on kitakami platform to boot compressed kernel images, which is what we use on open devices.

BustyLoli-Chan, 2017-02-14:
Even when on the v8 blobs my front camera is still 90 degrees counter-clockwise. 
I'm on the katakami satsuki, using kernel LA.BR.1.3.3_rb2.14
And using the Selinux, common, katakami, and satsuki stuff from 'n-mr1'

[2 more replies]

closed 2017-03-13 by BustyLoli-Chan