sonyxperiadev/kernel #1146: [1.3.3][D66x3] Reboot in call

nailyk-fr, 2017-01-15:
On LineageOS leo experience reboot in call after screen goes dark. 
This issue doesn't happens while plugged into usb, even after merging https://github.com/sonyxperiadev/kernel/pull/1130. 
If prevent screen going dark (setting timeout at 30min) and using headset for example, phone doesn't reboot until 30 minutes are reached. 

Don't know if this issue is related to https://github.com/sonyxperiadev/kernel/issues/797 or is relevant because of Lineage. 

Issue present on each LOS build. 

Partial output with this [kernel](https://github.com/nailyk-fr/kernel/tree/test) (include #1131 #1130 #1128 #1120): 
```
[  496.498192] Error: returning -512 value
[  496.513756] Unable to handle kernel NULL pointer dereference at virtual address 0000001c
[  496.513769] pgd = c3a78000
[  496.513788] [0000001c] *pgd=00000000
[  496.513810] Internal error: Oops: 5 [#1] PREEMPT SMP ARM
[  496.513839] CPU: 1 PID: 1331 Comm: system_server Not tainted 3.10.84-gaeaeb5a #1
[  496.513858] task: c4411f40 ti: c3ad8000 task.ti: c3ad8000
[  496.513888] PC is at snd_pcm_do_suspend+0x44/0x54
[  496.513907] LR is at snd_pcm_action_single+0x38/0x64
[  496.513929] pc : [<c0752cb0>]    lr : [<c0752910>]    psr: 60000093
[  496.513929] sp : c3ad9cd0  ip : 0000e64b  fp : c0e7bfec
[  496.513943] r10: 00000002  r9 : c0c7822c  r8 : c03a740c
[  496.513958] r7 : c0f9c914  r6 : 00000000  r5 : c0f8b080  r4 : c426ee00
[  496.513972] r3 : c426ee00  r2 : 00000000  r1 : 00000005  r0 : c426ee00
[  496.513993] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  496.514011] Control: 10c5787d  Table: 03a7806a  DAC: 00000015
[  496.514028] Process system_server (pid: 1331, stack limit = 0xc3ad8238)
[...]
[  496.515401] ---[ end trace 749da69a7594b5f2 ]---
[  496.537614] Kernel panic - not syncing: Fatal exception
[  496.537822] CPU0: stopping
[  496.537848] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G      D      3.10.84-gaeaeb5a #1
[  496.537907] [<c00139c0>] (unwind_backtrace+0x0/0x120) from [<c0011c54>] 
[...]
[  498.366798] Rebooting in 5 seconds..
[  503.372402] Going down for restart now
```
[Full oops](https://github.com/sonyxperiadev/kernel/files/706937/console-ramoops.txt)

I can provide any other information. 
IDK how to read stack traces to debug this. 

Error -512 doesn't seems to be related to this panic because can happens and the device doesn't reboot. However panic is *always* predate by this -512 line. 
```
nailyk :) % rgrep "Error: returning"                                                                                       [eac56be...] ~/cyanogen/14.1/kernel/sony/msm8974
drivers/usb/gadget/f_fs.c:970:          pr_err("Error: returning %zd value\n", ret);
```

rancidfrog, 2017-01-16:
This is similar to what happens in aries: https://github.com/sonyxperiadev/kernel/issues/797#issuecomment-256906275
I do not think it has anything to do with screen going dark, as screen goes dark when proximity sensor detects something and it does not always reboot.
Maybe devs will look into it know that someone else was able to duplicate it, as they do not seem to believe me when I do.

nailyk-fr, 2017-01-16:
Sorry for my English. I think it is related to sleep. 
Proximity sensor doesn't reboot in call, but if you lock screen with power button it will reboot 10 seconds after. 
On the z3c Lineage [thread](https://forum.xda-developers.com/z3-compact/orig-development/rom-cyanogenmod-14-1-t3514523) some users report the incall reboot probem is not present and your issue report random reboots that is why I don't know if it is related. 
Does all your reboot are predate with the -512 error?

rancidfrog, 2017-01-16:
```
[10763.437927] Error: returning -512 value
[10763.452033] Unable to handle kernel NULL pointer dereference at virtual address 0000001c
[10763.452046] pgd = c4a2c000
[10763.452065] [0000001c] *pgd=00000000
[10763.452087] Internal error: Oops: 5 [#1] PREEMPT SMP ARM
[10763.452116] CPU: 1 PID: 1259 Comm: system_server Tainted: G        W    3.10.84-gd77ff96 #1
[10763.452133] task: c42cddc0 ti: ea9ba000 task.ti: ea9ba000
[10763.452164] PC is at snd_pcm_do_suspend+0x44/0x54
[10763.452181] LR is at snd_pcm_action_single+0x38/0x64
[10763.452202] pc : [<c0730dac>]    lr : [<c0730a0c>]    psr: 60000093
[10763.452202] sp : ea9bbcd0  ip : 000ff749  fp : c0e43f6c
[10763.452215] r10: 00000002  r9 : c0c4a388  r8 : c038db04
[10763.452229] r7 : c0f05cb4  r6 : 00000000  r5 : c0ef4440  r4 : f2829200
[10763.452242] r3 : f2829200  r2 : 00000000  r1 : 00000005  r0 : f2829200
[10763.452261] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[10763.452276] Control: 10c5787d  Table: 04a2c06a  DAC: 00000015
[10763.452290] Process system_server (pid: 1259, stack limit = 0xea9ba238)
[10763.452303] Stack: (0xea9bbcd0 to 0xea9bc000)
[10763.452323] bcc0:                                     c0730d68 c0730a0c 00000000 f2829200
[10763.452348] bce0: f28292fc a0000013 c0f05cb4 c073334c f2829200 00000002 f2829000 c07333c4
[10763.452372] bd00: c0f05bcc 00000039 000005e4 c074cd98 c074cca4 00000000 00000000 f3f0bc10
[10763.452395] bd20: 00000002 c038db04 c0c4a388 c038db30 00000000 c0392990 c0e0a6d0 c003c41c
[10763.452419] bd40: 00000000 00000000 f3f0bc10 f3f0bc10 00000002 c113e674 f3f0bc44 f3f0bc10
[10763.452444] bd60: c0ebcdc8 c0393a84 009bbdd8 f3f0bc10 c42cddc0 f3c2a9ec f3c2a9ec 000ffbfc
[10763.452466] bd80: f3c2a000 c0392224 ea9bbd6c ffffffff ffffffff 00000000 00000000 00000000
[10763.452490] bda0: 00000000 00000000 c0f6da40 00000037 00000037 00000001 c0fe30b0 c022986c
[10763.452514] bdc0: c0f6da40 ea9bbdd0 c002f29c c0bd9b77 00000000 c0fe30a0 c0f6da40 00000037
[10763.452537] bde0: 00000037 c0f6da4f 00000046 c002f390 00000001 00000000 60000093 c099e690
[10763.452560] be00: 00000000 c0e40b90 00000000 c006d900 60000093 c099e7dc 00000f9e c002fd14
[10763.452583] be20: 00000400 c0f6d548 c0f6d548 00074e28 00008058 00000000 00000002 00000001
[10763.452605] be40: 00000002 00000000 c0e1b338 c0e0a4b0 00000001 00000037 00000002 c002fa00
[10763.452626] be60: 00000000 c0030160 00000000 00000000 00000000 00000000 c0f6de84 00000037
[10763.452650] be80: c0bff7dc ea9bbe9c 00000000 60000013 a7ecee5c c0082484 c0f6d548 00000003
[10763.452672] bea0: 00000000 c113e674 f3c04e74 f3f0bc74 00000000 c113e674 00000000 c0393bbc
[10763.452696] bec0: 0fd4b49b 000009ca 0fd4b49b 000009ca 00000003 00000003 c0f267f0 f3e65400
[10763.452720] bee0: f3e65500 c09ce430 a30654e8 c0070b94 00000003 00000007 c113e674 00000000
[10763.452743] bf00: 00000003 c0070e9c c1071ae0 00000003 00000003 eaad0000 ea9bbf80 c006fc84
[10763.452766] bf20: 00000003 f2140a40 f2140a58 c0221740 00000003 c01579dc a3065315 ea9bbf80
[10763.452789] bf40: edfcc000 00000003 ea9ba018 00000003 00000000 c010305c edfcc000 a3065315
[10763.452810] bf60: 000002b2 00000000 edfcc000 00000001 a3065315 00000003 00000000 c01031e4
[10763.452833] bf80: 000002b2 00000000 00000003 a3067048 8eae1888 8eae189c 00000004 c000e4c4
[10763.452856] bfa0: ea9ba000 c000e340 a3067048 8eae1888 0000003e a3065315 00000003 00000004
[10763.452880] bfc0: a3067048 8eae1888 8eae189c 00000004 000186a0 a3065290 a306704c a30654e8
[10763.452904] bfe0: a3065553 8eae1880 a30639e3 a5707f9c 000f0010 0000003e ffffffff ffffffff
[10763.452952] [<c0730dac>] (snd_pcm_do_suspend+0x44/0x54) from [<c0730a0c>] (snd_pcm_action_single+0x38/0x64)
[10763.452989] [<c0730a0c>] (snd_pcm_action_single+0x38/0x64) from [<c073334c>] (snd_pcm_suspend+0x34/0x5c)
[10763.453025] [<c073334c>] (snd_pcm_suspend+0x34/0x5c) from [<c07333c4>] (snd_pcm_suspend_all+0x50/0x64)
[10763.453077] [<c07333c4>] (snd_pcm_suspend_all+0x50/0x64) from [<c074cd98>] (snd_soc_suspend+0xf4/0x39c)
[10763.453139] [<c074cd98>] (snd_soc_suspend+0xf4/0x39c) from [<c038db30>] (platform_pm_suspend+0x2c/0x58)
[10763.453198] [<c038db30>] (platform_pm_suspend+0x2c/0x58) from [<c0392990>] (dpm_run_callback+0x44/0x7c)
[10763.453246] [<c0392990>] (dpm_run_callback+0x44/0x7c) from [<c0393a84>] (__device_suspend+0x1f0/0x284)
[10763.453291] [<c0393a84>] (__device_suspend+0x1f0/0x284) from [<c0393bbc>] (dpm_suspend+0xa4/0x200)
[10763.453337] [<c0393bbc>] (dpm_suspend+0xa4/0x200) from [<c0070b94>] (suspend_devices_and_enter+0xc0/0x258)
[10763.453374] [<c0070b94>] (suspend_devices_and_enter+0xc0/0x258) from [<c0070e9c>] (pm_suspend+0x170/0x2a0)
[10763.453405] [<c0070e9c>] (pm_suspend+0x170/0x2a0) from [<c006fc84>] (state_store+0x40/0x68)
[10763.453455] [<c006fc84>] (state_store+0x40/0x68) from [<c0221740>] (kobj_attr_store+0x14/0x20)
[10763.453500] [<c0221740>] (kobj_attr_store+0x14/0x20) from [<c01579dc>] (sysfs_write_file+0x104/0x148)
[10763.453548] [<c01579dc>] (sysfs_write_file+0x104/0x148) from [<c010305c>] (vfs_write+0xcc/0x1b4)
[10763.453590] [<c010305c>] (vfs_write+0xcc/0x1b4) from [<c01031e4>] (SyS_write+0x38/0x68)
[10763.453637] [<c01031e4>] (SyS_write+0x38/0x68) from [<c000e340>] (ret_fast_syscall+0x0/0x38)
[10763.453666] Code: 1a000004 e59320e0 e1a00003 e3a01005 (e592201c) 
[10763.453686] ---[ end trace e5cc1fca497956d7 ]---
[10763.536236] Kernel panic - not syncing: Fatal exception
[10763.536432] CPU0: stopping
[... cut ...]
[10763.537011] [<c098bdbc>] (rest_init+0x8c/0xa8) from [<c0db0a14>] (start_kernel+0x310/0x36c)
[10765.423962] Rebooting in 5 seconds..
[10770.429561] Going down for restart now
[10770.433400] Calling SCM to disable SPMI PMIC arbiter
```
Same error in Aries, you can compare above from aries with your log.
But, it might not be related to screen off, which is why z3c users report not present.
Incall reboots have been happening for a while.

There might be a difference between placing calls and receiving calls, when recieve calls there seems to be more reboots than when placing calls.

nailyk-fr, 2017-01-16:
Your log show the same entry point: 

> [10763.452952] [<c0730dac>] (snd_pcm_do_suspend+0x44/0x54) from [<c0730a0c>] (snd_pcm_action_single+0x38/0x64)

on leo no difference between incoming or outgoing calls from my smalls tests. However I notice the Lineage call app have different behavior for this. You can force sleep entering with power button. 
@rancidfrog  : Are you on Lineage, on AOSP or something else? 
Do someone know how to collect logcat for this? logcat > /cache/foo.log in a term app doesn't work anymore when device try to enter sleep mode.

rancidfrog, 2017-01-16:
For last logcat and last kmsg after a reboot, I use terminal emulator:
```
su
cd/sdcard
bugreport > bug.txt
```
It gets all necessary info, just search for last_kmsg and last logcat
As for Rom, I use aosp and DU aosp branches

nailyk-fr, 2017-01-16:
Relevant part of last logcat: 
```
01-16 13:51:02.489  1303  1974 I Simple PowerHAL: Device is asleep.
01-16 13:51:02.489  1303  1974 I Simple PowerHAL: Setting low power mode
```
Then some sleep for wifi and bluetooth and log stop. 

Maybe 'only' a missing wakelock into ril for shinano?

rancidfrog, 2017-01-16:
https://github.com/sonyxperiadev/kernel/issues/797#issuecomment-259080442

> It looks like your device started the suspend sequence while you were in call.
This should not happen.... While in call your device should stay awake, so something is not setting the wakelock

nailyk-fr, 2017-01-17:
Segfault happen [here](https://github.com/sonyxperiadev/kernel/blob/aosp/LA.BR.1.3.3_rb2.14/sound/core/pcm_native.c#L1082). 
substream & SNDRV_PCM_TRIGGER_SUSPEND are set. 
So ops should be null. Now need to identify where this object is created. No idea how to track this.

nailyk-fr, 2017-01-20:
@kholk with QUAT_MI2S_RX ignore_suspend=0: 
```
[  311.981859] SND: link name SLIMBUS_0_RX ingore_suspend 1
[  311.981870] SND: link name SLIMBUS_0_TX ingore_suspend 1
[  311.981881] SND: link name SLIMBUS_1_RX ingore_suspend 1
[  311.981891] SND: link name SLIMBUS_1_TX ingore_suspend 1
[  311.981901] SND: link name SLIMBUS_3_RX ingore_suspend 1
[  311.981911] SND: link name SLIMBUS_3_TX ingore_suspend 1
[  311.981921] SND: link name SLIMBUS_4_RX ingore_suspend 1
[  311.981932] SND: link name INCALL_RECORD_RX ingore_suspend 1
[  311.981942] SND: link name INCALL_RECORD_TX ingore_suspend 1
[  311.981952] SND: link name SLIMBUS_5_TX ingore_suspend 1
[  311.981963] SND: link name QUAT_MI2S_RX ingore_suspend 0
[  311.981971] SND: suspend_all called
[  311.982065] Unable to handle kernel NULL pointer dereference at virtual address 0000001c
[  311.982078] pgd = c4824000
[...]
```
Don't know where QUAT_MI2S_RX.ops is initiated but it should be this [one](https://github.com/nailyk-fr/kernel/blob/42c01d8e3f937f78a65e646aaa7f427ae6909c4f/sound/soc/msm/msm8974.c#L1476) which doesn't handle *trigger (where the segfault is). 
Also all my reading explain when phone call is established, rild communicate directly to RIL trough a serial interface (so without kernel interaction). 
Still agree this is not the best way but I can't find another one. If you wont merge it please explain and help me to do it. I'm still a C beginner and running out of ideas.

RyTekk, 2017-01-21:
https://github.com/MotorolaMobilityLLC/kernel-msm/commit/d4e5a04392aecc4cb158848fe412fd04719537d9
This commit should solve the returning error message.

nailyk-fr, 2017-03-01:
Thanks for merging. I think about leaving this issue open until the proper fix will be found. 

@RyTekk yes your patch fix it, thanks

bartcubbins, 2017-11-11:
@jerpelea looks like it can be closed

closed 2017-11-11 by jerpelea