sonyxperiadev/kernel #1152: Not poweroff

russel5, 2017-01-20:
Hello i'm using Z3Compact and build twrp recovery with your kernel sources. No matter what branch of omnirom i'm using, if i made recovery with your sources the phone are not poweroff at all. It's just always reboot after some time, ie blackscreen - 5 or 10 sec - reboot
But if i'm using your official kernel sources, like 291, all works perfect, ie i'm pressed power off in recovery and phone off not reboot as with your aosp sources. Can you check this issues, thanks.
I'm using omni - 6.0 & 7.0 for compiling, with your sources or default and always get this issue.

scafroglia93, 2017-01-20:
My Z3 got this problem when I charge the phone

@kholk

russel5, 2017-01-21:
One more things
Poweroff not work not only in recovery - reboot - poweroff
It's not work in terminal via busybox
And it's a bug in this kernel sources

russel5, 2017-01-22:
Seems like the latest commits for twrp fix this bug. Closed

closed 2017-01-22 by russel5