sonyxperiadev/kernel #1210: random32.so

mcgi5sr2, 2017-03-09:
CC      lib/random32.o
/home/mcgi5sr2/AOSP/android/system/kernel/sony/msm/lib/random32.c:179:50: error: expected ')' before numeric constant
 static DEFINE_TIMER(seed_timer, __prandom_timer, 0, 0);
                                                  ^
/home/mcgi5sr2/AOSP/android/system/kernel/sony/msm/lib/random32.c: In function '__prandom_timer':
/home/mcgi5sr2/AOSP/android/system/kernel/sony/msm/lib/random32.c:188:2: error: 'seed_timer' undeclared (first use in this function)
  seed_timer.expires = jiffies + (40 * HZ + (prandom_u32() % (40 * HZ)));
  ^
/home/mcgi5sr2/AOSP/android/system/kernel/sony/msm/lib/random32.c:188:2: note: each undeclared identifier is reported only once for each function it appears in
/home/mcgi5sr2/AOSP/android/system/kernel/sony/msm/lib/random32.c:189:2: error: implicit declaration of function 'add_timer' [-Werror=implicit-function-declaration]
  add_timer(&seed_timer);
  ^
/home/mcgi5sr2/AOSP/android/system/kernel/sony/msm/lib/random32.c: In function 'prandom_start_seed_timer':
/home/mcgi5sr2/AOSP/android/system/kernel/sony/msm/lib/random32.c:194:2: error: implicit declaration of function 'set_timer_slack' [-Werror=implicit-function-declaration]
  set_timer_slack(&seed_timer, HZ);
  ^
/home/mcgi5sr2/AOSP/android/system/kernel/sony/msm/lib/random32.c:194:19: error: 'seed_timer' undeclared (first use in this function)
  set_timer_slack(&seed_timer, HZ);
                   ^
/home/mcgi5sr2/AOSP/android/system/kernel/sony/msm/lib/random32.c: At top level:
/home/mcgi5sr2/AOSP/android/system/kernel/sony/msm/lib/random32.c:181:13: warning: '__prandom_timer' defined but not used [-Wunused-function]
 static void __prandom_timer(unsigned long dontcare)
             ^
cc1: some warnings being treated as errors
/home/mcgi5sr2/AOSP/android/system/kernel/sony/msm/scripts/Makefile.build:308: recipe for target 'lib/random32.o' failed
make[2]: *** [lib/random32.o] Error 1
/home/mcgi5sr2/AOSP/android/system/kernel/sony/msm/Makefile:821: recipe for target 'lib' failed
make[1]: *** [lib] Error 2
make[1]: *** Waiting for unfinished jobs....
Makefile:130: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2

kholk, 2017-03-09:
Did compile for me. Checking ASAP.

lowtraxx, 2017-03-10:
Got the same errors here. Cleaned and tried again, stillfailing.

lowtraxx, 2017-03-10:
Adding #include <linux/timer.h> into random32.c fixes the build for me.

ergoen, 2017-03-10:
In mainline <linux/sched.h> is included in random32.c, adding this include also fixes the compile problem (sched.h includes timer.h I suppose).

mcgi5sr2, 2017-03-10:
https://github.com/torvalds/linux/commit/f6a570333e554b48ad589e7137c77c57809eee81

mainline commit that adds sched.h include

jerpelea, 2017-03-10:
pr please

mcgi5sr2, 2017-03-10:
https://github.com/torvalds/linux/commit/a6a9c0f1bf5a9a5faa605773ea75e0b93c3ab108

is the commit from mainline thats adds sched.h include

kholk, 2017-03-10:
I'm on it.

mcgi5sr2, 2017-03-12:
n1 @kholk

closed 2017-03-12 by mcgi5sr2