sonyxperiadev/kernel #1214: [1.3.3] amami - Mobile Data Connection only work on LTE

Micha-Btz, 2017-03-12:
Hi there,

at first I don't know if this is the right place to post this but I have found nothing about the right place.
Also I know that the amami device is some years old and so maybe the developers have gone to the newer phone. But I ask anyway maybe some other older devices benefit also from this.
I build LOS with these manifest https://github.com/Micha-Btz/local_manifests/blob/cm-14.1/roomservice.xml
and like the topic says mobile data is only working on LTE. Since the AICP 12.1 have the same problem I hope here for some help.
I have make a big logcat and would be great when someone with the knowlege to solve this can take a look.
https://gist.github.com/Micha-Btz/2c65f7212c3438e83cbdcaa045408080

If you need more debugging please let me know how to get these and I will attach them here.
Thanks for reading.
Micha

jerpelea, 2017-03-12:
https://github.com/hahnjo/android_device_sony_common-aosp/commit/2028849ac52b84dc1680366cb3abf878bd16e5c5

hahnjo, 2017-04-02:
@jerpelea No, this doesn't solve the problem unfortunately. However, I think the problem description is not 100 percent accurate: Mobile data works with 2G and 3G, but only if the connection was established with LTE and later "downgraded". So the issue seems to be the initial connection. (see also https://github.com/sonyxperiadev/device-sony-amami/issues/43)

ergoen, 2017-04-02:
@hahnjo If this is related to the RIL problems causing the wakelock preventing the device from going to deep sleep, it can be fixed. One option is to hack the android framework to avoid calling a broken function in the RIL ( SuperLamic did something like that in LOS here: https://github.com/SonyLOS/android_device_sony_rhine-common/pull/1 ), the other option is to hack the binary blob to bypass the offending function ( https://forum.xda-developers.com/crossdevice-dev/sony/modified-libril-qc-qmi-blobs-aosp-8974-t3583370 ). I am currently testing this on amami, and it seems to work. I never encountered your 3G problem (since I mostly have LTE coverage), but setting the phone to force 3g followed by flight mode on and flightmode off seems to get me functional 3g data without connecting to LTE first.

Micha-Btz, 2017-04-03:
@ergoen @hahnjo 
I have both patches in my LOS build and I'm unable to get mobile data to work.
I don't have LTE in my area.
https://github.com/Micha-Btz/android_device_sony_common-aosp/commits/cm-14.1
https://github.com/Micha-Btz/android_device_sony_rhine-common/commits/cm-14.1

there are one report https://forum.xda-developers.com/showpost.php?p=71444201&postcount=90
that he get's it to work but with many workarounds.

ergoen, 2017-04-03:
@Micha-Btz Interesting bug! I have problems reproducing it (without physically going to a place without LTE I guess..). Stock AOSP doesn't allow me to force 3g, it goes into LTE whatever I set. With the modified binary I linked I can force 3g and reboot and get a working H+ connection, but it might be operator dependent anyway... I don't currently have a build with the linked patches unfortunately to compare.

Micha-Btz, 2017-04-03:
@ergoen you can try this build 
https://www.androidfilehost.com/?fid=457095661767144531
created with this manifest
https://github.com/Micha-Btz/local_manifests/blob/cm-14.1/roomservice.xml

hahnjo, 2017-04-03:
@ergoen I can reproduce 3G is working if data connection is activated by toggling the airplane mode. However if one later only toggles the mobile data connection, it can't reconnect. Can you test this with your build?

ergoen, 2017-04-03:
@Micha-Btz I will not try your build right now as going from AOSP to LOS would probably mess up my userdata, but I can try to build an AOSP with the equivalent patches this evening to compare with.

@hahnjo Hm, okay, now I can reproduce it kind of. Steps taken:
1. Set preferred network type to 3g and wait a few seconds for HSPA connection to establish.
2. Disable cellular data. Wait until it properly goes away.
3. Enable cellular data.

After these steps I have no data connection for 2-3 minutes. Then the H+ appears and it starts working... I will investigate a bit more later.

ergoen, 2017-04-03:
@Micha-Btz @hahnjo Setting persist.qcril_uim_vcc_feature=1 in the build.prop fixes the 3g data connection for me. Do you know if this is an issue for all Rhine, or only amami? (Should the pull request go to device/sony/rhine or device/sony/amami ?)

hahnjo, 2017-04-04:
@ergoen Nope, doesn't work for me. LTE is there within a few moments while 3G wasn't after half a minute

alviteri, 2017-05-26:
@hahnjo modem files (available for rhine) are for lollipop try https://github.com/SonyAosp/vendor_google/commit/020d70e6a8d547550e0dd633964f91589eb596f1

hahnjo, 2017-05-26:
@alviteri as written above, we already have that in place, see also https://github.com/SonyLOS/android_device_sony_rhine-common/pull/1

bartcubbins, 2017-11-11:
@jerpelea that is not the kernel issue. can be closed

closed 2017-11-11 by jerpelea