sonyxperiadev/kernel #1274: DTB not found

TitanNano, 2017-04-30:
I'm trying to build 7.1 for my `Ivy (E6533)` device. Therefore I cloned the AOSP source as described in the instructions for android 7.1 Kernel 3.10 (https://developer.sonymobile.com/open-devices/aosp-build-instructions/how-to-build-aosp-nougat-for-unlocked-xperia-devices/). At first I just built the recovery image to see if everything is working well. Unfortunately here I already encounter the first issue, when I try to boot the resulting `boot.img` or `recovery.img` with "fastboot boot boot.img`, it returns "FAILED (dtb not found)".

I know that the prebuild kernel already includes the dtb, so why is it not found here? Did I mess up something? I'd be glad if you could help me out here.

jerpelea, 2017-04-30:
please flash the latest Sony software to get a bootloader update

TitanNano, 2017-04-30:
@jerpelea I already unlocked my bootloader, so Device Companion told me that I can't update. Does this mean I have to lock it, update and then unlock it again? Thanks for your help

jerpelea, 2017-04-30:
https://developer.sonymobile.com/open-devices/

here you have the flashtool for unlocked devices and resources for our project

TitanNano, 2017-04-30:
@jerpelea I tried the flashtool, but it seems to only support the Z3+ (E6553) not the Z3+ Dual (E6533).

jerpelea, 2017-04-30:
i will check with my coleagues

kholk, 2017-04-30:
@TitanNano http://www.flashtool.net/downloads.php
Use this and flash your device

TitanNano, 2017-04-30:
@kholk I have this tool, but how can I download the official firmware?

kholk, 2017-04-30:
XperiFirm inside the tool itself.....

TitanNano, 2017-05-02:
okay I got it to work, thank you guys.

closed 2017-05-02 by TitanNano