sonyxperiadev/kernel #1292: LA.UM.5.7.r1 for kitakami

grayleshy, 2017-05-26:
Hi! 

Prompt please whether there will be for the Z5 series new kernel LA.UM.5.7.r1? 

At the moment, a kernel from the outdated LA.BR.1.3.3_rb2.14 version. 

_Sorry for my bad English._

jerpelea, 2017-05-26:
kitakami will remain on LA.BR.1.3.3_rb2.14

kholk, 2017-05-26:
Because of architectural problems (really bad support of QCOM IOMMUv1 and OCMEM), MSM8974 MSM8994 will be supported on LA.BR.1.3.3_rb2.14 for new versions of Android, but will not receive new kernel upgrades.

Sorry for this.

grayleshy, 2017-05-26:
Thanks for the response and for clarification.

NeoBeum, 2017-05-27:
Is this part of what humberos mentioned to me about the Kitakami platform changes in Omnirom to try and find a better way to port over the family to Omni?

closed 2017-07-14 by grayleshy

DroidThug, 2017-10-06:
@kholk (slightly off-OEM question) Is MSM8953 platform compaitable with the upgrade to MSM-4.4?

kholk, 2017-10-06:
Formal reply: no
Reality: no! :P not from qualcomm, at least... Neither was msm8976, but we have it here.
MSM8953 is easier to port compared to 8976, so if you really want to, you can.
Follow my msm8976 porting job and you'll get somewhere.

By the way, first and last warning: this is an old issue on github. If you want to ask a question, open a new one!!

makx1998998, 2019-03-14:
> Formal reply: no
> Reality: no! :P not from qualcomm, at least... Neither was msm8976, but we have it here.
> MSM8953 is easier to port compared to 8976, so if you really want to, you can.
> Follow my [...]

 Please give link