sonyxperiadev/kernel #1462: ecryptfs is broken

jamuir, 2017-10-17:
The sony kernel enables the ecryptfs filesystem; however, the following CAF commit breaks ecryptfs file operations:  8928f8683

If you install the `keyctl` utility, then you can demonstrate the defect as follows:

```
maple:/data/local/tmp # dd if=/dev/urandom of=mykey.txt bs=1 count=64
64+0 records in
64+0 records out
64 bytes transferred in 0.003 secs (21333 bytes/sec)
maple:/data/local/tmp # cat mykey.txt | keyctl padd user mykey @us
409613533
maple:/data/local/tmp # keyctl add encrypted 1000000000000000 "new ecryptfs user:mykey 64" @us                                                                                              
522453367
maple:/data/local/tmp # mkdir Private
maple:/data/local/tmp # mount -t ecryptfs -o ecryptfs_sig=1000000000000000,ecryptfs_cipher=aes,ecryptfs_key_bytes=32 Private Private                                                        
maple:/data/local/tmp # touch Private/foo.txt
touch: 'Private/foo.txt': Invalid argument
```
You are not able to create any files inside the ecryptfs mounted directory.

The problem is that 8928f8683 changes some keysize parameters in the file `fs/ecryptfs/keystore.c`:

https://github.com/sonyxperiadev/kernel/commit/8928f8683bcd0236f5653963deee3bc225fb2206#diff-9d49736092287dd68ae479bb68340895L2299

This causes an invalid keysize (64 bytes) to be passed into an aes setkey operation (64 is too large).  The value 64 comes from the function `ecryptfs_fill_auth_tok()` in `security/keys/encrypted-keys/ecryptfs_format.c`

Here is the relevant dmesg output for the error above:

```
[  150.085670] ecryptfs_initialize_file: Initializing crypto context
[  150.085795] ecryptfs_generate_new_key: Generated new session key:
[  150.085808] 0x69.0x56.0x94.0xf9.0xbb.0x5a.0x2b.0xd9.0x9c.0xab.0xfb.0xf1.0x0b.0xba.0xc7.0x2b.
[  150.085817] 0xaa.0x26.0xbe.0xb2.0xe3.0xcc.0x8e.0xc0.0x19.0xa3.0x30.0xaa.0x09.0x82.0xbc.0x3f.
[  150.085821] ecryptfs_dump_cipher: ecryptfs cipher is aes
[  150.085823] ecryptfs_dump_cipher: ecryptfs cipher mode is 
[  150.085826] ecryptfs_init_crypt_ctx: Initializing cipher [aes]; strlen = [3]; key_size_bits = [256]
[  150.086468] ecryptfs_dump_cipher: ecryptfs cipher is aes
[  150.086469] ecryptfs_dump_cipher: ecryptfs cipher mode is 
[  150.086471] ecryptfs_dump_cipher: ecryptfs cipher is aes
[  150.086472] ecryptfs_dump_cipher: ecryptfs cipher mode is 
[  150.086475] write_tag_3_packet: Using previously generated session key encryption key of size [64]
[  150.086477] write_tag_3_packet: Cached session key encryption key:
[  150.086486] 0x82.0xf5.0xbc.0x03.0xec.0x29.0x1e.0x40.0xb5.0xb2.0xfa.0x95.0xb6.0xc3.0xfe.0x22.
[  150.086487] write_tag_3_packet: Session key encryption key:
[  150.086495] 0x82.0xf5.0xbc.0x03.0xec.0x29.0x1e.0x40.0xb5.0xb2.0xfa.0x95.0xb6.0xc3.0xfe.0x22.
[  150.086500] write_tag_3_packet: Error setting key for crypto context; rc = [-22]
[  150.086502] ecryptfs_generate_key_packet_set: Error writing tag 3 packet
[  150.086504] ecryptfs_write_headers_virt: Error generating key packet set; rc = [-22]
[  150.086506] ecryptfs_write_metadata: Error whilst writing headers; rc = [-22]
[  150.086510] Error writing headers; rc = [-22]
```

jamuir, 2017-10-23:
This is a CAF / Android-for-MSM issue.  I'd be happy to file a bug report with them if someone could give me a pointer.

jerpelea, 2017-10-24:
You can try an port the patches from the Pixel 2 kernel since Google appears to have it fixed.
Additional to this you should leave the issue open here.

jamuir, 2017-10-24:
Looking at the Pixel 2 kernel source is a good idea.

https://dl.google.com/dl/android/aosp/taimen-opd1.170816.012-factory-20574403.zip

The factory image above uses revision 845c8307ee66:

https://android.googlesource.com/kernel/msm/+/845c8307ee66

Unfortunately, the contents of `fs/ecryptfs` are identical what is already on the branch aosp/LA.UM.5.7.r1 (so this defect exists in the Pixel 2 source as well).

It looks like Qualcomm added ICE support to ecryptfs.  They likely verified that ecryptfs with ICE works fine.  However, normal (i.e. non-ICE) ecryptfs operations are broken (and perhaps they do not care about that).

jamuir, 2017-10-27:
I will posted a message about this defect on the linux-arm-msm mailing list.

bartcubbins, 2017-11-28:
@jamuir did you still have the problem with ecryptfs?

jamuir, 2017-11-29:
https://github.com/sonyxperiadev/bug_tracker/issues/49 is a defect related to device encryption; it is not related to the current issue, which involves the ecryptfs file system.

closed 2018-02-16 by jerpelea