sonyxperiadev/kernel #1519: Precompiled kernel

msaw, 2017-11-18:
Hello,
I follow the instructions described on https://developer.sonymobile.com/open-devices/aosp-build-instructions/how-to-build-aosp-nougat-for-unlocked-xperia-devices/  but the boot image does not contain the precompiled kernel.

I tried also to to put:
```
ifeq ($(TARGET_PREBUILT_KERNEL),)
   LOCAL_KERNEL := device/sony/aries/kernel
else
   LOCAL_KERNEL:= $(TARGET_PREBUILT_KERNEL)
endif
PRODUCT_COPY_FILES := \
                      $(LOCAL_KERNEL):kernel
```
in the kernel/sony/msm/Makefile and export TARGET_PREBUILT_KERNEL=my/sony/zImage, but without success.
 
Anyone have a idea?

Thank you in advance.

PS: I try to build a boot image for sony xperia Z3 compact

jerpelea, 2017-11-18:
what android veraion do you try to bulit ?

msaw, 2017-11-18:
Nougat 7.1

jerpelea, 2017-11-18:
kernel 3.10 is provided for your device!

msaw, 2017-11-18:
Thank you for your quick response.
I know but I need some features that's not exist in the default kernel config!

jerpelea, 2017-11-18:
try the default one then we can start from there

msaw, 2017-11-18:
the default one work without problems! But I need just for an example the /proc/kallsyms that doesn't exist in the default Kernel!

jerpelea, 2017-11-18:
how do you build the kernel?

msaw, 2017-11-18:
I use the branch LA.BR.1.3.3_rb2.14 for the kernel:
```
$ make aosp_shinano_aries_defconfig
$ make xconfig # enable some features
$ make -j<nb_core> # the zImage was built successfuly 
```
I put the zImage in the folder  device/sony/aries in the AOSP directory as described in the guide (https://developer.sonymobile.com/open-devices/how-to-build-and-flash-a-linux-kernel/how-to-build-and-flash-a-linux-kernel-for-aosp-supported-devices/) and the arch/arm/boot/dts/msm8974pro-ac-shinano_aries.dtb to the folder device/sony/aries/dtbs (which doesn't exist by default). 

When I saw that doesn't wok I tried the alternative that I described in the first message.

Thank you @jerpelea.

closed 2017-11-19 by msaw