sonyxperiadev/kernel #1527: mdss is broken

hzj5158888, 2017-12-04:
I build the kernel and flash into the device but doesn"t boot
log:
<3>[   11.279262] mdss_mdp_pipe_destroy: unable to free pipe 3 while still in use
<4>[   11.279268] ------------[ cut here ]------------
<4>[   11.279277] WARNING: at /root/github/sony/drivers/video/msm/mdss/mdss_mdp_pipe.c:745 mdss_mdp_pipe_unmap+0x98/0xb0()
<4>[   11.279278] Unexpected free pipe during unmap
<4>[   11.279282] CPU: 3 PID: 285 Comm: mdss_fb0 Not tainted 3.10.84-g0a592d9-dirty #10
<4>[   11.279297] [<c00139c0>] (unwind_backtrace+0x0/0x120) from [<c0011c50>] (show_stack+0x10/0x14)
<4>[   11.279306] [<c0011c50>] (show_stack+0x10/0x14) from [<c002dec4>] (warn_slowpath_common+0x48/0x68)
<4>[   11.279313] [<c002dec4>] (warn_slowpath_common+0x48/0x68) from [<c002df3c>] (warn_slowpath_fmt+0x2c/0x3c)
<4>[   11.279319] [<c002df3c>] (warn_slowpath_fmt+0x2c/0x3c) from [<c026e634>] (mdss_mdp_pipe_unmap+0x98/0xb0)
<4>[   11.279326] [<c026e634>] (mdss_mdp_pipe_unmap+0x98/0xb0) from [<c028da38>] (mdss_mdp_overlay_pan_display+0x1d4/0x2fc)
<4>[   11.279336] [<c028da38>] (mdss_mdp_overlay_pan_display+0x1d4/0x2fc) from [<c02cdebc>] (__mdss_fb_display_thread+0x2c0/0x430)
<4>[   11.279346] [<c02cdebc>] (__mdss_fb_display_thread+0x2c0/0x430) from [<c004c870>] (kthread+0xdc/0xf0)
<4>[   11.279353] [<c004c870>] (kthread+0xdc/0xf0) from [<c000e3e0>] (ret_from_fork+0x14/0x34)
<4>[   11.279355] ---[ end trace 408065a3984296bc ]---

hzj5158888, 2017-12-04:
branch : LA.BR.1.3.3_rb2.14
device : togari

bartcubbins, 2017-12-04:
@hzj5158888 provide the full log please

hzj5158888, 2017-12-05:
@bartcubbins full log:
<6>[    4.375860] , IO base, CE = 0xc66c0000
<6>[    4.375860] , Consumer (IN) PIPE 4,    Producer (OUT) PIPE 5
<6>[    4.375860] IO base BAM = 0x  (null)
<6>[    4.375860] BAM IRQ 268
<6>[    4.375860] Engines Availability = 0x1010853
<6>[    4.400339] QCE50: qce_sps_init:  Qualcomm MSM CE-BAM at 0x00000000fd444000 irq 268
<6>[    4.407728] qcrypto qcrypto.0: qcrypto-ecb-aes
<6>[    4.412039] qcrypto qcrypto.0: qcrypto-cbc-aes
<6>[    4.416512] qcrypto qcrypto.0: qcrypto-ctr-aes
<6>[    4.420912] qcrypto qcrypto.0: qcrypto-ecb-des
<6>[    4.425365] qcrypto qcrypto.0: qcrypto-cbc-des
<6>[    4.429767] qcrypto qcrypto.0: qcrypto-ecb-3des
<6>[    4.434305] qcrypto qcrypto.0: qcrypto-cbc-3des
<6>[    4.438789] qcrypto qcrypto.0: qcrypto-xts-aes
<6>[    4.443249] qcrypto qcrypto.0: qcrypto-sha1
<6>[    4.447391] qcrypto qcrypto.0: qcrypto-sha256
<6>[    4.451725] qcrypto qcrypto.0: qcrypto-aead-hmac-sha1-cbc-aes
<6>[    4.457482] qcrypto qcrypto.0: qcrypto-aead-hmac-sha1-cbc-des
<6>[    4.463208] qcrypto qcrypto.0: qcrypto-aead-hmac-sha1-cbc-3des
<6>[    4.468994] qcrypto qcrypto.0: qcrypto-aead-hmac-sha256-cbc-aes
<6>[    4.474956] qcrypto qcrypto.0: qcrypto-aead-hmac-sha256-cbc-des
<6>[    4.480805] qcrypto qcrypto.0: qcrypto-aead-hmac-sha256-cbc-3des
<6>[    4.486824] qcrypto qcrypto.0: qcrypto-hmac-sha1
<6>[    4.491401] qcrypto qcrypto.0: qcrypto-hmac-sha256
<6>[    4.496195] qcrypto qcrypto.0: qcrypto-aes-ccm
<6>[    4.500593] qcrypto qcrypto.0: qcrypto-rfc4309-aes-ccm
<6>[    4.505629] qcrypto: FIPS140-2 Known Answer Tests: Skipped
<6>[    4.511773] SCSI Media Changer driver v0.25 
<6>[    4.516272] tun: Universal TUN/TAP device driver, 1.6
<6>[    4.520364] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>
<6>[    4.530985] PPP generic driver version 2.4.2
<6>[    4.534395] PPP BSD Compression module registered
<6>[    4.538899] PPP Deflate Compression module registered
<6>[    4.543964] PPP MPPE Compression module registered
<6>[    4.548708] NET: Registered protocol family 24
<6>[    4.553465] wcnss_wlan probed in built-in mode
<6>[    4.558991] NET: Registered protocol family 27
<6>[    4.559152] msm_sharedmem: msm_sharedmem_probe: Device created for client 'rmtfs'
<6>[    4.559284] msm_sharedmem: msm_sharedmem_probe: Device created for client 'rfsa_dsp'
<6>[    4.559414] msm_sharedmem: msm_sharedmem_probe: Device created for client 'rfsa_mdm'
<4>[    4.559528] msm_sharedmem: msm_sharedmem_probe: qcom,client-id property not found
<6>[    4.559638] msm_sharedmem: msm_sharedmem_probe: Device created for client 'bimc'
<3>[    4.566577] msm-dwc3 msm_dwc3: pget_irq for pwr_event_irq failed
<3>[    4.566783] Can't call dbm_l1_lpm_interrupt, dbp pointer == NULL
<6>[    4.566792] msm-dwc3 msm_dwc3: MHL switch is not enabled
<3>[    4.617678] IPC_RTR: ipc_router_create_log_ctx: Unable to create IPC logging for [local_IPCRTR]
<6>[    4.626122] msm_sharedmem: sharedmem_register_qmi: qmi init successful
<6>[    4.682518] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
<6>[    4.684991] msm-dwc3 msm_dwc3: DWC3 in low power mode
<6>[    4.693060] ehci-msm: Qualcomm On-Chip EHCI Host Controller
<6>[    4.698872] ehci-msm2: Qualcomm EHCI Host Controller
<6>[    4.704334] usbcore: registered new interface driver usb-storage
<6>[    4.710654] mousedev: PS/2 mouse device common for all mice
<6>[    4.715458] max1187x 0-0048: (INIT): Start
<6>[    4.717218] qcom,qpnp-rtc 00-qcom,pm8941_rtc: rtc core: registered qpnp_rtc as rtc0
<6>[    4.717402] i2c /dev entries driver
<6>[    4.718539] vidc: disabling clock-voltage handling
<6>[    4.720249] platform fdc00000.qcom,vidc: Driver msm_vidc_v4l2 requests probe deferral
<6>[    4.720440] vidc: disabling clock-voltage handling
<6>[    4.720444] msm_vidc:  err: Failed to determine dcvs min load: -22
<6>[    4.720447] msm_vidc:  err: Failed to determine dcvs min MB per frame: -22
<6>[    4.730550] CAM-SMMU cam_populate_smmu_context_banks:1054 getting QSMMU ctx : vfe
<6>[    4.730846] CAM-SMMU cam_populate_smmu_context_banks:1054 getting QSMMU ctx : vfe_secure
<6>[    4.731133] CAM-SMMU cam_populate_smmu_context_banks:1054 getting QSMMU ctx : cpp
<6>[    4.731439] CAM-SMMU cam_populate_smmu_context_banks:1054 getting QSMMU ctx : jpeg_enc0
<6>[    4.731743] CAM-SMMU cam_populate_smmu_context_banks:1054 getting QSMMU ctx : jpeg_enc1
<6>[    4.732042] CAM-SMMU cam_populate_smmu_context_banks:1054 getting QSMMU ctx : jpeg_dec
<3>[    4.732945] cci_enable_i2c_vcc: Failed to get vcc_i2c regulator
<3>[    4.732952] msm_cci_get_clk_info: clock-names[0][0] = camss_top_ahb_clk
[... 325 lines cut ...]
<13>[    9.969454] ueventd: Coldboot took 1.35s.
<6>[   10.143001] reset disable: pinctrl not enabled
<6>[   10.159773] reset enable: pinctrl not enabled
<6>[   10.414004] mdss_dsi_panel_disp_on: ctrl=f6256010 ndx=0
<6>[   10.436500] mdss_dsi_panel_off: ctrl=f6256010 ndx=0
<6>[   10.733001] reset disable: pinctrl not enabled
<6>[   10.735376] max1187x 0-0048: set_suspend_mode
<6>[   10.756968] reset enable: pinctrl not enabled
<5>[   10.942396] random: nonblocking pool is initialized
<6>[   11.008125] mdss_dsi_panel_disp_on: ctrl=f6256010 ndx=0
<6>[   11.008156] max1187x 0-0048: set_resume_mode
<3>[   11.029150] mdss_mdp_pipe_destroy: unable to free pipe 3 while still in use
<4>[   11.029156] ------------[ cut here ]------------
[... cut ...]
<4>[   11.029213] [<c026e668>] (mdss_mdp_pipe_unmap+0x98/0xb0) from [<c028da88>] (mdss_mdp_overlay_pan_display+0x1d4/0x2fc)
<4>[   11.029223] [<c028da88>] (mdss_mdp_overlay_pan_display+0x1d4/0x2fc) from [<c02ce000>] (__mdss_fb_display_thread+0x2c0/0x430)
<4>[   11.029232] [<c02ce000>] (__mdss_fb_display_thread+0x2c0/0x430) from [<c004c870>] (kthread+0xdc/0xf0)
<4>[   11.029239] [<c004c870>] (kthread+0xdc/0xf0) from [<c000e3e0>] (ret_from_fork+0x14/0x34)
<4>[   11.029242] ---[ end trace 70590e96b4929284 ]---

hzj5158888, 2017-12-09:
How to fix it?

closed 2017-12-23 by hzj5158888