sonyxperiadev/kernel #1693: [aosp/LE.UM.2.3.2.r1.4] init/Kconfig:340: can't open file "kernel/irq/Kconfig"

MartinX3, 2018-07-24:
System Ubuntu 18.04 / Linux Mint 19

I try to compile the XZ2 Akari kernel, but fail with this error:

```bash
$ make ARCH=arm64 CROSS_COMPILE=$CROSS_COMPILE aosp_tama_akari_defconfig
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
init/Kconfig:340: can't open file "kernel/irq/Kconfig"
scripts/kconfig/Makefile:112: recipe for target 'aosp_tama_akari_defconfig' failed
make[1]: *** [aosp_tama_akari_defconfig] Error 1
Makefile:544: recipe for target 'aosp_tama_akari_defconfig' failed
make: *** [aosp_tama_akari_defconfig] Error 2
```

Anyone know how to solve this problem?

nathanchance, 2018-07-24:
I would try specifying an out folder

`make ARCH=arm64 CROSS_COMPILE=$CROSS_COMPILE O=out aosp_tama_akari_defconfig`

MartinX3, 2018-07-24:
Thank you for your fast response.

But sadly stil lthe same error

```bash
$ make mrproper
  CLEAN   scripts/basic
  CLEAN   scripts/kconfig

#### build completed successfully (8 seconds) ####

$ make ARCH=arm64 CROSS_COMPILE=$CROSS_COMPILE o=out aosp_tama_akari_defconfig
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
init/Kconfig:340: can't open file "kernel/irq/Kconfig"
scripts/kconfig/Makefile:112: recipe for target 'aosp_tama_akari_defconfig' failed
make[1]: *** [aosp_tama_akari_defconfig] Error 1
Makefile:544: recipe for target 'aosp_tama_akari_defconfig' failed
make: *** [aosp_tama_akari_defconfig] Error 2

#### failed to build some targets (2 seconds) ####
```

kholk, 2018-07-24:
This seems to be an issue with your own environment variables and/or your specific GNU Make version.

Please make sure that you are using stable packages from your distribution and that you did not apply custom configuration to GNU Make (the suspect comes from the unusual Android build system style build success/failure messages).

MartinX3, 2018-07-24:
Thx for your response and sorry, the next issue will be in the issue tracker.

My GNU make version is the standard one shipped with Ubuntu 18.04 Bionic
"4.1-9.1ubuntu1"

And I did not change the GNU make settings.
I made a REPO init and REPO sync and used a fresh downloaded cross compiler.

Could it be the new Ubuntu 18.04?

Is there  a docker image or another kind of working build environment configuration I could install and test?

YumeMichi, 2018-07-25:
No build errors on archlinux.
https://i.imgur.com/rvOy5A0.png

MartinX3, 2018-07-25:
Hmm, maybe someone has the same problem on another Ubuntu 18.04?
Then something is in Ubuntu or in this Code.

If not, I did something wrong.
But I don't know what.

ab123321, 2018-07-25:
What's the creation time of that `kernel/irq/Kconfig`? I have got similar error with files that had creation date of 01-01-1970, GNU make simply couldn't find them for some reason

MartinX3, 2018-07-25:
Sadly the date of the akari/kernel/kernel/irq/Kconfig file is fine. :(

MartinX3, 2018-07-25:
Thank you! A shorter path fixed this problem.
According to
https://developer.sony.com/develop/open-devices/guides/kernel-compilation-guides/how-to-build-and-flash-a-linux-kernel-for-aosp-supported-devices
and
https://developer.sony.com/develop/open-devices/guides/aosp-build-instructions/build-aosp-nougat-8-1-oreo-4-4
I did the steps, but if I run "make" or "make bootimage" I get the following error:

```bash
martin@deus-ex-machina:/media/martin/extLinux/Develop/android$ make -j 8
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=8.1.0
TARGET_PRODUCT=aosp_h8266
TARGET_BUILD_VARIANT=eng
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=generic
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a9
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.15.0-29-generic-x86_64-with-LinuxMint-19-tara
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=OPM6.171019.030.E1
OUT_DIR=out
============================================
ninja: no work to do.
ninja: no work to do.
out/target/product/akari/clean_steps.mk was modified, regenerating...
[139/1004] including ./device/sony/common/Android.mk ...
hardware/qcom/media/sdm845/mm-video-v4l2/vidc/venc/Android.mk:45: warning: "venc: PQ compiled out"
/bin/bash: Zeile 0: pushd: out/target/product/akari/odm/: Datei oder Verzeichnis nicht gefunden
[675/1004] including ./system/sepolicy/Android.mk ...
./system/sepolicy/Android.mk:111: warning: BOARD_SEPOLICY_VERS not specified, assuming current platform version
[681/1004] including ./system/update_engine/Android.mk ...
./system/update_engine/Android.mk:504: warning: No PRODUCT_STATIC_BOOT_CONTROL_HAL configured but AB_OTA_UPDATER is true, no update sideload support.
[1004/1004] including ./vendor/qcom/opensource/wlan/wcnss-service/Android.mk ...
PRODUCT_COPY_FILES device/generic/goldfish/data/etc/apns-conf.xml:system/etc/apns-conf.xml ignored.
build/core/Makefile:54: warning: Duplicate header copy: out/target/product/akari/obj/include/qcom/display/qd_utils.h
build/core/Makefile:54: warning: Defined in: hardware/qcom/display/sdm845/libqdutils/../common.mk hardware/qcom/display/sdm845/libqdutils/../common.mk
No private recovery resources for TARGET_DEVICE akari
platform_testing/build/tasks/tests/instrumentation_metric_test_list.mk: warning: continuous_instrumentation_metric_tests: Unknown installed file for module perf-setup.sh 
platform_testing/build/tasks/tests/instrumentation_test_list.mk: warning: continuous_instrumentation_tests: Unknown installed file for module RecyclerViewTests 
platform_testing/build/tasks/tests/instrumentation_test_list.mk: warning: continuous_instrumentation_tests: Unknown installed file for module SettingsFunctionalTests 
platform_testing/build/tasks/tests/instrumentation_test_list.mk: warning: continuous_instrumentation_tests: Unknown installed file for module LauncherFunctionalTests 
platform_testing/build/tasks/tests/instrumentation_test_list.mk: warning: continuous_instrumentation_tests: Unknown installed file for module EmergencyInfoTests 
platform_testing/build/tasks/tests/native_metric_test_list.mk: warning: continuous_native_metric_tests: Unknown installed file for module perf-setup.sh 
build/core/Makefile:34: warning: overriding commands for target `out/target/product/akari/vendor/etc/gps.conf'
build/core/base_rules.mk:390: warning: ignoring old commands for target `out/target/product/akari/vendor/etc/gps.conf'
ninja: error: 'kernel/sony/msm-4.9/common-kernel/kernel-dtb-akari', needed by 'out/target/product/akari/kernel', missing and no known rule to make it
23:30:52 ninja failed with: exit status 1

#### failed to build some targets (41 seconds) ####
```

https://github.com/sonyxperiadev/vendor-sony-kernel/tree/aosp/LE.UM.2.3.2.r1.4
only contains a single .mk file.

So I assume, that it is too early to compile the 4.9 kernel and I should wait?
If this is right, could you give me an ETA? :)

YumeMichi, 2018-07-26:
4.4 kernel? 4.9 kernel?
```
error: 'kernel/sony/msm-4.4/kernel/arch/arm64/configs/aosp_tama_akari_defconfig'
```

MartinX3, 2018-07-26:
@YumeMichi I choose only ARM64 and the tama defconfig.
Tama is the XZ2 and XZ2C Plattform and the first with Kernel 4.9

YumeMichi, 2018-07-26:
@MartinX3 Why your kernel path for tama akari is kernel/sony/msm-4.4?
https://github.com/sonyxperiadev/local_manifests/blob/master/LE.UM.2.3.2.r1.4.xml#L6

kholk, 2018-07-26:
The 4.9 kernel is not ready yet for Tama.
Only recovery currently works, you can boot to Android if you hack the security, but the display hal currently crashes. 

If you want to go on, you can generate the dtb image and put it in that folder, that way your build will succeed. 

If you're not an experienced developer, I'd consider waiting for us to fix the last issues before trying to boot it!

MartinX3, 2018-07-26:
@kholk 
Good to know, thank you!

I only want to compile the dtb for the recovery image.
But "make -j8 recoveryimage" throws that error, too.
Do yu know where I can get a manual how to generate only the dtb image?

closed 2018-07-26 by MartinX3

kholk, 2018-07-27:
You're welcome!