sonyxperiadev/kernel #1838: Support user-supplied signing keys for Verified Boot.

gerg5c42g542g2c54g52c, 2018-12-03:
I assume Verified Boot isn't supported by Sony's AOSP build instructions.

This issue is about letting user use their own signing keys for Verified Boot to (greatly) aid in security.

Let me know if it's supported with vendor-enforced (Sony's keys) or user-supplied keys.

ix5, 2018-12-03:
Take a look at https://github.com/sonyxperiadev/device-sony-tone/blob/a04bad0770ecd80f8f5c991d3907372667070501/platform.mk#L165

The key will be in `build/make/target/product/security/verity.pk8`, or in Sony Open devices' case, `vendor/oss/release-keys`(those are obviously not the same keys Sony uses for their official builds).

But that is just dm-verity. You will still get the message about your unlocked bootloader at boot, and anyone will be able to tamper with your device via fastboot. AFAIK there is no way to enroll your own keys with the bootloader and re-locking it like you can with UEFI Secure Boot.

kholk, 2018-12-03:
The device is encrypted by default on Open Devices and as @ix5 said it's possible to change your keys for Verity.

Unfortunately, though, the bootloader doesn't support relocking with custom security keys (so there is no Pixel-style relocking).

jerpelea, 2018-12-04:
there is no implementation that lets users relock the bootloader using theyr own keys 

in the feature please use the bugtraker git to post questions

closed 2018-12-04 by jerpelea

gerg5c42g542g2c54g52c, 2018-12-04:
So, just to be sure. The bootloader can be re-locked, but only with Sony provided keys?
Does this mean private keys are provided by Sony to enable users to sign their builds for Verified Boot? Isn't this allowing anyone with access to https://developer.sony.com to create malicious builds and flash them to other people's phones without unlocking the bootloader and therefore completely pointless?

@jerpelea please re-open the issue. I obviously still want this implemented as it's one of the most useful security features. I'm sure no one doubts this is needed.

jerpelea, 2018-12-04:
this feature is not available and there is no way to lock the bootloader

gerg5c42g542g2c54g52c, 2018-12-24:
Can it be implemented?

jerpelea, 2018-12-24:
no it is not available on current devices