sonyxperiadev/kernel #1854: Excessive echo for VoIP call on Xperia X

XperiaDeveloper, 2018-12-14:
Within our VoIP-telephoning-application we encounter excessive echo with 
Sony Experia X under the following conditions:

1. Voice-Call using both of the phones in handset-mode

   * Note: If the playback-volume is turned down below 33% on both devices
           (which usually is just barely enough for a voice-call) the echo
           is reduced to a "just acceptable" level on the other-end phone. 
           This makes it apparent that the echo is produced by the air-gap-
           transmission in between the device's (handset-)loudspeaker and 
           it's microphone.

2. Voice-Call using both of the phones in handsfree-mode

   * Note: In this case we encounter excessive echo (up to the level where
           the echo due to massive feedback reaches self-oscillation). In
           this case it is not possible (unlike in case 1) to reduce echo
           to a sensible level where it doesn't lead to a bad user-experi-
           ence. Neither end party reaches understandable audio in this
           case.

3. Voice-Call using one of the phones in handsfree and the other in handset-
   mode

   * Note: This case is somewhat in between cases 1 and 2, for obvious rea-
           sons. If the playback-volume on the handset-party-end is taken
           down to approx. 20% a phone-call is possible but echo is still
           excessive.

We have this effect on Sony-AOSP, only. We do not encounter this effect of
excessive echo when using stock Sony firmware(s) [or other stock Android-
devices (eg. Huawei P20lite, Samsung Galaxy S4,...)]. From our side it looks
like this effect is caused by the hardware-DSP-AEC the devices are capable
of is turned off on AOSP.

We use PjSua2 as basis for the VoIP-capablilites of our app and this effect
can be reproduced by their Android-Example-Code, too. The same effect can
be reproduced on Sony Experia Z3 AOSP vs. stock-ROM.

This was tested on: Sony Experiy Z
                    AOSP 8.1.0, 
                    Dec 5, 2018, 
                    aosp_f5121-userdebug 8.1.0 
                    OPM7.181205.001 
                    eng_aosp.20181207.122338 
                    dev-keys

jerpelea, 2018-12-14:
please push the bugs on 
https://github.com/sonyxperiadev/bug_tracker/issues

closed 2018-12-14 by jerpelea

XperiaDeveloper, 2018-12-14:
moved to https://github.com/sonyxperiadev/bug_tracker/issues/264