sonyxperiadev/kernel #1879: Upstream to mainline Linux Kernel

alistair23, 2019-01-18:
This effort looks really cool! I wanted to check though if there are plans on upstreaming this work to the mainline Linux kernel?

jerpelea, 2019-01-22:
@alistair23 on a best effort we are submitting code upstream to Android and kernel as the gap closes between our fork and upstream. For some components the gap is too big and for now is not possible to reduce it.

kholk, 2019-01-22:
Hey.

I am glad to inform you that I am currently trying to upstream support for the platforms that we currently support on 4.9.
Regarding this, I've already sent some patches (GCC and RPMCC) upstream for MSM8956/76, even though my time is a lot limited, I'm trying to do my best on that matter: the mainline kernel already has a lot of support for QC SoCs and we can probably get good things from that.

If you mean to help me/us in that, please feel free to!

--Angelo

alistair23, 2019-01-22:
Yeah! I'm happy to help where I can upstream to the mainline Linux kernel (not the Android fork).

I recently bought a XA2 because of the Sailfish X support and now I'm curious to see if I can run pmOS with libhybris. Do you have a current status of what is/isn't upstream and any info for testing a mainline kernel?

closed 2020-01-22 by jerpelea