sonyxperiadev/kernel #1997: Latest update on LA.UM.6.4.r1 broke compilation

Thaodan, 2019-06-01:
The latest update to LA.UM.6.4.r1 brought the commit:
e5d7620af2fd50b50c3a37cd7a27e366e897747

Which results in
```
../../../../../../kernel/sony/msm-4.4/kernel/drivers/platform/msm/ipa/ipa_v3/ipa_rt.c: In function 'ipa3_reset_rt':
../../../../../../kernel/sony/msm-4.4/kernel/drivers/platform/msm/ipa/ipa_v3/ipa_rt.c:1473:13: error: 'struct ipa3_rt_entry' has no member named 'rule_id_valid'
    if (!rule->rule_id_valid)
             ^
make[6]: *** [drivers/platform/msm/ipa/ipa_v3/ipa_rt.o] Error 1
make[6]: *** Waiting for unfinished jobs....

```
The issue is that /rule_id_valid/ was only added in:
https://github.com/sonyxperiadev/kernel/commit/d0c2a1ebcecc549d1803787c1c9d78f3ed308a9e

Which is not inside this true.
So what should happen revert the commit or apply depended commit?

jerpelea, 2019-06-03:
please add the missing commit and PR

Thaodan, 2019-06-03:
Missing commit?
The commit thats the issue is mentioned above and the other pr is not part of 
this bug,  I discovered it afterwards.

Thaodan, 2019-06-03:
Currently I reverted the mentioned and the kernel builds fine now.
I just asked if reverting or porting the depended commit is better.

kholk, 2019-06-03:
@Thaodan Revert the commit. It's the best choice since k4.4 is in feature freeze since... A very long time ago.
The commit that introduces the fix also introduces a new feature, which will never be used anyway ;)

Thaodan, 2019-06-03:
Ok, I already did tat. I'll make a pr on that later.

jerpelea, 2019-06-03:
Please rever the commit 
the branch is feature freeze

Thaodan, 2019-06-04:
This depends also partly #1998.

Thaodan, 2019-06-14:
Fixed by: https://github.com/sonyxperiadev/kernel/commit/08932cbccbf7faaa049755da5ad7ae6aabcccba7

PR #1999 (2019-06-04): LA.UM.6.4.r1: Revert "msm:ipa:Prevent rt rule deletion if rt rule id is invalid"
- This reverts commit e5d7620af2fd50b50c3a37cd7a27e366e8977477
- Fixes #1997 :: The feature mentioned isn't aviable anyway in 4.4.x
commit eb590318 (2019-06-04): Revert "msm:ipa:Prevent rt rule deletion if rt rule id is invalid"

- This reverts commit e5d7620af2fd50b50c3a37cd7a27e366e8977477
- Fixes #1997 :: The feature mentioned isn't aviable anyway in 4.4.x
closed 2019-06-14 by Thaodan