sonyxperiadev/kernel #2054: LA.UM.7.1.r1 for ganges ?

Haxk20, 2019-09-11:
I see more and more devices getting added to 4.14 branch so i wanted to ask if ganges platform will get this upgrade too or we will be stuck with 4.9 ?
Thank you.

MarijnS95, 2019-09-11:
@Haxk20 No device is "really" on 4.14 yet, but we're almost there. Not to worry, all platforms and devices will go to 4.14 - it's just a matter of time to port configurations, fix issues and patching up the HALs to be compatible.

jerpelea, 2019-09-12:
If all is ok I will add ganges and nile defconfigs today after all 5 devices are tested

Haxk20, 2019-09-12:
> If all is ok I will add ganges and nile defconfigs today after all 5 devices are tested

Ahh OK. Thank you very much.
Closing the "issue".

closed 2019-09-12 by Haxk20