sonyxperiadev/kernel #2061: techpack/audio/Kconfig missing !

LinuXperia, 2019-09-16:
Hi!

This kernel can not be compiled using as a example the 

> arch/arm64/configs/aosp_nile_voyager_defconfig

Kernel Configuration as it has not all Files Completed !

Everytime the kernel try to load 
techpack/Kconfig

but this file just refers to a non existent FIle in the Tree

> source techpack/audio/Kconfig

Please upload the missing files or fix the problem !

MarijnS95, 2019-09-16:
Please use `repo` to sync this _separate repository_! (same for `wlan-qc` drivers)
https://github.com/sonyxperiadev/local_manifests/blob/master/LA.UM.7.1.r1.xml#L7

@jerpelea How about adding a `.gitmodules` to track the (now 4, or 5 if you count the optional defconfig repository) external repositories?

LinuXperia, 2019-09-16:
Ahhh thanks for the Tip.

I am building the Kernel using buildroot as described in the Official
Documentation.

I dont see anything there about repo sync mentioned !

[developer.sony.com: Build basic Linux environment for development – Buildroot for Xperia](https://developer.sony.com/develop/open-devices/guides/kernel-compilation-guides/build-basic-linux-environment-for-development-buildroot-for-xperia#tutorial-step-1)

How this can be solved using buildroot or other build environments ?

jerpelea, 2019-09-16:
lets keep them separate for now

closed 2020-01-22 by jerpelea