sonyxperiadev/kernel #2180: Why is there no

sorryusernameisalreadytaken, 2020-02-05:
I ask myself why is there no suzuran defconfig?
The ota is running out and we had problems building new roms without official sources.

Isn't Sony able to provide some code to prevent the Sony Xperia 5 (compact/premium/normal) from getting sorted out and land in the garbage?

jerpelea, 2020-02-05:
unfortunately not for the latest branch but you can use the LA.BR.1.3.3_rb2.14 branch for your Android build

sorryusernameisalreadytaken, 2020-02-05:
THX for that. Sadly is it quite old :)

Can you tell me the reason why there is nothing for the latest branch?

jerpelea, 2020-02-05:
because it is old and does not have the needed blobs

oshmoun, 2020-02-05:
the z3 is definitely not supported on the latest branch. you are probably mixing up xz3 and z3 :)

sorryusernameisalreadytaken, 2020-02-05:
> 
> 
> the z3 is definitely not supported on the latest branch. you are probably mixing up xz3 and z3 :)

oh yeah you right.

> 
> 
> because it is old and does not have the needed blobs

Ok, it is just development time for such an old device. That's a pity, because especially these need some support in the last moves to be able to be maintained by free developers.

jerpelea, 2020-02-05:
@sorryusernameisalreadytaken you can use the the branch I specified for your project since it has compatible blobs released on our website

closed 2020-02-05 by jerpelea