sonyxperiadev/kernel #2194: Sony Xperia 5 AOSP

Lucie013, 2020-02-16:
Hello all, excuse my bad English.

I have a phone Sony Xperia 5 (J9210), and I would like to install Android 10 AOSP into it (pure android without bloatware from Sony .etc)

I found this official guide:
[CODE]https://developer.sony.com/develop/open-devices/get-started/unlock-bootloader/how-to-unlock-bootloader
https://developer.sony.com/develop/open-devices/guides/aosp-build-instructions/build-aosp-android-android-10-0-0[/CODE]

I have 4 questions:

1, How can I make a backup of a ROM if I want to go back to the original ROM?

2, If someone have experience with build ROM for SONY, are there any known problems what won't work for me, example wifi, camera, bluetooth .etc?

3, Will download Android auto-updates work? or do I have to rebuild ROM with new updates every time?

4. Will this build AOSP contain translate Czech ?

Thanks for help me :o

MartinX3, 2020-02-16:
1. XDA is a good place, this is an issue tracker you shouldn't use at all
In this repo is one for SODP specific issues (but which doesn't include questions how to backup your phone :P)

2. The mentioned bug tracker is a good place
If you find a new issue, please add it there.

3. Of course not, someone needs to pay servers for a updater infrastructure including rom image building
If you want to create/donate one, you're welcome ^^

4. If the normal AOSP does, the chances are high, that you have Czech as well here.
It is not a language introduced into the world last year.

Haxk20, 2020-02-16:
@Lucie013 
1. Jednoducho povedané nemôžete. Jedna možnosť je tam dať stock ale data sú preč.
2.[Bug tracker](https://github.com/sonyxperiadev/bug_tracker/issues)
3. Rebuild.
4. Čeština ? Ano ;)

Lucie013, 2020-02-17:
@MartinX3 first I wrote on XDA forum (twice), no answer, then I wrote here.
I thought if there would be a minimal security auto-update from google (when I switch to AOSP) but if I understand correctly you say no.

@Haxk20
Ahoj moc děkuji za odpodět,

Moc tomu nerozumím, stock ROM je vydání (neoficiální vydaní komunitou) který obsahuje původní ROM od výrobce, pro lidi co si třeba koupili mobil od operátora a ten jim tam přidal své aplikace (Nebo něco změnil) a tímto způsobem se vrátí k původnímu ROM?

Jinak, nemáš Discord že bych ti napsala a poprosila o pomoc :-D

ix5, 2020-02-17:
There are "auto-updates" for the AOSP source, but you still need to build them yourself.

This project gives you a recipe, and if the recipe changes, you need to cook the soup yourself.

The only precompiled update you are going to get from Google are [APEX modules](https://www.xda-developers.com/android-q-apex-biggest-thing-since-project-treble/), but that's a big piece of vapourware...

Haxk20, 2020-02-19:
> @MartinX3 first I wrote on XDA forum (twice), no answer, then I wrote here.
> I thought if there would be a minimal security auto-update from google (when I switch to AOSP) but if I understand [...]

Stock ROM je ROM priamo od výrobcu.
Telecom alebo O2 ... iba pridáva veci.
Discord moc nepoužívam.
Ale Guide má všetko čo je potrebné.

kholk, 2020-03-01:
Rule #1: English.
Rule #2: Use the bug tracker. 

Both rules broken, congratulations.

Haxk20, 2020-03-01:
@kholk Sorry for breaking the rule 1.
@jerpelea Can you please close this ? Also possibly disable the issues on repo's other then bug tracker ?

closed 2020-03-02 by jerpelea