sonyxperiadev/kernel #2274: techpack/audio/kconfig something in config is wrong

winabs, 2020-05-31:
techpack/Kconfig:1: can't open file "techpack/audio/Kconfig"
make[1]: * [scripts/kconfig/Makefile:113: aosp_tone_dora_defconfig] Error 1
make: * [Makefile:534: aosp_tone_dora_defconfig] Error 2

winabs, 2020-05-31:
Please fix for Sony X and XZ

kholk, 2020-05-31:
You have to clone the techpacks gits to the proper folders. This is not a bug.

winabs, 2020-05-31:
I have to clone which one?

MartinX3, 2020-05-31:
I assume you search this
https://www.github.com/sonyxperiadev/local_manifests

ix5, 2020-05-31:
Follow [the Kernel compilation guide](https://developer.sony.com/develop/open-devices/guides/kernel-compilation-guides/how-to-build-and-flash-a-linux-kernel-for-aosp-supported-devices) to the letter and you will not have the issues you describe.

For your benefit, the issue lies in the fact that you are trying to compile a 4.14 kernel, which relies on git submodules.
The submodules and their location can be found in the [local_manifests xml file for the 4.14 kernel](https://github.com/sonyxperiadev/local_manifests/blob/2061f1b0129d2550e8318c05337e3dd730a5a958/LA.UM.7.1.r1.xml#L6-L12).

But your devices are not supported on 4.14 yet: Both Xperia X ("suzu") on the loire platform as well as Xperia XZ ("kagura") on the tone platform are only supported on 4.9.

Please close this issue.

ix5, 2020-05-31:
@jerpelea you might want to turn off the "Issues" functionality for this repo altogether, and redirect people to the common bug tracker.

closed 2020-06-01 by jerpelea