sonyxperiadev/kernel #2281: Kagura

winabs, 2020-06-04:
make aosp_tone_kagura_defconfig                     HOSTCC  scripts/basic/bin2c
backports/compat/Kconfig:57: warning: ignoring unknown option bp_modules#
# configuration written to .config
#
siswa@siswa:~/kernel$ make -j$(nproc --all)                             arch/arm64/Makefile:67: *** CROSS_COMPILE_ARM32 not defined or empty, the compat vDSO will not be built.  Stop.

ix5, 2020-06-04:
Please use the [issue tracker](https://github.com/sonyxperiadev/bug_tracker/issues/) for reporting any issues.

There, you will also be directed to fill out a form with more information. As it stands, we sadly cannot help you with your issue because we can only guess which configuration you are using to build.

iamsaalim, 2020-06-06:
export cross compile 32 path to the arm toolchain

erwinabs, 2020-06-07:
Oke sir thanks you

Haxk20, 2020-07-01:
@jerpelea Please close this

closed 2020-07-02 by jerpelea