sonyxperiadev/kernel #2431: automativ kernel build for Xperia 1II

mskyaxl, 2021-07-24:
Hi,

I'm trying to compile the kernel everytime I build AOSP but removing the common-kernel as documented here
https://developer.sony.com/develop/open-devices/guides/kernel-compilation-guides/how-to-build-and-flash-a-linux-kernel-for-aosp-supported-devices
does not work.

I know that  common-kernel has been moved to the kernel folder as shown here https://github.com/sonyxperiadev/local_manifests/blob/android-11.0.0_r39/LA.UM.9.12.r1.xml but I get the same result.

Is there a way to compile them automatically?

Should the documentation be updated for Android 11?

Thanks!

Haxk20, 2021-07-25:
There is already a way to compile the kernel only. That is trough Android source tree and going to the common-kernel folder and using the script provided inside. Also this is a kernel repo. We do not usually support issues here.

So if you could move it to our bug_tracker repo and close this one more people will try to help you :)

MarijnS95, 2021-07-25:
Unfortunately the documentation is outdated and only applies to way older Android trees. We're poking Sony every once in a while to either update the documentation, mention that it's only applicable to older trees, or just remove it altogether as you are not the first one to run into this confusion.

> Is there a way to compile them automatically?

There is not. Google went [out of their way to prevent in-line kernel builds from happening](https://android.googlesource.com/platform/build/+/master/Changes.md#PATH_Tools). It is possible with a lot of trickery or by modifying the build system again, but this project aims to stay as close to AOSP as possible without such changes. Prebuilts are not a problem to the vast majority of SODP users anyway, only few dabble in kernel development and prebuilts are regenerated whenever interesting - and tested! - changes land.

Keep in mind that `kernel/sony/msm-4.19` is for Xperia 1/5II, and `4.14` is for everything else (as of writing).

Haxk20, 2021-07-26:
@jerpelea Please move this.

mskyaxl, 2021-10-21:
can be closed in this case. Thanks for clarification!