sonyxperiadev/kernel #2510: Software binaries for AOSP Android 11.0 – Kernel 4.14 – Yoshino v9a is missing a ton of stuff

moralrecordings, 2022-05-17:
Not sure if this is the correct place to raise this, but this may at least let people know why AOSP isn't working for them. The latest binary blobs package for Yoshino is missing a number of crucial libraries. This includes most of the radio stack, so RIL will be broken.

```
--- v8a.sorted	2022-05-18 01:14:45.502006636 +0800
+++ v9a.sorted	2022-05-18 01:14:53.168739270 +0800
@@ -9,8 +9,6 @@
 ./bin/hw
 ./bin/hw/qcrild
 ./bin/hw/vendor.qti.qspmhal@1.0-service
-./bin/hw/vendor.somc.hardware.miscta@1.0-service
-./bin/hw/vendor.somc.hardware.modemswitcher@1.0-service
 ./bin/imsdatadaemon
 ./bin/imsqmidaemon
 ./bin/imsrcsd
@@ -19,6 +17,7 @@
 ./bin/ks
 ./bin/mdm_helper
 ./bin/mlog_qmi_service
+./bin/msm_irqbalance
 ./bin/netmgrd
 ./bin/pd-mapper
 ./bin/pm-proxy
@@ -33,6 +32,7 @@
 ./bin/sct_service
 ./bin/sensors.qcom
 ./bin/sony-modem-switcher
+./bin/sscrpcd
 ./bin/tad_static
 ./bin/ta_qmi_service
 ./bin/tftp_server
@@ -239,7 +239,6 @@
 ./lib64/lib-imsvtcore.so
 ./lib64/lib-imsvt.so
 ./lib64/lib-imsxml.so
-./lib64/libizat_core.so
 ./lib64/libkeymasterdeviceutils.so
 ./lib64/libkeymasterprovision.so
 ./lib64/libkeymasterutils.so
@@ -247,7 +246,6 @@
 ./lib64/libllvm-glnext.so
 ./lib64/libllvm-qcom.so
 ./lib64/liblocdiagiface.so
-./lib64/liblowi_client.so
 ./lib64/liblqe.so
 ./lib64/libmdmdetect.so
 ./lib64/libmdmimgload.so
@@ -333,7 +331,6 @@
 ./lib64/libvpphvx.so
 ./lib64/libvppimmotion.so
 ./lib64/libxml.so
-./lib64/libxtadapter.so
 ./lib64/qcrild_librilutils.so
 ./lib64/qtibus.so
 ./lib64/qtimutex.so
@@ -353,38 +350,13 @@
 ./lib64/vendor.qti.hardware.data.cne.internal.constants@1.0.so
 ./lib64/vendor.qti.hardware.data.cne.internal.server@1.0.so
 ./lib64/vendor.qti.hardware.data.connection@1.0.so
-./lib64/vendor.qti.hardware.data.connection@1.1.so
 ./lib64/vendor.qti.hardware.data.dynamicdds@1.0.so
-./lib64/vendor.qti.hardware.data.iwlan@1.0.so
 ./lib64/vendor.qti.hardware.data.latency@1.0.so
 ./lib64/vendor.qti.hardware.data.qmi@1.0.so
 ./lib64/vendor.qti.hardware.dsp@1.0.so
 ./lib64/vendor.qti.hardware.mwqemadapter@1.0.so
-./lib64/vendor.qti.hardware.radio.am@1.0.so
 ./lib64/vendor.qti.hardware.radio.atcmdfwd@1.0.so
 ./lib64/vendor.qti.hardware.radio.ims@1.0.so
-./lib64/vendor.qti.hardware.radio.ims@1.1.so
-./lib64/vendor.qti.hardware.radio.ims@1.2.so
-./lib64/vendor.qti.hardware.radio.ims@1.3.so
-./lib64/vendor.qti.hardware.radio.ims@1.4.so
-./lib64/vendor.qti.hardware.radio.ims@1.5.so
-./lib64/vendor.qti.hardware.radio.ims@1.6.so
-./lib64/vendor.qti.hardware.radio.internal.deviceinfo@1.0.so
-./lib64/vendor.qti.hardware.radio.lpa@1.0.so
-./lib64/vendor.qti.hardware.radio.qcrilhook@1.0.so
-./lib64/vendor.qti.hardware.radio.qtiradio@1.0.so
-./lib64/vendor.qti.hardware.radio.qtiradio@2.0.so
-./lib64/vendor.qti.hardware.radio.qtiradio@2.1.so
-./lib64/vendor.qti.hardware.radio.qtiradio@2.2.so
-./lib64/vendor.qti.hardware.radio.qtiradio@2.3.so
-./lib64/vendor.qti.hardware.radio.qtiradio@2.4.so
-./lib64/vendor.qti.hardware.radio.uim@1.0.so
-./lib64/vendor.qti.hardware.radio.uim@1.1.so
-./lib64/vendor.qti.hardware.radio.uim@1.2.so
-./lib64/vendor.qti.hardware.radio.uim_remote_client@1.0.so
-./lib64/vendor.qti.hardware.radio.uim_remote_client@1.1.so
-./lib64/vendor.qti.hardware.radio.uim_remote_client@1.2.so
-./lib64/vendor.qti.hardware.radio.uim_remote_server@1.0.so
 ./lib64/vendor.qti.hardware.slmadapter@1.0.so
 ./lib64/vendor.qti.ims.callcapability@1.0.so
 ./lib64/vendor.qti.ims.callinfo@1.0.so
@@ -512,7 +484,6 @@
 ./lib/lib-imsvtcore.so
 ./lib/lib-imsvt.so
 ./lib/lib-imsxml.so
-./lib/libizat_core.so
 ./lib/libjpegdhw.so
 ./lib/libjpegdmahw.so
 ./lib/libjpegehw.so
@@ -523,7 +494,6 @@
 ./lib/libllvm-glnext.so
 ./lib/libllvm-qcom.so
 ./lib/liblocdiagiface.so
-./lib/liblowi_client.so
 ./lib/liblqe.so
 ./lib/libmdmdetect.so
 ./lib/libmdsprpc.so
@@ -702,7 +672,6 @@
 ./lib/libvpphvx.so
 ./lib/libvppimmotion.so
 ./lib/libxml.so
-./lib/libxtadapter.so
 ./lib/qcrild_librilutils.so
 ./lib/qtibus.so
 ./lib/qtimutex.so
@@ -738,38 +707,13 @@
 ./lib/vendor.qti.hardware.data.cne.internal.constants@1.0.so
 ./lib/vendor.qti.hardware.data.cne.internal.server@1.0.so
 ./lib/vendor.qti.hardware.data.connection@1.0.so
-./lib/vendor.qti.hardware.data.connection@1.1.so
 ./lib/vendor.qti.hardware.data.dynamicdds@1.0.so
-./lib/vendor.qti.hardware.data.iwlan@1.0.so
 ./lib/vendor.qti.hardware.data.latency@1.0.so
 ./lib/vendor.qti.hardware.data.qmi@1.0.so
 ./lib/vendor.qti.hardware.dsp@1.0.so
 ./lib/vendor.qti.hardware.mwqemadapter@1.0.so
-./lib/vendor.qti.hardware.radio.am@1.0.so
 ./lib/vendor.qti.hardware.radio.atcmdfwd@1.0.so
 ./lib/vendor.qti.hardware.radio.ims@1.0.so
-./lib/vendor.qti.hardware.radio.ims@1.1.so
-./lib/vendor.qti.hardware.radio.ims@1.2.so
-./lib/vendor.qti.hardware.radio.ims@1.3.so
-./lib/vendor.qti.hardware.radio.ims@1.4.so
-./lib/vendor.qti.hardware.radio.ims@1.5.so
-./lib/vendor.qti.hardware.radio.ims@1.6.so
-./lib/vendor.qti.hardware.radio.internal.deviceinfo@1.0.so
-./lib/vendor.qti.hardware.radio.lpa@1.0.so
-./lib/vendor.qti.hardware.radio.qcrilhook@1.0.so
-./lib/vendor.qti.hardware.radio.qtiradio@1.0.so
-./lib/vendor.qti.hardware.radio.qtiradio@2.0.so
-./lib/vendor.qti.hardware.radio.qtiradio@2.1.so
-./lib/vendor.qti.hardware.radio.qtiradio@2.2.so
-./lib/vendor.qti.hardware.radio.qtiradio@2.3.so
-./lib/vendor.qti.hardware.radio.qtiradio@2.4.so
-./lib/vendor.qti.hardware.radio.uim@1.0.so
-./lib/vendor.qti.hardware.radio.uim@1.1.so
-./lib/vendor.qti.hardware.radio.uim@1.2.so
-./lib/vendor.qti.hardware.radio.uim_remote_client@1.0.so
-./lib/vendor.qti.hardware.radio.uim_remote_client@1.1.so
-./lib/vendor.qti.hardware.radio.uim_remote_client@1.2.so
-./lib/vendor.qti.hardware.radio.uim_remote_server@1.0.so
 ./lib/vendor.qti.hardware.slmadapter@1.0.so
 ./lib/vendor.qti.ims.callcapability@1.0.so
 ./lib/vendor.qti.ims.callinfo@1.0.so

```

moralrecordings, 2022-05-17:
Here's the logs from logcat showing qcrild not starting. 
```
04-28 01:13:53.532 12313 12313 I flags_health_check: ServerConfigurableFlagsReset reset_mode value: 1
04-28 01:13:53.533 12313 12313 I flags_health_check: ServerConfigurableFlagsReset updatable crashing detected, resetting flags.
04-28 01:13:54.231     0     0 E init    : Control message: Could not find 'android.hardware.secure_element@1.2::ISecureElement/SIM1' for ctl.interface_start from pid: 502 (/system/bin/hwservicemanager)
04-28 01:13:54.259 12268 12268 W HidlServiceManagement: Waited one second for android.hardware.secure_element@1.2::ISecureElement/SIM1
04-28 01:13:54.260   502   502 I hwservicemanager: Since android.hardware.secure_element@1.2::ISecureElement/SIM1 is not registered, trying to start it as a lazy HAL.
04-28 01:13:54.261 12268 12268 I HidlServiceManagement: getService: Trying again for android.hardware.secure_element@1.2::ISecureElement/SIM1...
04-28 01:13:54.264   502 12314 W libc    : Unable to set property "ctl.interface_start" to "android.hardware.secure_element@1.2::ISecureElement/SIM1": error code: 0x20
04-28 01:13:54.482     0     0 E init    : Control message: Could not find 'android.hardware.radio@1.5::IRadio/slot1' for ctl.interface_start from pid: 502 (/system/bin/hwservicemanager)
04-28 01:13:54.511  4037  4037 W HidlServiceManagement: Waited one second for android.hardware.radio@1.5::IRadio/slot1
04-28 01:13:54.512   502   502 I hwservicemanager: Since android.hardware.radio@1.5::IRadio/slot1 is not registered, trying to start it as a lazy HAL.
04-28 01:13:54.513  4037  4037 I HidlServiceManagement: getService: Trying again for android.hardware.radio@1.5::IRadio/slot1...
04-28 01:13:54.515   502 12315 W libc    : Unable to set property "ctl.interface_start" to "android.hardware.radio@1.5::IRadio/slot1": error code: 0x20
04-28 01:13:55.236     0     0 E init    : Control message: Could not find 'android.hardware.secure_element@1.2::ISecureElement/SIM1' for ctl.interface_start from pid: 502 (/system/bin/hwservicemanager)
04-28 01:13:55.265 12268 12268 W HidlServiceManagement: Waited one second for android.hardware.secure_element@1.2::ISecureElement/SIM1
04-28 01:13:55.266   502   502 I hwservicemanager: Since android.hardware.secure_element@1.2::ISecureElement/SIM1 is not registered, trying to start it as a lazy HAL.
04-28 01:13:55.267 12268 12268 I HidlServiceManagement: getService: Trying again for android.hardware.secure_element@1.2::ISecureElement/SIM1...
04-28 01:13:55.270   502 12316 W libc    : Unable to set property "ctl.interface_start" to "android.hardware.secure_element@1.2::ISecureElement/SIM1": error code: 0x20
04-28 01:13:55.485     0     0 E init    : Control message: Could not find 'android.hardware.radio@1.5::IRadio/slot1' for ctl.interface_start from pid: 502 (/system/bin/hwservicemanager)
04-28 01:13:55.514  4037  4037 W HidlServiceManagement: Waited one second for android.hardware.radio@1.5::IRadio/slot1
04-28 01:13:55.515   502   502 I hwservicemanager: Since android.hardware.radio@1.5::IRadio/slot1 is not registered, trying to start it as a lazy HAL.
04-28 01:13:55.516  4037  4037 I HidlServiceManagement: getService: Trying again for android.hardware.radio@1.5::IRadio/slot1...
04-28 01:13:55.518   502 12317 W libc    : Unable to set property "ctl.interface_start" to "android.hardware.radio@1.5::IRadio/slot1": error code: 0x20
04-28 01:13:56.207     0     0 D logd    : logdr: UID=0 GID=0 PID=12318 b tail=0 logMask=99 pid=0 start=0ns timeout=0ns
04-28 01:13:56.268 12268 12268 W HidlServiceManagement: Waited one second for android.hardware.secure_element@1.2::ISecureElement/SIM1
04-28 01:13:56.269   502   502 I hwservicemanager: Since android.hardware.secure_element@1.2::ISecureElement/SIM1 is not registered, trying to start it as a lazy HAL.
04-28 01:13:56.271 12268 12268 I HidlServiceManagement: getService: Trying again for android.hardware.secure_element@1.2::ISecureElement/SIM1...
04-28 01:13:56.275   502 12320 W libc    : Unable to set property "ctl.interface_start" to "android.hardware.secure_element@1.2::ISecureElement/SIM1": error code: 0x20
04-28 01:13:56.241     0     0 E init    : Control message: Could not find 'android.hardware.secure_element@1.2::ISecureElement/SIM1' for ctl.interface_start from pid: 502 (/system/bin/hwservicemanager)
04-28 01:13:56.517  4037  4037 W HidlServiceManagement: Waited one second for android.hardware.radio@1.5::IRadio/slot1
04-28 01:13:56.519   502   502 I hwservicemanager: Since android.hardware.radio@1.5::IRadio/slot1 is not registered, trying to start it as a lazy HAL.
04-28 01:13:56.521  4037  4037 I HidlServiceManagement: getService: Trying again for android.hardware.radio@1.5::IRadio/slot1...
04-28 01:13:56.525   502 12321 W libc    : Unable to set property "ctl.interface_start" to "android.hardware.radio@1.5::IRadio/slot1": error code: 0x20
04-28 01:13:56.491     0     0 E init    : Control message: Could not find 'android.hardware.radio@1.5::IRadio/slot1' for ctl.interface_start from pid: 502 (/system/bin/hwservicemanager)
04-28 01:13:57.272 12268 12268 W HidlServiceManagement: Waited one second for android.hardware.secure_element@1.2::ISecureElement/SIM1
04-28 01:13:57.273   502   502 I hwservicemanager: Since android.hardware.secure_element@1.2::ISecureElement/SIM1 is not registered, trying to start it as a lazy HAL.
[... cut ...]
04-28 01:13:58.537 12325 12325 E MSM-irqbalance: Failed to sleep for 5 seconds
04-28 01:13:58.537 12325 12325 I MSM-irqbalance: msm-irq_balance End
```
When invoked manually as root, we get a missing library error.
```
lilac:/ # /odm/bin/hw/qcrild                                                                                                                                              
CANNOT LINK EXECUTABLE "/odm/bin/hw/qcrild": library "vendor.qti.hardware.radio.ims@1.1.so" not found: needed by /odm/lib64/libril-qc-hal-qmi.so in namespace (default)
```

MarijnS95, 2022-05-17:
Now we're talking.

It looks like these have never been explicitly required:

https://github.com/sonyxperiadev/device-sony-common/blame/master/common-odm.mk#L189

And the build for `qcrild` probably messed up the compile-time dependency? @jerpelea this is not something I can easily debug on the OSS side :wink:

closed 2022-10-05 by jerpelea