sonyxperiadev/kernel #2539: Camera issue on Xperia 1(SO-03L)

kato-cuebs, 2022-09-26:
After executing the capture method of CameraCaptureSession, if CameraDevice is closed before onImageAvailableListner is executed, the camera can no longer be operated and the following error will occur.

```
2022-09-14 15:01:13.003 6583-11439/jp.co.recruit.erikura E/CameraCaptureSession: Session 0: Failed to create capture session; configuration failed
2022-09-14 15:01:13.003 6583-11473/jp.co.recruit.erikura V/CAMERA: CAMERA: CameraCaptureSession onConfigureFailed row：2140
2022-09-14 15:01:13.003 6583-11439/jp.co.recruit.erikura W/System.err: android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): endConfigure:527: Camera 0: Error configuring streams: Function not implemented (-38)
```

Could you tell me the cause and workaround?

sraase, 2022-09-26:
I need more logs in order to see what goes wrong.

Please run the following commands:
> setprop persist.vendor.camera.logErrorMask 0xFFFFFFFF
> setprop persist.vendor.camera.logWarningMask 0xFFFFFFFF
> setprop persist.vendor.camera.logInfoMask 0xFFFFFFFF

Then reboot the phone and collect the logs. Note that these commands will increase the log output a lot, so try to run "adb logcat -G 64m" before starting; otherwise some lines might be dropped.

Thank you.

kato-cuebs, 2022-10-06:
Thank you for your reply.

I don't have the device in question at the moment.
As soon as I get it, I will try to get the log in the form you taught me.

Thank you.

closed 2023-09-11 by jerpelea